project-chip/connectedhomeip #36561: Dependency via ifdef usage cleanup

andy31415, 2024-11-19:
We have significant logic that includes cpp files and then includes/excludes the entire file or large sections of it via `#ifdef`. We should switch to compile time dependencies instead and reasonable injection for members.

Starter is several such usages in CHEF (e.g. CHIP_CONFIG_ENABLE_ICD_SERVER) however also size optimizations are trying to use a `CHIP_CONFIG_ENABLE_READ_CLIENT` that dynamicall changes content and size of our large InteractionModelEngine (we should try to make our objects not as large contains-everything instead)

gabc-cpp, 2025-06-23:
List of `.cpp` files with `#ifdef` references:

```
./ble/BtpEngine.cpp
./ble/BLEEndPoint.cpp
./app/util/attribute-storage.cpp
./app/util/privilege-storage.cpp
./app/clusters/diagnostic-logs-server/diagnostic-logs-server.cpp
./app/clusters/color-control-server/color-control-server.cpp
./app/clusters/pump-configuration-and-control-server/pump-configuration-and-control-server.cpp
./app/clusters/mode-select-server/mode-select-server.cpp
./app/clusters/level-control/level-control.cpp
./app/clusters/mode-base-server/mode-base-server.cpp
./app/clusters/groups-server/groups-server.cpp
./app/clusters/general-diagnostics-server/general-diagnostics-server.cpp
./app/clusters/administrator-commissioning-server/administrator-commissioning-server.cpp
./app/clusters/on-off-server/on-off-server.cpp
./app/clusters/valve-configuration-and-control-server/valve-configuration-and-control-server.cpp
./app/clusters/wifi-network-diagnostics-server/wifi-network-diagnostics-server.cpp
./app/clusters/window-covering-server/window-covering-server.cpp
./app/server/java/CHIPAppServer-JNI.cpp
./platform/realtek/BEE/FactoryDataProvider.cpp
./platform/realtek/BEE/Logging.cpp
./platform/webos/Logging.cpp
./platform/qpg/OTAImageProcessorImpl.cpp
./platform/nxp/rt/rt1170/PlatformManagerImpl.cpp
./platform/nxp/rt/rt1060/PlatformManagerImpl.cpp
./platform/nxp/rt/rw61x/PlatformManagerImpl.cpp
./platform/nxp/common/NetworkCommissioningWiFiDriver.cpp
./platform/nxp/common/ble/BLEManagerImpl.cpp
./platform/nxp/common/ble_zephyr/BLEManagerImpl.cpp
./platform/nxp/mcxw71/Logging.cpp
./platform/nxp/mcxw71/PlatformManagerImpl.cpp
./platform/nxp/zephyr/DiagnosticDataProviderImplNxp.cpp
./platform/nxp/zephyr/ota/OTAImageProcessorImpl.cpp
./platform/Beken/Logging.cpp
./platform/tests/TestKeyValueStoreMgr.cpp
./platform/tests/TestPlatformTime.cpp
./platform/FreeRTOS/SystemTimeSupport.cpp
./platform/mbed/Logging.cpp
./platform/mbed/OTAImageProcessorImpl.cpp
./platform/ESP32/OpenthreadLauncher.cpp
./platform/ESP32/Logging.cpp
./platform/ESP32/ESP32SecureCertDACProvider.cpp
./platform/ESP32/ConnectivityManagerImpl_WiFi.cpp
./platform/ESP32/SystemTimeSupport.cpp
./platform/ESP32/ESP32FactoryDataProvider.cpp
./platform/ESP32/bluedroid/BLEManagerImpl.cpp
./platform/ESP32/DiagnosticDataProviderImpl.cpp
./platform/ESP32/OTAImageProcessorImpl.cpp
./platform/ESP32/nimble/BLEManagerImpl.cpp
./platform/ESP32/ConfigurationManagerImpl.cpp
./platform/stm32/FactoryDataProvider.cpp
./platform/stm32/CHIPMem-Platform.cpp
./platform/ASR/ASRUtils.cpp
./platform/ASR/ConnectivityManagerImpl.cpp
./platform/ASR/KeyValueStoreManagerImpl.cpp
./platform/ASR/BLEManagerImpl.cpp
./platform/ASR/BLEAppSvc.cpp
./platform/ASR/DiagnosticDataProviderImpl.cpp
./platform/ASR/ConfigurationManagerImpl.cpp
./platform/ASR/ASRFactoryDataProvider.cpp
./platform/Ameba/FactoryDataProvider.cpp
./platform/Darwin/dnssd/DnssdContexts.cpp
./platform/Infineon/crypto/trustm/DeviceAttestationCredsExampleTrustM.cpp
./platform/Infineon/CYW30739/BLEManagerImpl.cpp
./platform/Infineon/PSOC6/MTBKeyValueStore.cpp
./platform/Infineon/PSOC6/cycfg_bt_settings.cpp
./platform/Infineon/PSOC6/BLEManagerImpl.cpp
./platform/Infineon/PSOC6/OTAImageProcessorImpl.cpp
./platform/bouffalolab/common/BLEManagerImpl.cpp
./platform/bouffalolab/common/DiagnosticDataProviderImpl.cpp
./platform/nrfconnect/FactoryDataProvider.cpp
./platform/nrfconnect/DiagnosticDataProviderImplNrf.cpp
./platform/nrfconnect/OTAImageProcessorImpl.cpp
./platform/nrfconnect/Reboot.cpp
./platform/silabs/wifi/lwip-support/ethernetif.cpp
./platform/silabs/wifi/WifiInterface.cpp
./platform/silabs/wifi/wf200/WifiInterfaceImpl.cpp
./platform/silabs/wifi/rs911x/WifiInterfaceImpl.cpp
./platform/silabs/wifi/wiseconnect-interface/WiseconnectWifiInterface.cpp
./platform/silabs/wifi/SiWx/WifiInterfaceImpl.cpp
./platform/silabs/NetworkCommissioningWiFiDriver.cpp
./platform/silabs/Logging.cpp
./platform/silabs/SiWx917/CHIPCryptoPALTinyCrypt.cpp
./platform/silabs/SiWx917/OTAImageProcessorImpl.cpp
./platform/silabs/multi-ota/OTATlvProcessor.cpp
./platform/silabs/multi-ota/OTAFactoryDataProcessor.cpp
./platform/silabs/multi-ota/OTAFirmwareProcessor.cpp
./platform/silabs/platformAbstraction/GsdkSpam.cpp
./platform/silabs/platformAbstraction/WiseMcuSpam.cpp
./platform/silabs/rs911x/BLEManagerImpl.cpp
./platform/silabs/efr32/ThreadStackManagerImpl.cpp
./platform/silabs/efr32/OTAImageProcessorImpl.cpp
./platform/silabs/ConfigurationManagerImpl.cpp
./platform/telink/OTAImageProcessorImpl.cpp
./platform/Zephyr/Logging.cpp
./platform/Zephyr/SysHeapMalloc.cpp
./platform/Zephyr/BLEManagerImpl.cpp
./platform/Zephyr/DiagnosticDataProviderImpl.cpp
./platform/Zephyr/ConfigurationManagerImpl.cpp
./platform/SyscallStubs.cpp
./platform/ti/cc13xx_26xx/BLEManagerImpl.cpp
./platform/ti/cc13xx_26xx/ThreadStackManagerImpl.cpp
./platform/mt793x/Logging.cpp
./platform/mt793x/DnssdContexts.cpp
./platform/mt793x/ConnectivityManagerImpl_WIFI.cpp
./platform/mt793x/BLEManagerImpl.cpp
./platform/mt793x/ConfigurationManagerImpl.cpp
./controller/python/chip/internal/ChipThreadWork.cpp
./controller/java/AndroidCallbacks.cpp
./controller/java/AndroidDeviceControllerWrapper.cpp
./controller/java/CHIPDeviceController-JNI.cpp
./system/SystemClock.cpp
./inet/InetInterface.cpp
./inet/tests/inet-layer-test-tool.cpp
./inet/UDPEndPointImplLwIP.cpp
./inet/UDPEndPointImplSockets.cpp
./inet/TCPEndPointImplSockets.cpp
./wifipaf/WiFiPAFLayer.cpp
./wifipaf/WiFiPAFTP.cpp
./wifipaf/WiFiPAFEndPoint.cpp
./lib/core/tests/TestTLV.cpp
./lib/shell/streamer_esp32.cpp
./lib/shell/streamer_nxp.cpp
./lib/shell/streamer_asr.cpp
./lib/support/Base64.cpp
./lib/support/JniReferences.cpp
./lib/support/CHIPMem-Malloc.cpp
./lib/support/PersistentStorageAudit.cpp
./lib/dnssd/Discovery_ImplPlatform.cpp
./lib/dnssd/Advertiser_ImplMinimalMdns.cpp
./lib/dnssd/Resolver_ImplMinimalMdns.cpp
./protocols/user_directed_commissioning/UserDirectedCommissioningClient.cpp
./credentials/CHIPCert.cpp
```

gabc-cpp, 2025-06-23:
List of `.h` files with `#ifdef` references:

```
./test_driver/efr32/include/FreeRTOSConfig.h
./test_driver/openiotsdk/unit-tests/freertos-config/FreeRTOSConfig.h
./crypto/tests/TestCryptoLayer.h
./transport/raw/Tuple.h
./transport/SessionManager.h
./ble/BleConfig.h
./app/util/af-types.h
./app/util/attribute-storage.h
./app/util/types_stub.h
./app/clusters/color-control-server/color-control-server.h
./app/clusters/level-control/level-control.h
./app/clusters/on-off-server/on-off-server.h
./lwip/cc32xx/arch/cc.h
./lwip/standalone/arch/perf.h
./lwip/standalone/lwipopts.h
./lwip/bl702l/arch/cc.h
./lwip/bl702/arch/cc.h
./lwip/silabs/lwipopts-wf200.h
./lwip/silabs/lwipopts-rs911x.h
./lwip/silabs/arch/cc.h
./lwip/k32w1/arch/cc.h
./lwip/k32w0/arch/cc.h
./lwip/mt793x/arch/cc.h
./lwip/mt793x/lwipopts.h
./include/platform/ConnectivityManager.h
./include/platform/PlatformManager.h
./include/platform/KeyValueStoreManager.h
./include/platform/internal/NFCCommissioningManager.h
./include/platform/internal/BLEManager.h
./include/platform/internal/GenericConnectivityManagerImpl_NoBLE.h
./include/platform/internal/GenericPlatformManagerImpl_CMSISOS.h
./include/platform/internal/GenericConnectivityManagerImpl_NoWiFi.h
./include/platform/internal/GenericConnectivityManagerImpl_WiFi.h
./include/platform/CHIPDeviceConfig.h
./include/platform/NFCOnboardingPayloadManager.h
./include/platform/ConfigurationManager.h
./include/platform/CHIPDeviceEvent.h
./include/platform/ThreadStackManager.h
./platform/cc32xx/FreeRTOSConfig.h
./platform/nxp/crypto/se05x/CHIPCryptoPALHsm_se05x_utils.h
./platform/nxp/common/CHIPNXPPlatformDefaultConfig.h
./platform/nxp/common/ble_zephyr/BLEManagerImpl.h
./platform/nxp/common/CHIPDeviceNXPPlatformDefaultConfig.h
./platform/nxp/common/ram_storage.h
./platform/nxp/mcxw71/CHIPDevicePlatformConfig.h
./platform/nxp/mcxw71/k32w1-chip-mbedtls-config.h
./platform/nxp/mcxw71/ram_storage.h
./platform/nxp/zephyr/DiagnosticDataProviderImplNxp.h
./platform/nxp/zephyr/CHIPDevicePlatformConfig.h
./platform/nxp/zephyr/CHIPPlatformConfig.h
./platform/mbed/PlatformManagerImpl.h
./platform/ESP32/OTAImageProcessorImpl.h
./platform/ESP32/CHIPDevicePlatformEvent.h
./platform/ESP32/ESP32SecureCertDACProvider.h
./platform/ESP32/CHIPDevicePlatformConfig.h
./platform/ESP32/BLEManagerImpl.h
./platform/ESP32/CHIPPlatformConfig.h
./platform/ESP32/route_hook/ESP32RouteTable.h
./platform/ESP32/route_hook/ESP32RouteHook.h
./platform/ESP32/nimble/blecent.h
./platform/ESP32/ESP32FactoryDataProvider.h
./platform/ESP32/ChipDeviceScanner.h
./platform/stm32/STM32FreeRtosHooks.h
./platform/ASR/BLEAppSvc.h
./platform/ASR/route_hook/asr_route_hook.h
./platform/ASR/route_hook/asr_route_table.h
./platform/ASR/ASRFactoryDataParser.h
[... 55 lines cut ...]
./platform/mt793x/mt793x-mbedtls-config.h
./pw_backends/assert/public_overrides/pw_assert_backend/check_backend.h
./controller/java/AndroidDeviceControllerWrapper.h
./system/SystemLayerImpl.h
./system/SystemConfig.h
./system/SystemError.h
./inet/TCPEndPointImpl.h
./inet/UDPEndPointImpl.h
./inet/InetConfig.h
./lib/core/CHIPEncoding.h
./lib/core/CHIPConfig.h
./lib/core/CHIPSafeCasts.h
./lib/address_resolve/AddressResolve.h
./lib/support/CHIPPlatformMemory.h
./lib/support/Pool.h
./lib/support/CHIPFaultInjection.h
./lib/support/PrivateHeap.h
./lib/dnssd/TxtFields.h
./darwin/Framework/CHIP/MTRDefines.h
./darwin/Framework/CHIP/MTRDefines_Internal.h
./darwin/Framework/CHIP/MTRDeviceController_XPC.h
./darwin/Framework/CHIP/MTRDevice_Internal.h
./darwin/Framework/CHIP/MTRDevice_Concrete.h
./darwin/Framework/CHIPTests/TestHelpers/MTRTestDeclarations.h
```

andy31415, 2025-07-10:
https://github.com/project-chip/connectedhomeip/blob/master/src/controller/CommissionerDiscoveryController.cpp#L32 does the same thing.

andy31415, 2025-07-10:
https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/joint-fabric-datastore-server/joint-fabric-datastore-server.cpp#L32 also seems wrong - since these files are included by a zap config, why are we having an extra ifdef? That seems dodgy

gabc-cpp, 2025-07-11:
Pivoting to specific cases mentioned above by @andy31415

gabc-cpp, 2025-08-22:
Working on this issue and related subissues. Refactoring the file inclusion to use regular `#include` instead of `#ifdef` for the whole files.
Doing changes for subissue `#40021`.