project-chip/connectedhomeip #36574: [trivial]

aravindone, 2024-11-20:
### Type

Other

### Explanation

need to know how the command works in code ,do we have any API for that ?  my need is to detect the command received from controller for on and off cluster and to execute my function. and i can see whether i can do on same firmware for turning on and off command in on-off cluster. 

### Platform

esp32

### Platform Version(s)

1.3

### Testing

Manually tested with SDK

### Anything else?

@bzbarsky-apple please comment

aravindone, 2024-11-21:
Hi @bzbarsky-apple 

>  How that cluster then interacts with the application depends on the cluster

I want to understand this and I'm currently working in on-off Cluster in light-switch app

bzbarsky-apple, 2024-11-22:
So is this a _client_ cluster or a _server_ cluster?

aravindone, 2024-11-22:
Hi @bzbarsky-apple  the The cluster in zap tool says _client_ cluster but i need it as _server_ cluster, when i try to activate as  _server_  and compile code im getting error which is based out of ZAP, do i need to change any file apart from Zap file

bzbarsky-apple, 2024-11-22:
@aravindone The _light switch app_ does not have a server On/Off cluster.  It has a client.

It's really hard to say anything clear about your error, since you did not bother to say what exact error you got.  Please attach (not paste) logs as needed.

aravindone, 2024-11-25:
Hello @bzbarsky-apple 
Sorry for the late reply

Here my log 
[Light_switch_error_log.TXT](https://github.com/user-attachments/files/17903171/Light_switch_error_log.TXT)

this is the Zap configuration, but in the tool it clearly mentions that on-off cluster is dedicated for Client. can i able to change it ? 
[image]
[image]

bzbarsky-apple, 2024-12-05:
@aravindone After you changed the ZAP to have a server cluster, did you rerun codegen for the application?  Something like `./scripts/tools/zap/generate.py examples/light-switch-app/light-switch-common/light-switch-app.zap`?

PR #38288 (2025-04-24): reconcile Common Area / Location namespaces to current spec
Common Area: remove duplicates with different names; add an item
Common Location: added
FloorSurface Tags: removed

fixes #36574 

#### Testing

- [x] I think this is the first instance of using numbered `Reserved` values, so need to check how those are handled by various codegens.