project-chip/connectedhomeip #36579: [Feature] Fix OnOff state shadowing by the LevelControl cluster while OnOff transitions are in progress

lmapii, 2024-11-20:
### Feature description

In the current implementation of the `level-control.cpp` cluster and `on-off-server.cpp` cluster, the LevelControl server shadows the OnOff state if it is performing an On/Off transition and incorrectly changes the CurrentLevel in case an On transition is interrupted by an OnOff::Command::Off. Details explained below. Assume a DimmableLight device with LevelControl/OnOffTransitiontime set to 3 seconds.

# Problem (1) OnOff::Command::Off may unintentionally manipulate the CurrentLevel

- Turn on the device using the command OnOff::Command::On (or `chip-tool onoff toggle <node-id> 1`)
- The dimmable light starts the Off->On transition
- While the transition is in progress, issue the command OnOff::Command::Off

Only at the end of the sequence the `level-control.cpp` implementation updates the OnOff state to "Off" and sets "CurrentLevel" - but it now updates the "CurrentLevel" to the level that was present when receiving the OnOff::Command::Off.

Imagine you had a dimmable light at home that has such an OnOffTransitiontime. If you interrupt the effect, e.g., because you inadvertently turned on a light and just want to turn it off again, you know changed the light setting to the level where you interrupted your light. I'd not expect the next On level to change just because I turned my light back off.

# Problem (2) OnOff::Command::On is ignored while the LevelControl cluster is turning off the light

- Turn on the device using the command OnOff::Command::On (or `chip-tool onoff toggle <node-id> 1`)
- The dimmable light starts the Off->On transition
- While the transition is in progress, issue the command OnOff::Command::Off
- The dimmable light now starts the On->Off transition, but the OnOff state is still "On"
- While the transition is in progress, issue the command OnOff::Command::On
- This command is now ignored since the current implementation in `on-off-server.cpp`

The `on-off-server.cpp` currently has no means to know that the `level-control.cpp` is performing a transition to "Off" and thus the future state of the OnOff attribute will be set to "Off" once the effect is complete.

---

I have a rough implementation ready which would fix both problems, but I don't know if the behavior is expected to be as it is (though I'd be bummed if I couldn't turn my light on/off at will just because an effect is in progress). However, this fix also requires a callback (similar to `emberAfOnOffClusterLevelControlEffectCallback`) such that `on-off-server.cpp` can read the "future OnOff state" from the LevelControl cluster, if present.

I can open a pull request if you also think that the behavior should be different.

### Platform

all

jmartinez-silabs, 2024-11-21:
@lmapii you are right that this seems like a faulty behavior. Of course I haven't seen the cod you provided yet, but from memory of the cluster implementation I think we can solve the situation with a small change.
If we change the StoredLevel variable to be nullable. Set it to null after the transition is completed and storedLevel was reapplied to currentLevel, and on a transition start only update StoredLevel if it is null

lmapii, 2024-11-21:
I think that changing StoredLevel to nullable alone won't do the trick. What I currently use is just a hack and I'm not very happy about it but you can see it here: https://github.com/lmapii/sdk-connectedhomeip/commit/fe07f7488af129c6fc207c23630a46e7b5b4c6ad

It'd be some work to align this with the current version of this repository - I'm using the nRF SDK which uses its own downstream branch where they cherry-pick commits for their SDK releases. I'd also still need to get a standalone CHIP/Matter project up and running to test this (e.g., compile a dimmable light host application).

Using a nullable won't fix problem (2) since OnOff commands are ignored:
https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/on-off-server/on-off-server.cpp#L377

I hope the mess I've created in the given commit helps a bit.

jmartinez-silabs, 2024-12-12:
@lmapii Sorry, I have been working on an internal project and didn't have time to look deeper into this matter. I will be able to work on it at the beginning of 2025 butIf you want to contribute your fix in a PR in this repo before that please do so and you can @ me I'll review it there too.

lmapii, 2024-12-18:
Hi - no worries. The next couple of months I prob. won't have any time to provide a proper pull request, sorry. As mentioned, this would be especially difficult for me since I never used the project out of the Zephyr/NCS context, but if I find some time I'd try to contribute properly. For me and for now it'd be fine to just keep the issue open until either has the time to fix it. Happy holidays!

lmapii, 2024-12-20:
Oh well, seems like Nordic decided to release a new SDK version which contains the latest changes from this repository. Happy to continue the discussion in the pull request.

closed 2025-10-16 by lmapii