project-chip/connectedhomeip #36582: [BUG] TC-ICDM-2.1 Fails when DUT does not support ICDM.S.A0006 and ICDM.S.A0007 optional attributes

jrhees-cae, 2024-11-20:
### Reproduction steps

When TH test TC-ICDM-2.1 python test is run against a DUT that does not support ICDM.S.A0006 and ICDM.S.A0007 optional attributes, the test errors out with the following:

```
[MatterTest] 11-20 18:00:19.777 ERROR Exception occurred in test_TC_ICDM_2_1.
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/dist-packages/mobly/base_test.py", line 818, in exec_one_test
    test_method()
  File "/root/python_testing/scripts/sdk/matter_testing_support.py", line 1985, in async_runner
    return _async_runner(body, self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/root/python_testing/scripts/sdk/matter_testing_support.py", line 1973, in _async_runner
    return asyncio.run(runner_with_timeout)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/runners.py", line 194, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/base_events.py", line 687, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/asyncio/tasks.py", line 520, in wait_for
    return await fut
           ^^^^^^^^^
  File "/root/python_testing/scripts/sdk/TC_ICDM_2_1.py", line 234, in test_TC_ICDM_2_1
    self.set_bits_count(uatHintInstructionDepedentBitmap), 1, "UserActiveModeTriggerHint has more than 1 bit that is dependent on the UserActiveModeTriggerInstruction")
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnboundLocalError: cannot access local variable 'uatHintInstructionDepedentBitmap' where it is not associated with a value
[MatterTest] 11-20 18:00:19.778 INFO <<< [E:8992i S:56786 M:70646265 (Ack:15164239)] (S) Msg TX from 000000000001B669 to 1:0000000012344321 [E4B6] [UDP:[fd11:22::92d5:aa90:2c56:e060]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:34)
[MatterTest] 11-20 18:00:19.793 ERROR 

******************************************************************
*
* Test test_TC_ICDM_2_1 failed for the following reason:
* cannot access local variable 'uatHintInstructionDepedentBitmap' where it is not associated with a value
*
* File "/root/python_testing/scripts/sdk/TC_ICDM_2_1.py", line 234, in test_TC_ICDM_2_1
* UnboundLocalError: cannot access local variable 'uatHintInstructionDepedentBitmap' where it is not associated with a value
*
*******************************************************************

[MatterTest] 11-20 18:00:19.794 INFO [Test] test_TC_ICDM_2_1 ERROR
[MatterTest] 11-20 18:00:19.803 INFO Summary for test class TC_ICDM_2_1: Error 1, Executed 1, Failed 0, Passed 0, Requested 1, Skipped 0
[MatterTest] 11-20 18:00:19.805 INFO Summary for test run MatterTest@11-20-2024_18-00-15-355:
Total time elapsed 4.401238357997499s
Artifacts are saved in "/tmp/matter_testing/logs/MatterTest/11-20-2024_18-00-15-355"
Test summary saved in "/tmp/matter_testing/logs/MatterTest/11-20-2024_18-00-15-355/test_summary.yaml"
Test results: Error 1, Executed 1, Failed 0, Passed 0, Requested 1, Skipped 0
```

Changing the python code as follows allows the test to complete:
```
diff TC_ICDM_2_1.py TC_ICDM_2_1.py.orig
233,234c233,234
<             asserts.assert_less_equal(
<                 self.set_bits_count(uatHintInstructionDepedentBitmap), 1, "UserActiveModeTriggerHint has more than 1 bit that is dependent on the UserActiveModeTriggerInstruction")
---
>         asserts.assert_less_equal(
>             self.set_bits_count(uatHintInstructionDepedentBitmap), 1, "UserActiveModeTriggerHint has more than 1 bit that is dependent on the UserActiveModeTriggerInstruction")
264,267c264,266
<             if self.pics_guard(self.check_pics("ICDM.S.A0006")):
<                 # Check if the UserActiveModeTriggerInstruction was required
<                 asserts.assert_false(uatHintInstructionDepedentBitmap in kUatInstructionMandatoryBitMask,
<                                      "UserActiveModeTriggerHint requires the UserActiveModeTriggerInstruction")
---
>             # Check if the UserActiveModeTriggerInstruction was required
>             asserts.assert_false(uatHintInstructionDepedentBitmap in kUatInstructionMandatoryBitMask,
>                                  "UserActiveModeTriggerHint requires the UserActiveModeTriggerInstruction")
```

### Bug prevalence

Consistent

### GitHub hash of the SDK that was being used

main

### Platform

python

bzbarsky-apple, 2024-11-20:
@mkardous-silabs

mkardous-silabs, 2024-11-26:
@jrhees-cae Test Script in the SDK have the issue fixed but the update script didn't make it in the test harness for 1.4.