project-chip/connectedhomeip #36622: Enforce ember decoupling from libCHIP

andy31415, 2024-11-22:
Some basic restrictions on ember calls exists on https://github.com/project-chip/connectedhomeip/blob/master/.github/workflows/lint.yml#L275 

It currently focuses on `emberAfReadAttribute` and `emberAfWriteAttribute` however over time we should disallow any `emberAf` (or related calls) within the entire repository except `src/app/util` and `src/app/codegen-data-mode-provider`.

- [ ] figure out what methods to deny (a start is `emAf` and `emberAf` calls, however overall we want to deny anything inside `src/app/util`
- [ ] determine temporary or permanent exception lists for the above: some tests still define or use ember functions and migration would take a while. We have to be able to have exceptions (maybe we just allow all tests to start)
- [ ] Add enforcement either via grep or via inspecting the final symbol table in libCHIP

khodya, 2025-02-07:
break down:
ember functions can have different naming conventions, 95% of them will have 'em' prefix, need to:
1) find and document found functions
2) replace 
3) add checks to CI

jennygaz, 2025-02-20:
While working on this, this was my process:
1. Find all symbols for ember, so all symbols that start with `emberAf` or `emAf` for now. This was done with regular expressions and some `grep` + `find` + `sed`, the script used for it is here:
```sh
#!/bin/bash

# Created by jennygaz

# Create temporary files to store results
touch temp_results.txt
touch filtered_results.txt
touch final_results.txt

echo "Searching for symbols starting with 'emAf' or 'emberAf'..."

# Find all C and C++ files and search for both patterns
# Using two separate greps and combining results
find . -type f \( -name "*.c" -o -name "*.cpp" -o -name "*.cc" -o -name "*.h" -o -name "*.hpp" \) -exec grep -H '\<emAf[A-Za-z0-9_]*' {} \; > temp_results.txt
find . -type f \( -name "*.c" -o -name "*.cpp" -o -name "*.cc" -o -name "*.h" -o -name "*.hpp" \) -exec grep -H '\<emberAf[A-Za-z0-9_]*' {} \; >> temp_results.txt

# Process results to extract symbols and format output
while IFS= read -r line; do
    file=$(echo "$line" | cut -d':' -f1)
    content=$(echo "$line" | cut -d':' -f2-)
    
    # Extract both emAf and emberAf symbols from the line
    symbols=$(echo "$content" | grep -o '\<\(emAf\|emberAf\)[A-Za-z0-9_]*')
    
    # For each symbol found in the line
    while IFS= read -r symbol; do
        if [ ! -z "$symbol" ]; then
            escaped_content=$(echo "$content" | sed 's/"/\\"/g' | sed 's/|/\\|/g' | tr -s ' ')
            echo "Symbol: $symbol | File: $file | Line: $escaped_content" >> filtered_results.txt
        fi
    done <<< "$symbols"
done < temp_results.txt

# Sort by symbol name, then by file path, and remove exact duplicates
# The 'sort -u' will keep only unique complete lines
sort -u filtered_results.txt > final_results.txt

# Count unique symbols
echo -e "\nUnique symbols found:"
cat final_results.txt | cut -d'|' -f1 | sort -u | sed 's/Symbol: //'

# Print full results
echo -e "\nFull results (with context):"
cat final_results.txt

# Display statistics
symbol_count=$(cat final_results.txt | cut -d'|' -f1 | sort -u | wc -l)
occurrence_count=$(cat final_results.txt | wc -l)
echo -e "\nStatistics:"
echo "Unique symbols found: $symbol_count"
echo "Total occurrences: $occurrence_count"

# Clean up temporary files
rm temp_results.txt filtered_results.txt final_results.txt
```

2. Afterwards, I edited the `lint.yml` file to reflect the changes and came up with the following version:

[YAML file](https://github.com/jennygaz/connectedhomeip/blob/update-lint-exclude-ember-symbols/.github/workflows/lint.yml)

3. Currently, the issues to tackle are:
* Finish testing the Github Action, [it is running in my fork, called Lint Code Base](https://github.com/jennygaz/connectedhomeip/actions)
* Solve issues that might arise
* Clean up in case there are more jobs regarding ember symbols
* Possibly, speed up the action

jennygaz, 2025-03-10:
When testing (and excluding the `src/app/util` directory), I found the following files that are being used right now:
- [ ] at `src/app/dynamic_server/DynamicDispatcher.cpp:278` using `Protocols::InteractionModel::Status emAfReadOrWriteAttribute(const EmberAfAttributeSearchRecord * attRecord...`
- [ ] at `src/app/dynamic_server/DynamicDispatcher.cpp:324` using `Protocols::InteractionModel::Status emAfWriteAttributeExternal(const ConcreteAttributePath & path...`
- [ ] at `src/darwin/Framework/CHIP/ServerEndpoint/MTRIMDispatch.mm:42` using `Protocols::InteractionModel::Status emAfWriteAttributeExternal(const ConcreteAttributePath & path...`
- [ ] at `src/data-model-providers/codegen/CodegenDataModelProvider_Read.cpp:130 using `Protocols::InteractionModel::Status status = emAfReadOrWriteAttribute(...`
- [ ] at `src/data-model-providers/codegen/CodegenDataModelProvider_Write.cpp:190` using `status = emAfWriteAttributeExternal(request.path, dataInput);`
- [ ] at `src/data-model-providers/codegen/tests/EmberReadWriteOverride.cpp:81` using `Status emAfReadOrWriteAttribute(const EmberAfAttributeSearchRecord * attRecord, const EmberAfAttributeMetadata ** metadata...`
- [ ] at `src/data-model-providers/codegen/tests/EmberReadWriteOverride.cpp:106` using `Status emAfWriteAttributeExternal(const chip::app::ConcreteAttributePath & path, const EmberAfWriteDataInput & input)`
- [ ] at `src/data-model-providers/codegen/tests/EmberReadWriteOverride.cpp:136` using `return emAfWriteAttributeExternal(chip::app::ConcreteAttributePath(endpoint, cluster, attributeID)...`
- [ ] at `src/data-model-providers/codegen/tests/EmberReadWriteOverride.cpp:142` using `return emAfWriteAttributeExternal(path, input);`
- [ ] at `src/data-model-providers/codegen/tests/EmberReadWriteOverride.h:[27](https://github.com/jennygaz/connectedhomeip/actions/runs/13770601107/job/38507998003#step:22:28)` using ```/// specify what the next `emAfReadOrWriteAttribute` will contain```

PR #38006 (2025-05-26): Update lint workflow to exclude ember symbols
Fixes #36622 

Update the `lint.yml` file so that it also checks for symbols `emberAf` and `emAf`. This is done by adding a separate script, `check_symbols.sh`, which searches for uses of those symbols and has the ability to exclude specific directories with the `--skip-dir` flag.

This is the first step of many to solving #36622, but it requires refactoring the places marked by the script to make the script pass and avoid failing future PRs unnecessarily. It is also expected that the job created by this script will fail since we still need the ref [... cut]