project-chip/connectedhomeip #36652: [1.3] resumption of a deactivated secure thread session

varnholtm, 2024-11-27:
### Reproduction steps

A Thread accessory won't send messages on a deactivated secure session after it has been reactivated again. 

**Test-Setup:** 
Nordic nRF52840 Development Kit; 
Home-Controller(s): Apple HomePod Mini, AeotecHub (SmartThings), Homeassistant on Raspi with OTBR

Used Firmware: Matter Door-Lock Sample 

**Scenario:**
- the accessory (Development Kit) looses the thread network connection --> "DETACHED"
- the accessory tries to send a message (default 4 times) and sets the used secure session to --> "kDefunct" 
- the accessory reconnects to the thread network --> "CHILD"
- the Home-Controller sends a message to the accessory (e.g. "LockDoor")
- the accessory acknowledges this message and performs the operation but then **doesn't** send the updated cluster state via the reactivated secure session
- you have to wait, until the home-controller establishes a new secure session
- the accessory will automatically send its momentary cluster states on the new session

This behavior seems strange to me. Shouldn't the accessory send its cluster state messages after the old secure session has been reactivated?

Further more the time until the home-controller establishes a new secure session is a lot longer with the 1.3 version than it is with the 1.2, even though there shouldn't be a difference.
In the 1.2 door lock sample the home-controller sends a new CASE Session < 10s, after the secure session deactivates on the accessory. In the 1.3 sample this takes at least 30s most of the time more than 2 min.

The attached logs show the problem. 

[1_3_DK_ha_otbr.log](https://github.com/user-attachments/files/17932214/1_3_DK_ha_otbr.log)

[1_3_DK_apple.log](https://github.com/user-attachments/files/17932127/1_3_DK_apple.log)

Log with 1.2 Door Lock Sample to show fast new CASE session establishment
[1_2_apple.log](https://github.com/user-attachments/files/17932312/1_2_apple.log)

### Bug prevalence

always reproducible

### GitHub hash of the SDK that was being used

022f2c92f6fc04ddebeea443513d050a6bd1d222

### Platform

nrf connect

### Platform Version(s)

1.3

### Type

Platform Issue

bzbarsky-apple, 2024-12-05:
@varnholtm Which message did the device fail to send in the "tries to send a message (default 4 times)" step?  If this is a subscription report, won't this tear down the subscription at that point, in addition to marking the session defunct?

varnholtm, 2024-12-10:
@bzbarsky-apple You are right, the messages are subscription reports (IM: ReportData). But if the subscription will also be teared down when loosing the thread connection, why isn't there a mechanism that will get it working again in an acceptable timerange (within seconds)? 
It would be logically to me that if the device reactivates an old secure session, on which it had sent out reports to a subscriber, it should also resume sending the reports again to the same subscriber. I'm not sure if there is such an mechanism or feature that would help with this problem.

bzbarsky-apple, 2024-12-13:
> why isn't there a mechanism that will get it working again in an acceptable timerange (within seconds)?

Devices can enable an optional feature to attempt to re-establish the subscription from their end.  But if that fails a few times, at some point the subscription is torn down, because you can't keep trying this sort of thing forever and need to reclaim resources at some point.

varnholtm, 2024-12-13:
@bzbarsky-apple: I did check the "persistent subscription" option, which seems to improve the behavior a little bit, but as you said, after a certain amount of resumptions this wont't work anymore, until a new session was established.

One thing that is not clear to me is the different behavior in the version 1.2. To me it seems, that with this version the home controller is recognizing a lost accessory a lot faster, so that you always get an SIGMA Case messages right after the accessory has lost its thread network connection. 

Do you know of any difference from 1.2 to 1.3 regarding that?

Thanks!

bzbarsky-apple, 2024-12-18:
> Do you know of any difference from 1.2 to 1.3 regarding that?

Not off the top of my head, no.