project-chip/connectedhomeip #36661: Remove direct calls for emberAfAttributeChanged from libCHIP (reporting.cpp) and make these part of CodegenDataModel

andy31415, 2024-11-27:
https://github.com/project-chip/connectedhomeip/blob/master/src/app/reporting/reporting.cpp contains just a wrapper for attribute change callbacks.

However this is directly ember coupled and uses a global attribute change listener. For backwards compatibility we should maintain `MatterReportingAttributeChangeCallback` existing, however because it is ember-tied it should be part of the Codegen data model provider rather than direct libCHIP.

yufengwangca, 2024-12-05:
The initial decoupling work was done by https://github.com/project-chip/connectedhomeip/pull/36364/files

andy31415, 2025-01-30:
This is currently decoupled using a `Temporary_...` call that goes into DataModel::Provider.
Forcing DataModel::Provider to update versions is not ideal, however at least it is decoupled.