project-chip/connectedhomeip #36666: Standardize build logic for SDK code to GN even when using cmake-based builds

andy31415, 2024-11-28:
Currently libCHIP is built by GN:

https://github.com/project-chip/connectedhomeip/blob/master/config/common/cmake/chip_gn.cmake

However some things are still off:

- Ameba has separate logic: https://github.com/project-chip/connectedhomeip/blob/master/config/ameba/chip.cmake
- even though chip_data_model.gni exists, we also have https://github.com/project-chip/connectedhomeip/blob/master/src/app/chip_data_model.cmake
- ESP32 compiles cluster logic (or overall icd/util/server etc) as a completely separate list of 'compile everything': https://github.com/project-chip/connectedhomeip/blob/master/examples/all-clusters-app/esp32/main/CMakeLists.txt#L31

What we should have is a single path to compile things so we can track dependencies. CMake logic can exist to fix up includes, however I would expect any `static_library` or similar to be generate by GN only. We should start with chip_data_model and unifying ESP32.

andy31415, 2024-11-28:
Expected tasks:
  - [ ] Make Ameba not require special logic
  - [ ] Make ESP32 not require specia logic
  - [ ] Make the cmake-gn build logic capable to include any targets (specifically including data models)
  - [ ] Update examples that use cmake to use gn-builds for both libCHIP and data models (this way all dependencies will be pulled from GN rather than custom cmake logic)

khodya, 2025-02-07:
find common things how modules build use cmake

luis4a0, 2025-02-20:
I am taking this ticket. I'll investigate what can be done and separate in subtickets if needed.

luis4a0, 2025-02-20:
Hi @andy31415, I am investigating first the issue with ameba. Could you please point me to a folder which is built using the logic you expect. I see various folders with different building blocks.

andy31415, 2025-02-20:
I just grepped for cmake, so not fully familiar with the build systems.

However to build you can check the CI:  https://github.com/project-chip/connectedhomeip/blob/master/.github/workflows/examples-ameba.yaml#L54

It seems to use build_examples. you can do `--dry-run` to see what console commands it executes. It would be https://github.com/project-chip/connectedhomeip/blob/master/scripts/build/builders/ameba.py#L75

The scary bit is that it uses another script to trigger the build :/