project-chip/connectedhomeip #36672: [BUG] referential integrity for groupIds

jonsmirl, 2024-11-28:
### Reproduction steps

I have not tested this extensively, but I appear to be able to delete GroupKeyMap entries while GroupInfo entries still reference the map entries. What kind of errors should this generate, it doesn't cause any currently.  Should you be required to have a GroupKeyMap entry before a GroupInfo is constructed?

If this is not an error, you can get into a run-time error situation when you try to use a group which is missing an associated groupKeyMap entry and there is no UI for communicating the error to the user.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

17b1a38e909e7874593bcb87c31be03a5866f1d4

### Platform

core

jonsmirl, 2024-12-05:
The GroupKeyMap on the GKM cluster can also get out of sync.

bzbarsky-apple, 2024-12-05:
> The GroupKeyMap on the GKM cluster can also get out of sync.

Right, that's "removal of group keys for an already-existing group", no?

jonsmirl, 2024-12-05:
There are two places -- the Group cluster and the GroupKeyMap on the GKM.

bzbarsky-apple, 2024-12-05:
Yes.... Just to be clear:

1. You cannot add a group via Group cluster that does not have a corresponding key defined in the GKM.
2. You _can_ remove a GKM entry for a group that has been added via Group cluster, which will then end up key-less.

jonsmirl, 2024-12-05:
Why is the key-less group state allowed? And has that state been tested?

turon, 2024-12-06:
We need to both specify and implement these interdependencies explicitly.

bzbarsky-apple, 2024-12-13:
> Why is the key-less group state allowed? 

My guess?  Oversight.

> And has that state been tested?

Doubtful.