project-chip/connectedhomeip #36673: [CERT-TEST-FAILURE] [TC-RR-1.1] When CHIP_CONFIG_MAX_GROUPS_PER_FABRIC sets to 24, step 14 Setting group key map timeout

KaMeHaMeHaaaa, 2024-11-29:
### Feature Area

Other

### Test Case

TC-RR-1.1

### Reproduction steps

1. Set CHIP_CONFIG_MAX_GROUPS_PER_FABRIC to 24 (4 * 6)
2. docker run -v /home/ubuntu/[certs:/paa_roots](http://certs/paa_roots) -v /var/run/dbus/[system_bus_socket:/var/run/dbus/system_bus_socket](http://system_bus_socket/var/run/dbus/system_bus_socket) -v /home/ubuntu/certification-tool/backend/test_collections/matter/sdk_tests/sdk_checkout/[python_testing:/root/python_testing/](http://python_testing/root/python_testing/) -v /launch_dir --privileged --network host -it connectedhomeip/chip-cert-bins:50bcad170e6def6d56f2f5b34522a82aea9f3190
3. rm -f admin_storage.json && python3 python_testing/scripts/sdk/TC_RR_1_1.py --discriminator 3550 --passcode 32590277 --commissioning-method ble-wifi --paa-trust-store-path /paa_roots --storage-path admin_storage.json --wifi-ssid Matter --wifi-passphrase 12345678 --timeout 600 --ble-interface-id 0

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

13ab158f107c06ad9d4a2591aba438d212500b2d

### Platform

ameba

### Anything else?

Same issue mentioned in this link:  [https://github.com/project-chip/connectedhomeip/issues/33539](url)

KaMeHaMeHaaaa, 2024-11-29:
==== test log of device =====
chip[EM] >>> [E:52356r S:21575 M:72010717] (S) Msg RX from 1:000000000001B669 [6C51] --- Type 0001:06 (IM:WriteRequest)
[2024-11-28 19:32:24]  
chip[EM] Handling via exchange: 52356r, Delegate: 0x1002948c
[2024-11-28 19:32:24]  
chip[IM] Received Write request
[2024-11-28 19:32:24]  
chip[DMG] IM WH moving to [Initialized]
[2024-11-28 19:32:24]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679c1
[2024-11-28 19:32:24]  
chip[DMG] IM WH moving to [AddStatus]
[2024-11-28 19:32:24]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679c2
[2024-11-28 19:32:25]  
chip[DMG] IM WH moving to [AddStatus]
[2024-11-28 19:32:25]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679c3
[2024-11-28 19:32:25]  
chip[DMG] IM WH moving to [AddStatus]
[2024-11-28 19:32:25]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679c4
[2024-11-28 19:32:26]  
chip[DMG] IM WH moving to [AddStatus]
[2024-11-28 19:32:26]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679c5
[2024-11-28 19:32:26]  
chip[DMG] IM WH moving to [AddStatus]
[2024-11-28 19:32:26]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679c6
[2024-11-28 19:32:27]  
chip[DMG] IM WH moving to [AddStatus]
[2024-11-28 19:32:27]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679c7
[2024-11-28 19:32:27]  
chip[DMG] IM WH moving to [AddStatus]
[2024-11-28 19:32:27]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679c8
[2024-11-28 19:32:27]  
chip[DMG] IM WH moving to [AddStatus]
[2024-11-28 19:32:27]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679c9
[2024-11-28 19:32:28]  
chip[DMG] IM WH moving to [AddStatus]
[2024-11-28 19:32:28]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679ca
[2024-11-28 19:32:28]  
chip[DMG] IM WH moving to [AddStatus]
[2024-11-28 19:32:28]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679cb
[2024-11-28 19:32:29]  
chip[DMG] IM WH moving to [AddStatus]
[2024-11-28 19:32:29]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679cc
[2024-11-28 19:32:29]  
chip[DMG] IM WH moving to [AddStatus]
[2024-11-28 19:32:29]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679cd
[2024-11-28 19:32:30]  
chip[DMG] IM WH moving to [AddStatus]
[2024-11-28 19:32:30]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679ce
[2024-11-28 19:32:30]  
chip[DMG] IM WH moving to [AddStatus]
[2024-11-28 19:32:30]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679cf
[2024-11-28 19:32:30]  
chip[DMG] IM WH moving to [AddStatus]
[2024-11-28 19:32:30]  
chip[DMG] Endpoint 0, Cluster 0x0000_003F update version to 1db679d0
[... 71 lines cut ...]
chip[EM] Found matching exchange: 52356r, Delegate: 0x0
[2024-11-28 19:32:35]  
chip[EM] Forcing tx of solitary ack for duplicate MessageCounter:72010717 on exchange 52356r
[2024-11-28 19:32:35]  
chip[EM] <<< [E:52356r S:21575 M:257801649 (Ack:72010717)] (S) Msg TX to 1:000000000001B669 [6C51] [UDP:192.168.137.51:35847] --- Type 0000:10 (SecureChannel:StandaloneAck)
[2024-11-28 19:32:35]  
chip[IN] Received a duplicate message with MessageCounter:72010717 on exchange 52356r
[2024-11-28 19:32:35]  
chip[EM] >>> [E:52356r S:21575 M:72010717] (S) Msg RX from 1:000000000001B669 [6C51] --- Type 0001:06 (IM:WriteRequest)
[2024-11-28 19:32:35]  
chip[EM] Found matching exchange: 52356r, Delegate: 0x0
[2024-11-28 19:32:35]  
chip[EM] Forcing tx of solitary ack for duplicate MessageCounter:72010717 on exchange 52356r
[2024-11-28 19:32:35]  
chip[EM] <<< [E:52356r S:21575 M:257801650 (Ack:72010717)] (S) Msg TX to 1:000000000001B669 [6C51] [UDP:192.168.137.51:35847] --- Type 0000:10 (SecureChannel:StandaloneAck)
[2024-11-28 19:32:35]  
chip[EM] Retransmitting MessageCounter:**257801646** on exchange 52356r Send Cnt 1
[2024-11-28 19:32:36]  
chip[EM] Retransmitting MessageCounter:**257801646** on exchange 52356r Send Cnt 2
[2024-11-28 19:32:36]  
chip[EM] Retransmitting MessageCounter:**257801646** on exchange 52356r Send Cnt 3
[2024-11-28 19:32:36]  
chip[EM] Retransmitting MessageCounter:**257801646** on exchange 52356r Send Cnt 4
[2024-11-28 19:32:37]  
chip[EM] Failed to Send CHIP MessageCounter:**257801646** on exchange 52356r sendCount: 4 max retries: 4

KaMeHaMeHaaaa, 2024-11-29:
==== test log of TH ====
2024-11-28 19:32:24]  [MatterTest] 11-28 11:32:23.776 INFO <<< [E:52356i S:35599 M:**72010717**] (S) Msg TX to 1:0000000012344321 [6C51] [UDP:192.168.137.56%eth0:5540] --- Type 0001:06 (IM:WriteRequest)
[2024-11-28 19:32:24]  [MatterTest] 11-28 11:32:24.129 INFO Retransmitting MessageCounter:**72010717** on exchange 52356i Send Cnt 1
[2024-11-28 19:32:25]  [MatterTest] 11-28 11:32:24.538 INFO Retransmitting MessageCounter:**72010717** on exchange 52356i Send Cnt 2
[2024-11-28 19:32:25]  [MatterTest] 11-28 11:32:25.152 INFO Retransmitting MessageCounter:**72010717** on exchange 52356i Send Cnt 3
[2024-11-28 19:32:26]  [MatterTest] 11-28 11:32:26.101 INFO Retransmitting MessageCounter:**72010717** on exchange 52356i Send Cnt 4
[2024-11-28 19:32:27]  [MatterTest] 11-28 11:32:27.676 ERROR Failed to Send CHIP MessageCounter:**72010717** on exchange 52356i sendCount: 4 max retries: 4
[2024-11-28 19:32:28]  [MatterTest] 11-28 11:32:30.003 INFO SecureSession[0xffff7406a8f0, LSID:35599]: State change 'kActive' --> 'kDefunct'
[2024-11-28 19:32:30]  [MatterTest] 11-28 11:32:30.004 ERROR Time out! failed to receive write response from Exchange: 52356i
[2024-11-28 19:32:30]  [MatterTest] 11-28 11:32:30.009 ERROR Exception occurred in test_TC_RR_1_1.
[2024-11-28 19:32:30]  Traceback (most recent call last):
[2024-11-28 19:32:30]    File "/usr/local/lib/python3.10/dist-packages/mobly/base_test.py", line 783, in exec_one_test
[2024-11-28 19:32:30]      test_method()
[2024-11-28 19:32:30]    File "/root/python_testing/scripts/sdk/matter_testing_support.py", line 1547, in async_runner
[2024-11-28 19:32:30]      return asyncio.run(runner_with_timeout)
[2024-11-28 19:32:30]    File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
[2024-11-28 19:32:30]      return loop.run_until_complete(main)
[2024-11-28 19:32:30]    File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
[2024-11-28 19:32:30]      return future.result()
[2024-11-28 19:32:30]    File "/usr/lib/python3.10/asyncio/tasks.py", line 445, in wait_for
[2024-11-28 19:32:30]      return fut.result()
[2024-11-28 19:32:30]    File "/root/python_testing/scripts/sdk/TC_RR_1_1.py", line 471, in test_TC_RR_1_1
[2024-11-28 19:32:30]      await self.fill_and_validate_group_key_map(
[2024-11-28 19:32:30]    File "/root/python_testing/scripts/sdk/TC_RR_1_1.py", line 580, in fill_and_validate_group_key_map
[2024-11-28 19:32:30]      await client.WriteAttribute(
[2024-11-28 19:32:30]    File "/usr/local/lib/python3.10/dist-packages/chip/ChipDeviceCtrl.py", line 1047, in WriteAttribute
[2024-11-28 19:32:30]      return await future
[2024-11-28 19:32:30]  chip.exceptions.ChipStackError: src/app/WriteClient.cpp:504: CHIP Error 0x00000032: Timeout
[2024-11-28 19:32:30]  [MatterTest] 11-28 11:32:30.022 INFO ***** Test Failure : 
[2024-11-28 19:32:30]  [MatterTest] 11-28 11:32:30.022 INFO Finished test in 150018ms
[2024-11-28 19:32:30]  [MatterTest] 11-28 11:32:30.023 INFO [Test] test_TC_RR_1_1 ERROR

jonsmirl, 2024-11-29:
Are you part of Zemismart? I am trying to get these Zemismart devices working and they have problems.

KaMeHaMeHaaaa, 2024-11-29:
> Are you part of Zemismart? I am trying to get these Zemismart devices working and they have problems.

Sorry, I think you might have mistaken me for someone else. I’m not with that company

cecille, 2024-11-29:
This isn't just a test problem - any client that attempts to do this will also time out. If the device takes 11 seconds to respond to a client, the client will definitely think something has gone wrong.

tcarmelveilleux, 2024-12-03:
Here are the full logs from a run.
[TC_RR_1_1_DUT_LOG.log.txt](https://github.com/user-attachments/files/17994349/TC_RR_1_1_DUT_LOG.log.txt)
[TC_RR_1_1_TH_LOG.log.txt](https://github.com/user-attachments/files/17994351/TC_RR_1_1_TH_LOG.log.txt)

cecille, 2024-12-03:
> > This isn't just a test problem - any client that attempts to do this will also time out. If the device takes 11 seconds to respond to a client, the client will definitely think something has gone wrong.
> 
> Regarding this issue, could you provide any advice? Does the device absolutely need to resolve the slow processing speed of setting the group key map to meet the requirements of the TC-RR-1.1 test case? Or, if possible, could there be any workarounds to provide the device with more processing time before replying with a response? For example, increasing the timeout duration, adjusting the retransmission count or interval, or any other feasible approach. Any insights would be greatly appreciated. Thank you!

There is no workaround for this in the test because this is a case of the test catching a real problem with a device. This is actually good - it's catching a problem that would cause flake in the field. 

I don't know what platform you are using, but if this is a problem with key storage it probably means that the storage write isn't optimized properly. Flash, for example, can be quite long to write if you don't consolidate the writes properly. 

I would suggest contacting your chip vendor to see if they have an updated solution for fast writes of small data. OR you can try deferring and consolidating the writes out to storage.

KaMeHaMeHaaaa, 2024-12-04:
Thanks a lot for your suggestion. I will follow up on optimizing the storage write speed.