project-chip/connectedhomeip #36688: [TC_ACE_2.1]

FIREWORKMBY, 2024-12-02:
### Test issue(s)

Test issue(s)
TH Version: v2.11+fall2024
TH SHA: 8f86ccd

First, use the web interface of the test harness to test it. The error screen and log are as follows:
[TC-ACE-2-1_2024_11_28_08_16_43.log](https://github.com/user-attachments/files/17946878/TC-ACE-2-1_2024_11_28_08_16_43.log)
image

After that, the Raspberry PI equipped with TH was used to enter docker, and the pyrhon script was used for testing. The error log is as follows:
[
[【TC-ACE-2.1】UI-Python test python error.txt](https://github.com/user-attachments/files/17976579/TC-ACE-2.1.UI-Python.test.python.error.txt)
](url)

Test command:
python3 /root/python_testing/scripts/sdk/TC_AccessChecker.py --commissioning-method on-network --discriminator 3840 --passcode 20202021 --endpoint 0 --timeout 1000 --paa-trust-store-path /paa-root-certs --storage-path admin_storage.json --tests test_TC_ACE_2_1 --timeout 1000

Clearly visible error is:
[MatterTest] 11-29 06:40:21.415 INFO
Problem: ProblemSeverity.ERROR
    test_name: Read access Checker - 1
    location:
       Endpoint: 1,
       Cluster:  1107 (0x453) ThreadNetworkDirectory
      Attribute:0 (0x00)
    problem: Error reading <class 'chip.clusters.Objects.ThreadNetworkDirectory'>:<class 'chip.clusters.Objects.ThreadNetworkDirectory.Attributes.PreferredExtendedPanID'> = ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.UnsupportedAccess: 126>))
    spec_location:

[MatterTest] 11-29 06:40:21.416 INFO
Problem: ProblemSeverity.ERROR
    test_name: Read access Checker - 1
    location:
       Endpoint: 1,
       Cluster:  1107 (0x453) ThreadNetworkDirectory
      Attribute:1 (0x01)
    problem: Error reading <class 'chip.clusters.Objects.ThreadNetworkDirectory'>:<class 'chip.clusters.Objects.ThreadNetworkDirectory.Attributes.ThreadNetworks'> = ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.UnsupportedAccess: 126>))
    spec_location:

[MatterTest] 11-29 06:40:21.416 INFO
Problem: ProblemSeverity.ERROR
    test_name: Read access Checker - 2
    location:
       Endpoint: 1,
       Cluster:  1107 (0x453) ThreadNetworkDirectory
      Attribute:0 (0x00)
    problem: Error reading <class 'chip.clusters.Objects.ThreadNetworkDirectory'>:<class 'chip.clusters.Objects.ThreadNetworkDirectory.Attributes.PreferredExtendedPanID'> = ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.UnsupportedAccess: 126>))
    spec_location:

[MatterTest] 11-29 06:40:21.417 INFO
Problem: ProblemSeverity.ERROR
    test_name: Read access Checker - 2
    location:
       Endpoint: 1,
       Cluster:  1107 (0x453) ThreadNetworkDirectory
      Attribute:1 (0x01)
    problem: Error reading <class 'chip.clusters.Objects.ThreadNetworkDirectory'>:<class 'chip.clusters.Objects.ThreadNetworkDirectory.Attributes.ThreadNetworks'> = ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.UnsupportedAccess: 126>))
    spec_location:

[MatterTest] 11-29 06:40:21.417 INFO
Problem: ProblemSeverity.ERROR
    test_name: Read access Checker - 3
    location:
       Endpoint: 1,
       Cluster:  1107 (0x453) ThreadNetworkDirectory
      Attribute:0 (0x00)
    problem: Error reading <class 'chip.clusters.Objects.ThreadNetworkDirectory'>:<class 'chip.clusters.Objects.ThreadNetworkDirectory.Attributes.PreferredExtendedPanID'> = ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.UnsupportedAccess: 126>))
    spec_location:

FIREWORKMBY, 2024-12-10:
@bzbarsky-apple Can you help me to resolve this problem?

bzbarsky-apple, 2024-12-18:
@FIREWORKMBY sounds like your ThreadNetworkDirectory cluster implementation is returning errors?  Is that expected?