project-chip/connectedhomeip #36712: [BUG] Segmentation Fault in WebSocket Server Message Handling

Chapoly1305, 2024-12-04:
### Reproduction steps

## Issue
Segmentation fault occurs when attempting to write messages to a WebSocket connection that has been closed or destroyed.

## Root Cause Analysis
The crash occurs in `lws_callback_on_writable()` when `gWebSocketInstance` is NULL, which happens when trying to write messages after a client disconnection.

### Stack Trace

```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00005f46f48fc5d6 in lws_callback_on_writable (wsi=0x0)
```

### Problematic Code Path

1. In `WebSocketServer.cpp`, messages are queued using the `Send()` method:
```
void WebSocketServer::Send(const char * msg)
{
    std::lock_guard<std::mutex> lock(gMutex);
    gMessageQueue.push_back(msg);
}
```

2. The main server loop attempts to process queued messages without checking connection state:
```
while (mRunning)
{
    lws_service(context, -1);

    std::lock_guard<std::mutex> lock(gMutex);
    if (gMessageQueue.size())
    {
        lws_callback_on_writable(gWebSocketInstance);  // Crashes here if gWebSocketInstance is null
    }
}
```

3. The WebSocket instance pointer is managed in callback handlers:

```
else if (LWS_CALLBACK_ESTABLISHED == reason)
{
    gWebSocketInstance = wsi;
}
else if (LWS_CALLBACK_WSI_DESTROY == reason)
{
    gWebSocketInstance = nullptr;
}
```

### Impact

Application crashes when attempting to send messages after client disconnection
Messages in queue become stale and may cause crashes when new connections are established

### Recommended Fix
Add connection state validation before attempting to write:
```
while (mRunning)
{
    lws_service(context, -1);

    std::lock_guard<std::mutex> lock(gMutex);
    if (gMessageQueue.size() && gWebSocketInstance != nullptr)
    {
        lws_callback_on_writable(gWebSocketInstance);
    }
}
```
Additionally, clean up the message queue on connection termination:
```
else if (LWS_CALLBACK_WSI_DESTROY == reason)
{
    std::lock_guard<std::mutex> lock(gMutex);
    gWebSocketInstance = nullptr;
    gMessageQueue.clear();  // Prevent stale messages from causing issues
}
```

## Test Cases

- Normal message sending with connected client
- Client disconnection while messages are in queue
- Message queuing during no active connection
- Reconnection handling with existing message queue

## Related Files

- WebSocketServer.cpp
- InteractiveCommands.cpp (caller of WebSocket server)
- [core.chip-tool.zip](https://github.com/user-attachments/files/18003897/core.chip-tool.zip)
- [Crash_WebSocketServer_lws_callback_on_writable.zip](https://github.com/user-attachments/files/18003908/Crash_WebSocketServer_lws_callback_on_writable.zip)

## Contributors
@Chapoly1305 @agatah2333

### Bug prevalence

Always, stable replication during our testing. Test script attached.

### GitHub hash of the SDK that was being used

2b382d6357e9bfbb54f930d17156e79ed07091f3

### Platform

other

### Platform Version(s)

Tested on 1.4 and earlier branch. Git blame indicate the code has been there for a long while.

bzbarsky-apple, 2024-12-05:
@vivien-apple

vivien-apple, 2025-06-05:
Thanks a lot for the detailed issue report — and especially for the test case! It made reproducing and verifying the fix so much easier ❤️

PR #39406 (2025-06-05): [Examples] Fix crash when sending to invalidated WebSocket connection
#### Description

Fixes a crash that occurs when attempting to send messages to a WebSocket client that has already disconnected...

Fixes #36712

#### Testing

Used the test case provided (Thanks!) in #36712:
 - Crash is reproducible without the patch
 - No crash occurs after applying the patch
closed 2025-06-05 by mergify[bot]