project-chip/connectedhomeip #36740: Remove CodegenDataModelInstance and make applications provide theyr codegen data model (likely via a `static` for example)

andy31415, 2024-12-05:
See https://github.com/project-chip/connectedhomeip/pull/36658#discussion_r1872111079

The usage of a mutable persistence provider that is used at startup is awkward. It seems better to have the persistence provider established in the constructor.

Expected changes:
  - [ ] Make the persistence provider to use part of the CodegenDataModel constructor (may be null, which means unchanged)
  - [ ] Remove CodegenDataModelProviderInstance (header and cpp files and all linkages), let applications build their own codegen data model provider instead and allocate space for them.

andy31415, 2024-12-05:
Note that this means a LOT of file changes, however making applications be responsible of what data model provider they use seems to be the best dependency use (this way it is clear that applications use zap/ember for example, so they provide that one as their underlying model)

bzbarsky-apple, 2024-12-09:
This is also the sanest way to have applications have a provider that uses ZAP codegen for some endpoints and dynamic things for others.

andy31415, 2025-02-04:
This has #36230 as a prerequisite or at least very high overlapping. we seem to want to make applications to setup their data model provider explicitly. I believe this was done in #36558. After that, we want to make the construction and dependencies better (since application has the persistence provider, application can allocate a codegendatamodelprovider (likely static variable) and initialize it with the right parameters.

andy31415, 2025-03-26:
TODO as first step: determine if this is still an issue or not , as we updated our server startup to take the datamodel provider as an argument and we have no hardcoded defaults in the interactionmodelengine code.