project-chip/connectedhomeip #36746: [BUG] Operational discovery fails if highest scored IP address is not reachable

nozols, 2024-12-06:
### Reproduction steps

There is a difference in behaviour between commissionable node and operational node discovery.

- Commisionable node discovery will try to send messages over all/multiple discovered addresses if the highest scored address fails
- Operational node discovery will only attempt the highest scored address in some cases, which I believe is a bug.

Commit https://github.com/project-chip/connectedhomeip/commit/b590431f30f2abbff84ad94211c11b049e2bd702 created and implemented the `NodeLookupResults::UpdateResults` function. The problem is that the `results` list can get filled with addresses which are all the same.

This can cause problems. 

- During commissioning the PASE session will be created over an IP address that doesn’t have the highest score. However, when operational discovery takes place to finalise commissioning, it wil fail because it only tries the IP with the highest score again.
- If for whatever reason the highest scored IP doesn’t work, operational discovery will never succeed.

I think the `NodeLookupResults::UpdateResults` function should be updated to also check if the address is already present in the `results` list, and not replace another address if this is the case (even if the score of address in the list is lower).

**Reproduction path:**

- To make it a bit easier to see what is happening you could at `LogDetail()` to `NodeLookupResults::ConsumeResults()` and `NodeLookupResults::UpdateResults()`
- Add `#define CHIP_CONFIG_MDNS_RESOLVE_LOOKUP_RESULTS 5` to `examples/chip-tool/include/CHIPProjectAppConfig.h`
    - Without this, operational discovery doesn’t even attempt multiple addresses, while commisionable node discovery does do this.
- Compile `chip-tool`
- Now starts the tricky path, we need to create a device that creates an mDNS advertisement, but doesn’t respond to udp messages. I’ve used `matter.js` for this, as it exposed the APIs to do this. I’ve attached these scripts.
- Commissionable node (which works as expected):
    - Run as javascript (with matter.js installed):

```javascript
import { Network, VendorId } from '@matter/main';
import { MdnsBroadcaster } from '@matter/main/protocol';

const broadcaster = await MdnsBroadcaster.create(Network.get());

// Pairing code = MT:Y.K90YDA13CJO35X700
await broadcaster.setCommissionMode(5540, 1, {
    discriminator: 1203,
    vendorId: VendorId(65521),
    name: 'Dummy MDNS Commissionable Node',
    deviceType: 0x0001,
    productId: 32768,
});
```
  - Run `chip-tool pairing code 12345 MT:Y.K90YDA13CJO35X700`
      - This obviously fails, because no one is responding. Important is that we see that all advertised addresses are used:
          - `Attempting PASE connection to UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540`
          - `Attempting PASE connection to UDP:[fd01:54a2:7aa9:9caf:21c:42ff:fe11:b7fc]:5540`
    - [commissionable-node.log](https://github.com/user-attachments/files/18038338/commissionable-node.log)

- Operational node (which can fail):
    - Run as javascript:
```javascript
import { FabricId, FabricIndex, Network, NodeId, Crypto, VendorId } from '@matter/main';
import { Fabric, MdnsBroadcaster } from '@matter/main/protocol';

const broadcaster = await MdnsBroadcaster.create(Network.get());

const compressedFabricId = 0xB4341FFC3A38E11Bn; // Note: replace with the compressed fabric id thats created by chip-tool
const nodeId = NodeId(12345);

const operationalId = Buffer.alloc(8);
operationalId.writeBigUint64BE(compressedFabricId);

await broadcaster.setFabrics(5540, [
    new Fabric({
        fabricIndex: FabricIndex(1),
        fabricId: FabricId(compressedFabricId),
        nodeId,
        rootNodeId: NodeId(1),
        operationalId,
        rootPublicKey: new Uint8Array(),
        keyPair: Crypto.createKeyPair(),
        rootVendorId: VendorId(0x1234),
        rootCert: new Uint8Array(),
        identityProtectionKey: new Uint8Array(),
        operationalIdentityProtectionKey: new Uint8Array(),
        intermediateCACert: undefined,
        operationalCert: new Uint8Array(),
        label: 'testing',
    }),
]);
```
  - Run `chip-tool interactive start` and then `onoff toggle 12345 1`
      - Interactive mode is required, because `chip-tool` will exit after 30s
      - With the added logs, we see that the results table is getting filled with duplicate addresses.
```
[-] NodeLookupResults::UpdateResults ->
[DIS]  #1: UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540
[DIS]  #2: UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540
[DIS]  #3: UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540
[DIS]  #4: UDP:[fd01:54a2:7aa9:9caf:21c:42ff:fe11:b7fc]:5540
[DIS]  #5: UDP:[fd01:54a2:7aa9:9caf:21c:42ff:fe11:b7fc]:5540
```
  - Because (in this case) the highest three addresses are the same, it will try three times to send the messages to this address. While the intended behaviour (I think) is that it should try a different address.
  - [operational-node.log](https://github.com/user-attachments/files/18038333/operational-node.log)

        
        
> That the link-local/highest score address is not reachable should not happen of course. But we don't live in a perfect world 😅 

> I was not able to reproduce this when running `chip-tool` on Mac, only on Ubuntu. So there also seems to be some discovery differences between platforms.

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

9b008bc10d48bef1184cf9f0405b12b41af5d328

### Platform

core

bzbarsky-apple, 2024-12-09:
@nozols Which DNS-SD implementation is the client using in this case?  It's quite odd that it's ending up with multiple entries for the same IP _on the same interface_ in NodeLookupResults::UpdateResults.

Is this chip-tool running on Linux or Mac?  If on Linux, is it using avahi or minmdns?

bzbarsky-apple, 2024-12-09:
But in either case, I would expect the `TryNextResult` in `OperationalSessionSetup::OnSessionEstablishmentError` to in fact walk through all available results...  There is no limit of "three" there.

nozols, 2024-12-10:
Hi @bzbarsky-apple, thanks for getting back to me. This test was done using chip-tool on linux, using the minimal mdns implementation.

> But in either case, I would expect the TryNextResult in OperationalSessionSetup::OnSessionEstablishmentError to in fact walk through all available results... There is no limit of "three" there.

Yes that's true, but I would expect to immediately try the next (non-link local) address. Instead of retrying the link local address again. I've also seen that the entire list gets filled with the same address (which would also have been the case with `CHIP_CONFIG_MDNS_RESOLVE_LOOKUP_RESULTS` set to 3 here).

I tested it with `chip_mdns = "platform"`, I only get a single result (seems to be random wether or not this is the IPv4 address or IPv6 link local address).

```
[1733820408.493] [51837:51839] [DIS] Node ID resolved for 41754DA88DED80AC-0000000000003039
[1733820408.493] [51837:51839] [DIS]    Hostname: 001C4211B7FC0000
[1733820408.493] [51837:51839] [DIS]    IP Address #1: 192.168.0.70
[1733820408.493] [51837:51839] [DIS]    Port: 5540
[1733820408.493] [51837:51839] [DIS]    Mrp Interval idle: 500 ms
[1733820408.493] [51837:51839] [DIS]    Mrp Interval active: 300 ms
[1733820408.493] [51837:51839] [DIS]    Mrp Active Threshold: 4000 ms
[1733820408.493] [51837:51839] [DIS]    TCP Client Supported: 0
[1733820408.493] [51837:51839] [DIS]    TCP Server Supported: 0
[1733820408.493] [51837:51839] [DIS]    ICD: not present
[1733820408.493] [51837:51839] [DIS] Lookup clearing interface for non LL address
[1733820408.493] [51837:51839] [-] NodeLookupResults::UpdateResults ->
[1733820408.493] [51837:51839] [DIS]    #1: UDP:192.168.0.70:5540
[1733820408.493] [51837:51839] [DIS] UDP:192.168.0.70%enp0s5:5540: new best score: 2 (for 41754DA88DED80AC-0000000000003039)
[1733820408.493] [51837:51839] [DIS] Checking node lookup status for 41754DA88DED80AC-0000000000003039 after 2 ms
[1733820408.493] [51837:51839] [DIS] Keeping DNSSD lookup active
[1733820408.694] [51837:51839] [DIS] Checking node lookup status for 41754DA88DED80AC-0000000000003039 after 203 ms
[1733820408.694] [51837:51839] [-] Consuming: 0
[1733820408.694] [51837:51839] [DIS]    #1: UDP:192.168.0.70:5540
```

With `chip_mdns = "minimal", I instantly get more results, but the table is getting filled up with the link local address:
```
[1733820454.889] [52728:52730] [DIS] Checking node lookup status for 41754DA88DED80AC-0000000000003039 after 3 ms
[1733820454.889] [52728:52730] [DIS] Keeping DNSSD lookup active
[1733820454.889] [52728:52730] [DIS] UDP:[fd59:dfdf:9f75:3fa2:21c:42ff:fe11:b7fc%enp0s5]:5540: score has not improved: 5 (for 41754DA88DED80AC-0000000000003039)
[1733820454.889] [52728:52730] [DIS] UDP:[fdcc:3d97:9273:cfa1:21c:42ff:fe11:b7fc%enp0s5]:5540: score has not improved: 5 (for 41754DA88DED80AC-0000000000003039)
[1733820454.889] [52728:52730] [DIS] UDP:[fd01:54a2:7aa9:9caf:21c:42ff:fe11:b7fc%enp0s5]:5540: score has not improved: 5 (for 41754DA88DED80AC-0000000000003039)
[1733820454.889] [52728:52730] [-] NodeLookupResults::UpdateResults ->
[1733820454.889] [52728:52730] [DIS]    #1: UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540
[1733820454.889] [52728:52730] [DIS]    #2: UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540
[1733820454.889] [52728:52730] [DIS]    #3: UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540
[1733820454.889] [52728:52730] [DIS]    #4: UDP:[fd59:dfdf:9f75:3fa2:21c:42ff:fe11:b7fc]:5540
[1733820454.889] [52728:52730] [DIS]    #5: UDP:[fdcc:3d97:9273:cfa1:21c:42ff:fe11:b7fc]:5540
[1733820454.889] [52728:52730] [DIS] UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540: new best score: 7 (for 41754DA88DED80AC-0000000000003039)
[1733820454.889] [52728:52730] [DIS] Checking node lookup status for 41754DA88DED80AC-0000000000003039 after 3 ms
[1733820454.889] [52728:52730] [DIS] Keeping DNSSD lookup active
[1733820454.889] [52728:52730] [DIS] UDP:[fd59:dfdf:9f75:3fa2:21c:42ff:fe11:b7fc%enp0s5]:5540: score has not improved: 5 (for 41754DA88DED80AC-0000000000003039)
[1733820454.890] [52728:52730] [DIS] UDP:[fdcc:3d97:9273:cfa1:21c:42ff:fe11:b7fc%enp0s5]:5540: score has not improved: 5 (for 41754DA88DED80AC-0000000000003039)
[1733820454.890] [52728:52730] [DIS] UDP:[fd01:54a2:7aa9:9caf:21c:42ff:fe11:b7fc%enp0s5]:5540: score has not improved: 5 (for 41754DA88DED80AC-0000000000003039)
[1733820454.890] [52728:52730] [-] NodeLookupResults::UpdateResults ->
[1733820454.890] [52728:52730] [DIS]    #1: UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540
[1733820454.890] [52728:52730] [DIS]    #2: UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540
[1733820454.890] [52728:52730] [DIS]    #3: UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540
[1733820454.890] [52728:52730] [DIS]    #4: UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540
[1733820454.890] [52728:52730] [DIS]    #5: UDP:[fd59:dfdf:9f75:3fa2:21c:42ff:fe11:b7fc]:5540
[1733820454.890] [52728:52730] [DIS] UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540: new best score: 7 (for 41754DA88DED80AC-0000000000003039)
[1733820454.890] [52728:52730] [DIS] Checking node lookup status for 41754DA88DED80AC-0000000000003039 after 4 ms
[1733820454.890] [52728:52730] [DIS] Keeping DNSSD lookup active
[1733820454.890] [52728:52730] [DIS] UDP:[fd59:dfdf:9f75:3fa2:21c:42ff:fe11:b7fc%enp0s5]:5540: score has not improved: 5 (for 41754DA88DED80AC-0000000000003039)
[1733820454.890] [52728:52730] [DIS] UDP:[fdcc:3d97:9273:cfa1:21c:42ff:fe11:b7fc%enp0s5]:5540: score has not improved: 5 (for 41754DA88DED80AC-0000000000003039)
[... cut ...]
[1733820454.890] [52728:52730] [DIS]    #3: UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540
[1733820454.890] [52728:52730] [DIS]    #4: UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540
[1733820454.890] [52728:52730] [DIS]    #5: UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540
[1733820454.890] [52728:52730] [DIS] UDP:[fe80::21c:42ff:fe11:b7fc%enp0s5]:5540: new best score: 7 (for 41754DA88DED80AC-0000000000003039)
[1733820454.890] [52728:52730] [DIS] Checking node lookup status for 41754DA88DED80AC-0000000000003039 after 4 ms
[1733820454.890] [52728:52730] [DIS] Keeping DNSSD lookup active
```

bzbarsky-apple, 2024-12-12:
Well, those sound like bugs in those DNS-SD implementations, then.  There's not much operational discovery can do if the underlying layer is providing it garbage data...

andy31415, 2024-12-14:
I hope to get a chance to take a look soon. It would help a lot if I can readily reproduce it (i.e. not trying to figure out myself how to do that) - @nozols do you have a dns responder setup or similar that I could use? I wonder what makes your addresses repeated - multiple nics? something else?

I can also make some blind change with unit tests and hope for the best, but would really appreciate knowing I did the right thing? Would you at least be able to test if I have some "proposed fix" PR?

PR #36861 (2024-12-17): Make AddressResolver not keep duplicate IP addresses in its cache
This fixes #36746 by deduplicating IP addresses in AddressResolver results.

### Changes

- do not insert duplicate addresses in the list
- fixed and added unit tests for this
- Attempted to push MDNS result size to 3 however that seems to take too much RAM for various platforms, so new tests are enabled only if mdns result size is 3 or more.

### Testing

Unit tests were updated and added to cover the new code. The unit tests were failing before the code was updated. CI will also validate general functionality.
closed 2024-12-17 by mergify[bot]