project-chip/connectedhomeip #36772: [BUG] Adaptive Lighting mode to Hue & Saturation mode transition failed

xshuqun, 2024-12-10:
### Reproduction steps

Facing a problem with Adaptive Lighting transiting to another color option

**DUT is using FreeRTOS and LwIP.**

After investigating the SDK, I observed the following:

1. The DUT is set to Adaptive Lighting by selecting it on the Homekit GUI.
2. Homekit sends a CommandRequest for `moveToColorTemp` and DUT handles callback `updateTempCommand` to set color temp.
3. Once the transition stabilizes, the DUT continuously triggers `updateTempCommand`.
4. A HueSaturation color option is selected on the Homekit GUI.
5. Homekit sends a CommandRequest `moveToHueAndSaturation` and DUT receives it + register timer with scheduleTimerCallbackMs for `updateHueSatCommand`.
6. Some Messages (for ReadRequest or other request) are received, which triggers PostEvent and added to the task queue (duplicate message packets are also observed).
7. The DUT enters RunEventLoop and continuously processes the task queue.
8. Homekit sends a CommandRequest to `stopAllColorTransitions` (sent by Apple on time, regardless of whether the DUT has completed the transition, I assumed).
9. The DUT handles `stopAllColorTransitions` and cancels/stops all ongoing timers for color control.
10. The DUT fails to complete the color transition.

Basically, before the DUT is able to handle the active timer, Homekit  has sent another CommandRequest to stop all color transition and cancel all the timers.

Expected outcome:

1. The DUT is set to Adaptive Lighting by selecting it on the Homekit GUI.
2. Homekit sends a CommandRequest for `moveToColorTemp` and DUT handles callback `updateTempCommand` to set color temp.
3. A HueSaturation color option is selected on the Homekit GUI.
4. Homekit sends a CommandRequest `moveToHueAndSaturation` and DUT receives it + register timer with scheduleTimerCallbackMs for `updateHueSatCommand`.
5. DUT triggers callback `updateHueSatCommand` and change Hue + Saturation.

I have attached the DUT with analysis of how FreeRTOS queue is being handled.
[1] [DUT_adaptive_lighting_success_logs.txt](https://github.com/user-attachments/files/18070738/DUT_adaptive_lighting_success_logs.txt)
[2] [DUT_adaptive_lighting_failed_logs.txt](https://github.com/user-attachments/files/18070737/DUT_adaptive_lighting_failed_logs.txt)

This is a slight video on the operation using Homekit:
[1] https://github.com/user-attachments/assets/9edbafef-4ed0-4b2d-9e3c-dfbea94c19ca

### Bug prevalence

always

### GitHub hash of the SDK that was being used

master

### Platform

ameba

bzbarsky-apple, 2024-12-11:
@xshuqun So are you claiming this is a problem with the _server_ (and if so, what exactly is the problem), or with the _client_ (HomeKit) sending the wrong commands?

xshuqun, 2024-12-12:
Apparently, I think the server hit a corner case where there is insufficient time to execute the necessary command.
However, whether the client is sending these command correctly, I am uncertain.

I had provided details in the description, where server (using FreeRTOS) has to clear its queue before handling active timer, but before the server could handle the timer, it was cancelled by the next command sent by the client. 

I would like some advise if this can be handled in another way or other solutions? (e.g., Like handling the active timer first and clearing the queue?)

bzbarsky-apple, 2024-12-12:
@xshuqun Honestly, to me it sounds like the client is doing something wrong (stopping transitions before they have finished) and that you should report this issue to Apple....

xshuqun, 2024-12-17:
I did report this issue to the Apple Team through the Feedback Assistant, but apparently, there is no response.

bzbarsky-apple, 2024-12-18:
Yeah, that one is resolved with "ask the reporter these questions" back in October....  I have no idea why no one reached out to you, but I will try to prod it along.  Chances are nothing will happen until Jan 6 at the earliest, though; everyone involved is on vacation until then.

xshuqun, 2025-02-06:
@bzbarsky-apple wondering if there is any feedback from the team on this discussion.

woody-apple, 2025-02-15:
Please ping the feedback ticket at Apple, a CSA forum is not the place to provide feedback to Apple Home’s implementation

closed 2025-02-15 by woody-apple