project-chip/connectedhomeip #36815: [Feature] Matter build system Intergrate to another build system.

tongjiang, 2024-12-12:
### Feature description

The Matter build system need to call the . scripts/activate.sh to download and set the build enviroment. 
1.  If we just copy this directory to officemate, the . scripts/activate.sh executtion will report error. 
2. There is a lot absolute path in the .enviroment files which make it hard for the developer of our company to use, and it's not possible to modify all the absolute path.
3. The company has a lot of security measure, just as the internet network is not working for every one. It is not possible for every one to download the enviorment.

So is there a way to let every one has a stable build enviroment instead of let everyone to download?

### Platform

all

bzbarsky-apple, 2024-12-12:
@andy31415

andy31415, 2024-12-13:
@tongjiang you may be able to use a docker/podman image with everything preconfigured and copy that.

andy31415, 2024-12-13:
https://pigweed.dev/pw_env_setup/ describes a `pw.pw_env_setup.virtualenv.pip_install_offline` that is supposed to allow for offline pip parts. 

for CIPD I am unsure, CLI documents a `CIPD_CACHE_DIR` which could be potentially referenced and copied as well.

I imagine we may have an easier time allowing for "offline bootstrap based on some caches/repository" rather than allowing for `copy .environment and expect it to work` as .environment may reference various O/S things and I am unsure how much similarity between source and destination machines we would need to enforce.