project-chip/connectedhomeip #36827: Example app conformance failed: chip-energy-management-app

andy31415, 2024-12-12:
- [ ] IDM-10.3: 40 (0x28) BasicInformation: Revision found on cluster (3) does not match revision listed in the spec (4)
- [ ] IDM-10.3: 47 (0x2f) PowerSource: Revision found on cluster (2) does not match revision listed in the spec (3)
- [ ] IDM-10.3: 48 (0x30) GeneralCommissioning: Revision found on cluster (1) does not match revision listed in the spec (2)
- [ ] IDM-10.3: 3 (0x03) Identify: Revision found on cluster (4) does not match revision listed in the spec (5)
- [ ] IDM-10.3: 148 (0x94) WaterHeaterManagement: Revision found on cluster (ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.Failure: 1>))) does not match revision listed in the spec (2)
- [ ] IDM-10.3: 153 (0x99) EnergyEvse: Revision found on cluster (2) does not match revision listed in the spec (3)
- [ ] IDM-10.3: 157 (0x9d) EnergyEvseMode: Revision found on cluster (1) does not match revision listed in the spec (2)
- [ ] IDM-10.3: 159 (0x9f) DeviceEnergyManagementMode: Revision found on cluster (1) does not match revision listed in the spec (2)
- [ ] IDM-10.5: 152 (0x98) DeviceEnergyManagement: Extra cluster found on endpoint with device types [Descriptor.Structs.DeviceTypeStruct(deviceType=1292, revision=1), Descriptor.Structs.DeviceTypeStruct(deviceType=1296, revision=1)]
- [ ] IDM-10.5: 148 (0x94) WaterHeaterManagement: Extra cluster found on endpoint with device types [Descriptor.Structs.DeviceTypeStruct(deviceType=1292, revision=1), Descriptor.Structs.DeviceTypeStruct(deviceType=1296, revision=1)]
- [ ] IDM-10.5: 158 (0x9e) WaterHeaterMode: Extra cluster found on endpoint with device types [Descriptor.Structs.DeviceTypeStruct(deviceType=1292, revision=1), Descriptor.Structs.DeviceTypeStruct(deviceType=1296, revision=1)]
- [ ] IDM-10.5: 159 (0x9f) DeviceEnergyManagementMode: Extra cluster found on endpoint with device types [Descriptor.Structs.DeviceTypeStruct(deviceType=1292, revision=1), Descriptor.Structs.DeviceTypeStruct(deviceType=1296, revision=1)]
- [ ] IDM-10.5: 47 (0x2f) PowerSource: Extra cluster found on endpoint with device types [Descriptor.Structs.DeviceTypeStruct(deviceType=22, revision=1)]

soares-sergio, 2024-12-12:
All of these have been fixed in https://github.com/project-chip/connectedhomeip/pull/36201

Last time I tested, the only conformance failure now is the lack of a thermostat implementation in the water heater app.

andy31415, 2025-01-06:
Assigned to myself as a "worked on" placeholder pending getting seats for new developers in the project

closed 2025-03-03 by mergify[bot]