project-chip/connectedhomeip #36839: [Build] CIPD 包管理器下载和安装 `fuchsia/third_party/clang/linux-amd64` 和 `fuchsia/third_party/sysroot/bionic` 时失败了

Ahdjdus, 2024-12-13:
### Build issue(s)

有谁可以帮帮我，在执行 `./install.sh` 出现了下面的错误，已经重新构建了很多便了，还是这样。真要疯了。。。。

```shell

[P55267 15:33:45.646 storage.go:245 I] [fetch 2/2] Fetching:  98.6/268.2 MB ( 37%, 6.76 MB/s)
[P55267 15:33:46.286 storage.go:245 I] [fetch 2/2] Fetching: 101.0/268.2 MB ( 38%, 6.61 MB/s)
[P55267 15:33:46.787 storage.go:245 I] [fetch 2/2] Fetching: 105.5/268.2 MB ( 39%, 6.84 MB/s)
[P55267 15:33:47.349 storage.go:245 I] [fetch 2/2] Fetching: 108.1/268.2 MB ( 40%, 6.47 MB/s)
[P55267 15:33:47.854 storage.go:245 I] [fetch 2/2] Fetching: 111.0/268.2 MB ( 41%, 6.53 MB/s)
[P55267 15:33:48.358 storage.go:245 I] [fetch 2/2] Fetching: 113.7/268.2 MB ( 42%, 6.36 MB/s)
[P55267 15:33:48.860 storage.go:245 I] [fetch 2/2] Fetching: 116.6/268.2 MB ( 43%, 6.23 MB/s)
[P55267 15:33:49.181 storage.go:264 W] [fetch 2/2] Transient error: writing to the output instance file: unexpected EOF
[P55267 15:33:51.183 client.go:1867 E] [fetch 2/2] Failed to fetch fuchsia/third_party/sysroot/bionic:9Fyg2LRoELlNKn28ZfkJIzfWqVaLJgtRFzpauTFNol4C: failed to download after multiple attempts
[P55267 15:33:51.233 client.go:1963 E] [unzip 2/2] Failed to install fuchsia/third_party/sysroot/bionic in "pigweed/clang_sysroot": failed to download after multiple attempts
Errors:
  failed to install fuchsia/third_party/clang/linux-amd64 in "pigweed": failed to download after multiple attempts
  failed to install fuchsia/third_party/sysroot/bionic in "pigweed/clang_sysroot": failed to download after multiple attempts
/home/yang-esp/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/cipd ensure -ensure-file /home/yang-esp/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages.ensure -root /home/yang-esp/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages -log-level debug -json-output /home/yang-esp/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages.json -max-threads 0 -cache-dir /root/.cipd-cache-dir

None
Traceback (most recent call last):
  File "/home/yang-esp/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 1002, in <module>
    sys.exit(main())
  File "/home/yang-esp/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 994, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/yang-esp/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 579, in setup
    result = step(spin)
  File "/home/yang-esp/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 696, in cipd
    if not cipd_update.update(
  File "/home/yang-esp/esp/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/update.py", line 389, in update
    subprocess.check_call(cmd, stdout=outs, stderr=subprocess.STDOUT)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/yang-esp/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/cipd', 'ensure', '-ensure-file', '/home/yang-esp/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages.ensure', '-root', '/home/yang-esp/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages', '-log-level', 'debug', '-json-output', '/home/yang-esp/esp/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages.json', '-max-threads', '0', '-cache-dir', '/root/.cipd-cache-dir']' returned non-zero exit status 1.

```

packages.json 文件内容

```json

{
  "error": "failed to install fuchsia/third_party/clang/linux-amd64 in \"pigweed\": failed to download after multiple attempts (and 1 other error)",
  "error_code": "cas_error",
  "error_details": {
    "package": "fuchsia/third_party/clang/linux-amd64",
    "version": "nPyRR3dznfRL_lvkABEzYUFenSo424F5eQzxyvF73bYC",
    "subdir": "pigweed"
  },
  "result": null
}

```

### Platform

esp32

Ahdjdus, 2024-12-14:
关于 fuchsia/third_party/sysroot/bionic 和 fuchsia/third_party/clang/linux-amd64 我下载好了压缩包，该把他们放到哪里？

Alami-Amine, 2025-01-10:
it looks like this was executed through ESP's [install.sh]( https://github.com/espressif/esp-matter/blob/main/install.sh) in esp-matter? I would say check with them first.
 Will close the issue for now, we can reopen it if needed.

closed 2025-01-10 by Alami-Amine