project-chip/connectedhomeip #36846: [Feature] Standardizing Thermostat SystemMode support detection in Matter

dusikasss, 2024-12-14:
### Feature description

I am working on an ESP32-based air conditioner device implementing the Matter protocol. I have encountered a problem in the current implementation of Matter integration in Home Assistant, in particular with regard to the Thermostat cluster.

Currently, Home Assistant determines the supported SystemModes for the Thermostat cluster based on manufacturer and product IDs. This approach is not ideal, as it lacks flexibility and scalability, especially for new devices entering the market.

The crux of the problem seems to be that Matter does not provide a standardized way to specify support for various SystemModes (for example, fan-only mode or dehumidification mode) through function bits or other universal methods.

Questions:

1. What is the recommended way to implement support for various SystemModes in the device to ensure proper detection by the Matter server? Is there a standard approach that does not rely on hard-coded lists of manufacturer/product IDs?

2. If there is currently no universal method, what possible solutions could be considered to standardize this process? What are the advantages and disadvantages of different approaches?

3. Are there any plans or ongoing discussions to improve this aspect of the Matter specification? If so, what approaches are being considered?

4. How does the Matter community see the solution to the problem of dynamic detection of device capabilities in the long term? What steps can be taken to improve this situation?

I would appreciate any ideas, recommendations or information about the ongoing discussions on this issue. If there is a need for additional research or prototyping to explore possible solutions, I am ready to contribute to this process.

Thank you for your attention to this issue!

### Platform

esp32

### Anything else?

[https://github.com/home-assistant/core/blob/dev/homeassistant/components/matter/climate.py](url)

In the file `climate.py `, who is responsible for integrating Matter climate devices into the Home Assistant, has the following comment:

```python
# In the list below specify tuples of (vendorid, productid) of devices that

bzbarsky-apple, 2024-12-18:
@hasty

dusikasss, 2025-01-08:
Update: I've created a specific issue in Home Assistant repository demonstrating how this problem affects real-world integrations: home-assistant/core#135124

The issue shows that Home Assistant currently has to maintain hardcoded lists of vendor/product IDs to determine supported modes like DRY and FAN_ONLY, since there's no standardized way in Matter to detect these capabilities.

This creates problems for:
1. Device manufacturers - need to request adding their devices to HA lists
2. Integration maintainers - need to maintain device lists
3. Users - devices don't work properly out of the box

Having a standardized way to detect supported modes in Matter would solve these issues for all Matter-based platforms, not just Home Assistant.

dusikasss, 2025-01-08:
After investigating the Matter specification and ControlSequenceOfOperation attribute, I'd like to propose a potential solution.

Currently, ControlSequenceOfOperation in Matter spec has following values:
- 0: CoolingOnly
- 1: CoolingWithReheat
- 2: HeatingOnly
- 3: HeatingWithReheat
- 4: CoolingAndHeating
- 5: CoolingAndHeatingWithReheat

The attribute is already used to indicate basic thermostat capabilities, so it could be a natural place to add support for additional modes. We could extend it with new values:

Proposed new values:
- 6: CoolingAndHeatingWithDry (All cooling/heating modes + dry mode)
- 7: CoolingAndHeatingWithFanOnly (All cooling/heating modes + fan-only mode)
- 8: CoolingAndHeatingWithDryAndFan (Full feature set including dry and fan modes)
- etc.

Would appreciate feedback on this approach from Matter community. Are there any technical limitations or considerations that would prevent extending ControlSequenceOfOperation this way?

samryanuofl, 2025-03-06:
Could it use the FeatureMap attribute to indicate which modes are supported? For example add a [DRY] and [FANONLY] option and then in SystemMode make those modes depend on the feature bits.

TRGercek, 2025-08-29:
Hello,

Thank you for writing this response. I am also bothered by the same situation. Why aren't integrators adding these modes even though they exist on the air conditioning side?

Could you share your latest information on this subject with me?