project-chip/connectedhomeip #36848: light-switch-app -- Example code not able to compile

Siddhiq74, 2024-12-15:
### Build issue(s)

1.
2.
3.
...
I tired to compile light-switch-app after set-target to esp32h2.  It is stalling and the error and verbose output is given in the file.  But, the same file is able to compile when i set-targt to esp32c3.  can you suggest what is the problem.

### Anything else?

[image]
[idf_py_stdout_output_306290.txt](https://github.com/user-attachments/files/18138877/idf_py_stdout_output_306290.txt)

Siddhiq74, 2025-02-10:
Any update on the build issue

dhrishi, 2025-02-22:
@Siddhiq74 ESP32-H2 does not have Wi-Fi interface, so ideally the code related to Wi-Fi should not get compiled. Can you please try a clean build -
```
$ cd /path/to/examples/lighting-app/esp32
$ rm -rf sdkconfig sdkconfig.old build/
$ idf.py set-target esp32h2
$ idf.py build
```

If this does not resolve your problem, please share the details about the ESP-IDF and connectedhomeip version/branch you are using so that we can recreate this

Siddhiq74, 2025-02-24:
I am getting the same error.  My current Connectionhomeip version and commit hash is v1.3.0.0-3063-g43a8a9b8e1
43a8a9b8e1433c338ac8fabd772c3ba040d7518e

And esp-idf ver is commit hash is v5.5-dev-1998-gc71d74e2f8
c71d74e2f853b1135c63f47e349f2a08f63f3e01

Kindly, check and let me know the results

closed 2025-02-24 by Siddhiq74