project-chip/connectedhomeip #36884: [CI Testing] Improve TC_SWTCH testing on CI

arkq, 2024-12-18:
### Test issue(s)

The TC_SWTCH runs all test cases for endpoint 1, 2, 3, 4. However, not all this endpoints have features required for particular tests, so some of them are skipped. Such testing scenario fails in case when we enforce all test cases to pass (fail-on-skip) in order to catch tests skipped by mistake (e.g. endpoint has been changed).

For now the CI runs are filtered by endpoint and matching tests cases, e.g.: on endpoint 1 only test_TC_SWTCH_2_2 is run. However, this is also not ideal, because one might add another test case to the class without adding `--test-case` to the script args, and such test will not be run on CI at all...

Maybe this TC_SWTCH.py file should be split into particular test cases, so testing on CI will be more relievable and resilient to configuration omissions.

### Platform

other

cecille, 2024-12-19:
Some are skipped on purpose. The all clusters app is set up to run these tests across all endpoints. This is always the case, BTW, whether or not tests are selected automatically or by PICS.

arkq, 2024-12-19:
>  tests are selected automatically or by PICS.

Test cases are selected based on particular switch cluster features being present on selected endpoint.