project-chip/connectedhomeip #36898: [CERT-TEST-FAILURE] Unexpected error writing attribute - expected Unsupported Access, got 0

KaMeHaMeHaaaa, 2024-12-19:
### Feature Area

Other

### Test Case

TC-ACE-2.2

### Reproduction steps

**Step1:  Commissioning**
    ./chip-tool pairing ble-wifi 1 Matter 12345678 86155769 3518
**Step2:  Set privilege as View(3) for second controller**
    ./chip-tool accesscontrol write acl '[{ "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, { "privilege": 3, "authMode": 2, "subjects": [112234], "targets": null}]' 1 0
**Step3: Write nodeLabel attribute of basicinformation cluster on endpoint 0(rootnode)，and get reponse with unsupported access**
    ./chip-tool basicinformation write node-label "TEST“ 1 
**Step4: Write nodeLabel attribute of bridgedbasicinfomation cluster on dynamic endpoint，and get response with success**
    ./chip-tool bridgeddevicebasicinformation write node-label "TEST“ 1 5 --commissioner-nodeid 112234

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

13ab158f107c06ad9d4a2591aba438d212500b2d

### Platform

ameba

### Anything else?

- When the privilege for the second controller is set to view (3), attempting to write to the nodeLabel attribute of the basicInformation cluster on endpoint 0 (RootNode) returns : 0x7e (UNSUPPORTED_ACCESS), which is the expected behavior.
- However, when writing to the nodeLabel attribute of the bridged device basicInformation cluster, it returns SUCCESS instead of UNSUPPORTED_ACCESS, causing the test case to fail.
- I tried enabling the bridgedBasicInformationCluster as a server role in the .zap file and checked the nodeLabel attribute. However, I couldn’t find any definition for the access privileges of attributes of the bridged basicInformation cluster within the zap-generated/access.h file.
- Additionally, My declaration for the nodeLabel attribute of the bridged basicInformation cluster is as follows:
`DECLARE_DYNAMIC_ATTRIBUTE(BridgedDeviceBasicInformation::Attributes::NodeLabel::Id, CHAR_STRING, 33, ATTRIBUTE_MASK_WRITABLE)`
- Can someone tell me how to set access privileges for the nodeLabel attribute of the bridged basicInformation cluster?

KaMeHaMeHaaaa, 2024-12-19:
In step 3, the log shows that the requestPrivilege is manage (4)

> AccessControl: checking f=1 a=c s=0x000000000001B66A t= c=0x0000_0028 e=0 **p=m**

but in step 4, the log shows that the requestPrivilege is operate (3)

> AccessControl: checking f=1 a=c s=0x000000000001B66A t= c=0x0000_0039 e=5 **p=o**

cecille, 2024-12-19:
This just looks like a case where the test found an SDK bug. The proper solution here is to add the appropriate access markers into the zap XML here: https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/bridged-device-basic-information.xml, then re-run https://github.com/project-chip/connectedhomeip/blob/master/scripts/tools/zap_regen_all.py to propagate that change into the access.h and into the examples.