project-chip/connectedhomeip #36920: [BUG] Heap-Use-After-Free (UAF) in tv-casting-app when deleting a fabric after casting

BoB13-Matter, 2024-12-20:
### Reproduction steps

1. Build the `tv-casting-app` with AddressSanitizer (ASAN) enabled.
2. Open Terminal 1 and start the `tv-casting-app`:
    
    ```bash
    $ ./tv-casting-app
    ```
    
3. Open Terminal 2 and start the `tv-app`:
    
    ```bash
    $ ./tv-app
    ```
    
4. In the `tv-casting-app` shell, send a cast request:
    
    ```bash
    tv-casting-app > cast request 0
    ```
    
5. In the `tv-app` shell, respond with:
    
    ```bash
    tv-app > controller ux ok
    ```
    
6. In the `tv-casting-app`, print the list of fabrics and delete a specific fabric:
    
    ```bash
    tv-casting-app > cast print-fabrics
    tv-casting-app > cast delete-fabric <fabricId>
    ```
    
7. Observe the ASAN output in `tv-casting-app` indicating a heap-use-after-free issue.
[tv-casting-app UAF.txt](https://github.com/user-attachments/files/18207572/tv-casting-app.UAF.txt)

### Summary

A heap-use-after-free (UAF) issue occurs in the `tv-casting-app` when the `delete-fabric` command is executed. After removing a fabric, the system attempts to access a freed `ReadClient` object.

### Analysis and Description

The issue arises from a lifecycle mismatch between fabric cleanup and the `ReadClient` object references. The sequence of events leading to the issue is as follows:

1. Fabric Removal (FabricTable.cpp):
    - The `Delete()` method of `FabricTable` is called, which iterates through its delegates and invokes `OnFabricRemoved()`:
        
        ```cpp
        CHIP_ERROR FabricTable::Delete(FabricIndex fabricIndex)
        {
            if (mDelegateListRoot != nullptr)
            {
                FabricTable::Delegate * delegate = mDelegateListRoot;
                while (delegate)
                {
                    FabricTable::Delegate * nextDelegate = delegate->next;
                    delegate->OnFabricRemoved(*this, fabricIndex);
                    delegate = nextDelegate;
                }
            }
        }
        
        ```
        
2. ReadClient Cleanup (InteractionModelEngine.cpp):
    - The `OnFabricRemoved()` method loops through active `ReadClient` objects and calls `Close()`:
        
        ```cpp
        void InteractionModelEngine::OnFabricRemoved(const FabricTable & fabricTable, FabricIndex fabricIndex)
        {
            for (auto * readClient = mpActiveReadClientList; readClient != nullptr; readClient = readClient->GetNextClient())
            {
                if (readClient->GetFabricIndex() == fabricIndex)
                {
                    readClient->Close(CHIP_ERROR_IM_FABRIC_DELETED, false);
                }
            }
        }
        
        ```
        
3. ReadClient Deallocation (ReadClient.cpp):
    - The `Close()` method triggers the callback `mpCallback.OnDone(this)`, marking the `ReadClient` for destruction:
        
        ```cpp
        void ReadClient::Close(CHIP_ERROR aError, bool allowResubscription)
        {
            ...
            mpCallback.OnDone(this);
        }
        
        ```
        
4. Use of Freed Memory:
    - Despite being destroyed, the `ReadClient` remains in the `mpActiveReadClientList`. Subsequent iterations in `OnFabricRemoved()` access the dangling pointer. Specifically, the `readClient->GetNextClient()` call accesses the `mpNext` pointer of the destroyed `ReadClient` object, leading to a UAF:
        
        ```cpp
        ReadClient * GetNextClient() { return mpNext; }
        ```
        
    - This happens because `GetNextClient()` does not verify the validity of the `mpNext` pointer, which now points to a deallocated memory region.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

04e6a68cf8aa2c2d24dd1decb74d71a6f7a70dfe

### Platform

core

### Platform Version(s)

all versions

bzbarsky-apple, 2025-01-03:
@BoB13-Matter This would be an issue for any fabric removal with a live ReadClient, right?  Not specific to tv-casting?

yunhanw-google, 2025-01-08:
Just looking at this issue, application would take care of the delete for readClient inside ReadClient::OnDone, in the destructor of readClient, we will take care of the object removal in mpActiveReadClientList via mpImEngine->RemoveReadClient(this); this issue should be fine?

maybe tv-casting-app does not delete readClient object in onDone?

ReadClient::~ReadClient()
{
...
        if (mpImEngine)
        {
            mpImEngine->RemoveReadClient(this);
        }
    }
}

bzbarsky-apple, 2025-01-10:
The problem is we are holding a reference to the now-deleted thing in the loop in InteractionModelEngine::OnFabricRemoved, no?  @yunhanw-google

yunhanw-google, 2025-01-19:
@bzbarsky-apple 
you mean that we hold the reference for now deleted readClient in the below loop code in InteractionModelEngine::OnFabricRemoved ?
    for (auto * readClient = mpActiveReadClientList; readClient != nullptr; readClient = readClient->GetNextClient())
    {
        if (readClient->GetFabricIndex() == fabricIndex)
        {
            ChipLogProgress(InteractionModel, "Fabric removed, deleting obsolete read client with FabricIndex: %u", fabricIndex);
            readClient->Close(CHIP_ERROR_IM_FABRIC_DELETED, false);
        }
    }

I think mpImEngine->RemoveReadClient(this) in ~ReadClient() has removed that readClient reference?

void InteractionModelEngine::RemoveReadClient(ReadClient * apReadClient)
{
    ReadClient * pPrevListItem = nullptr;
    ReadClient * pCurListItem  = mpActiveReadClientList;

    while (pCurListItem != apReadClient)
    {
        //
        // Item must exist in this tracker list. If not, there's a bug somewhere.
        //
        VerifyOrDie(pCurListItem != nullptr);

        pPrevListItem = pCurListItem;
        pCurListItem  = pCurListItem->GetNextClient();
    }

    if (pPrevListItem)
    {
        pPrevListItem->SetNextClient(apReadClient->GetNextClient());
    }
    else
    {
        mpActiveReadClientList = apReadClient->GetNextClient();
    }

    apReadClient->SetNextClient(nullptr);
}

bzbarsky-apple, 2025-01-24:
> you mean that we hold the reference for now deleted readClient in the below loop code

Yes.

> I think mpImEngine->RemoveReadClient(this) in ~ReadClient() has removed that readClient reference?

How could it?  The reference is held on the stack in the `readClient` variable.

andy31415, 2025-01-24:
So the stack trace shows that Close eventually calls delete:

```
    #1 0x5dc1137ca60d in chip::Platform::MemoryFree(void*) /home/kcw/Matter/connectedhomeip/out/linux-x64-tv-casting-app-asan-ubsan-clang/../../examples/tv-casting-app/linux/third_party/connectedhomeip/src/lib/support/CHIPMem-Malloc.cpp:116:5
...
   #17 0x5dc11438fa8d in chip::app::ReadClient::Close(chip::ChipError, bool) /home/kcw/Matter/connectedhomeip/out/linux-x64-tv-casting-app-asan-ubsan-clang/../../examples/tv-casting-app/linux/third_party/connectedhomeip/src/app/ReadClient.cpp:235:16
    #18 0x5dc114292c19 in chip::app::InteractionModelEngine::OnFabricRemoved(chip::FabricTable const&, unsigned char) /home/kcw/Matter/connectedhomeip/out/linux-x64-tv-casting-app-asan-ubsan-clang/../../examples/tv-casting-app/linux/third_party/connectedhomeip/src/app/InteractionModelEngine.cpp:1943:25
```

And the loop is

```cpp
    for (auto * readClient = mpActiveReadClientList; readClient != nullptr; readClient = readClient->GetNextClient())
    {
        if (readClient->GetFabricIndex() == fabricIndex)
        {
            ChipLogProgress(InteractionModel, "Fabric removed, deleting obsolete read client with FabricIndex: %u", fabricIndex);
            readClient->Close(CHIP_ERROR_IM_FABRIC_DELETED, false);
        }
    }
```

So we are deleting `readClient` and then trying to do `readClient->GetNextClient()`. This looks buggy for sure.

yunhanw-google, 2025-01-25:
Agree, the loop variable is pointing to deleted memory.... And the body of the for will delete the object readClient points to.  So then readClient->GetNextClient() will be use-after-free.

yunhanw-google, 2025-01-25:
Just create the [PR](https://github.com/project-chip/connectedhomeip/pull/37199) for the fix and corresponding unit test to cover, thanks

closed 2025-01-27 by yunhanw-google