project-chip/connectedhomeip #36926: [TC-ACE-2.1] Issue when reading ThreadNetworkDiagnostics.Attributes.RouteTable on endpoint 2

Martin-NXP, 2024-12-20:
### Test issue(s)

**Failure identify with tests:**
1.TC-ACE-2.1
2.TC-IDM-2.2

**Error logs (TC-ACE-2.1):**
Problem: ProblemSeverity.ERROR
    test_name: Read access Checker - 1
    location:
       Endpoint: 2,
       Cluster:  53 (0x35) ThreadNetworkDiagnostics
      Attribute:8 (0x08)
    problem: Error reading <class 'chip.clusters.Objects.ThreadNetworkDiagnostics'>:<class 'chip.clusters.Objects.ThreadNetworkDiagnostics.Attributes.RouteTable'> = ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.Failure: 1>))
    spec_location:

[MatterTest] 12-13 07:33:24.218 INFO
Problem: ProblemSeverity.ERROR
    test_name: Read access Checker - 2
    location:
       Endpoint: 2,
       Cluster:  53 (0x35) ThreadNetworkDiagnostics
      Attribute:8 (0x08)
    problem: Error reading <class 'chip.clusters.Objects.ThreadNetworkDiagnostics'>:<class 'chip.clusters.Objects.ThreadNetworkDiagnostics.Attributes.RouteTable'> = ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.Failure: 1>))
    spec_location:

[MatterTest] 12-13 07:33:24.219 INFO
Problem: ProblemSeverity.ERROR
    test_name: Read access Checker - 3
    location:
       Endpoint: 2,
       Cluster:  53 (0x35) ThreadNetworkDiagnostics
      Attribute:8 (0x08)
    problem: Error reading <class 'chip.clusters.Objects.ThreadNetworkDiagnostics'>:<class 'chip.clusters.Objects.ThreadNetworkDiagnostics.Attributes.RouteTable'> = ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.Failure: 1>))
    spec_location:

[MatterTest] 12-13 07:33:24.219 INFO
Problem: ProblemSeverity.ERROR
    test_name: Read access Checker - 4
    location:
       Endpoint: 2,
       Cluster:  53 (0x35) ThreadNetworkDiagnostics
      Attribute:8 (0x08)
    problem: Error reading <class 'chip.clusters.Objects.ThreadNetworkDiagnostics'>:<class 'chip.clusters.Objects.ThreadNetworkDiagnostics.Attributes.RouteTable'> = ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.Failure: 1>))
    spec_location:

[MatterTest] 12-13 07:33:24.220 INFO
Problem: ProblemSeverity.ERROR
    test_name: Read access Checker - 5
    location:
       Endpoint: 2,
       Cluster:  49 (0x31) NetworkCommissioning
      Attribute:1 (0x01)
    problem: Error reading <class 'chip.clusters.Objects.NetworkCommissioning'>:<class 'chip.clusters.Objects.NetworkCommissioning.Attributes.Networks'> = ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.Failure: 1>))
    spec_location:

[MatterTest] 12-13 07:33:24.220 INFO
Problem: ProblemSeverity.ERROR
    test_name: Read access Checker - 5
    location:
       Endpoint: 2,
       Cluster:  53 (0x35) ThreadNetworkDiagnostics
      Attribute:8 (0x08)
    problem: Error reading <class 'chip.clusters.Objects.ThreadNetworkDiagnostics'>:<class 'chip.clusters.Objects.ThreadNetworkDiagnostics.Attributes.RouteTable'> = ValueDecodeFailure(TLVValue=None, Reason=InteractionModelError(<Status.Failure: 1>))
    spec_location:

**Context:**
NXP Thermostat application with OTBR support (examples/thermostat/nxp/zap/thermostat_matter_br.zap) with wifi network commissioning on endpoint 0 and Thread network commissioning on endpoint 2. Thread network is not started by default.

**The issue could be reproduce manually:**
1/ Run Wifi commissioning (pass)
2/ Read ThreadNetworkDiagnostics.Attributes.RouteTable (./chip-tool threadnetworkdiagnostics read route-table 1 2)
chip-tool error: 
[TOO] Run command failure: IM Error 0x00000501: General error: 0x01 (FAILURE)

**Observation:**
As the Thread network is not started, there is not Thread route, an error is returned here: 
https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/thread-network-diagnostics-server/thread-network-diagnostics-provider.cpp#L275C20-L275C27

**Questions**
1/ Is it normal to return an error if there no Thread route ? 
2/ If yes, should test prerequisites be updated in order to start the Thread network before reading attributes ?

bzbarsky-apple, 2025-01-03:
>  1/ Is it normal to return an error if there no Thread route ?

Spec does not very clearly define behavior, but I would expect empty list to be returned for RouteTable if there is no Thread network....  @jmartinez-silabs thoughts?

jmartinez-silabs, 2025-01-09:
@bzbarsky-apple @Martin-NXP imo we should return a empty list if the thread network is not up.

I didn't see any specifications on the spec tho what should be the expected result in such situations.

The test plan could  also specify the state of the device ( on a thread network)  as a prerequisite to this test

bzbarsky-apple, 2025-01-10:
> I didn't see any specifications on the spec tho what should be the expected result in such situations.

That needs to be fixed....

@Martin-NXP This is not my area of expertise (either thread or certification), so I don't have any specific recommendations.

PR #37032 (2025-01-13): Return empty thread route table when thread network is not up
Fixes #36926 
Command "./chip-tool threadnetworkdiagnostics read route-table" return an error if the thread network is not up (OTBR NXP thermostat app). Issue identify by the test lab during matter 1.4 certification.
Proposal: return empty route table instead of an error when there is no route.

### Testing
Tested with the NXP thermostat OTBR example, with RW612 platform.
The chip-tool command is not failing any more with this fix and print 0 entrance for the route table when the thread network is not up.
closed 2025-01-13 by mergify[bot]