project-chip/connectedhomeip #36943: [Build] CHIPTool app

BlackLordj, 2024-12-30:
### Build issue(s)

I'm trying to follow the instructions in https://github.com/project-chip/connectedhomeip/blob/v1.4-branch/docs/guides/android_building.md to build the CHIPtool app but I'm getting some errors
1.Unresolved reference: matter
2.
3.
...

### Anything else?

[image]
[image]

BlackLordj, 2024-12-30:
I read that this command will create some .jar files in Chiptool/app/libs but when I checked after running it it didn't create at all. My CPU is X86_64 and I use the command TARGET_CPU=x64 ./scripts/examples/android_app_ide.sh

andy31415, 2025-01-06:
@BlackLordj what were the exact commands you ran and under what O/S and configuration?

Our automated builds that are validated in CI generally only compile via command-line and your screenshot seems to show an Android Studio build, so I wonder if that is the cause. Does command-line build (via build-examples) work?