project-chip/connectedhomeip #36990: Python controller crashes with error: 'Aborted (Core Dumped)'

Harshith-GRL, 2025-01-08:
**Description:**
When executing the TC_Darwin_pair.py script from Stress and stability tests [repo](https://github.com/CHIP-Specifications/matter-qa), in numerous Occasions the python controller is crashing is crashing with  error 
                        `free (): double free detected in tcache 2 `
                        `Aborted (core dumped)`

GitHub hash of the SDK that were being used are :  1805b3ecad8c516f5f2b6a678bf0cdf29877c449     1b4c56cf5aead8cf45357bf32060fa0de848f186

**Platform**
I Have performed the execution on two different DUT (Raspberrypi and Nordic board), using ble-wifi and ble-thread provisioning 
_Controller Platform_ - python controller running on Raspberrypi 4B -8GB,
_Nordic Board as DUT_ - all-cluster-app running on Nordic nrf52840-DK board
_Raspberrypi as DUT_ - all-cluster-app running  on Raspberrypi 4B -8GB
Provisioning method for _Raspberrypi DUT is Ble-wifi_ and _nordic DUT is Ble-thread_

**Test scenario**

1. Pair/Provision the DUT (All-cluster-app running on raspi) as per drawiin pairing flow using ble-wifi/ ble-thread.
2. Read device information after the device is provisioned
3. sleep for 2 secs
4. Unpair the DUT (i.e. removing the fabric)
5. Factory reset the DUT
6. repeat step1 - 6 for 500 iterations
7. Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Darwin_Pair.py

**Screenshot**
This error will not appear in the python logs, Hence I am attaching a screenshot to represent how the error looks
[image]

### Execution Details
_Git hub hash id: 1b4c56cf5aead8cf45357bf32060fa0de848f186
Iterations scheduled to run: 500
Iterations executed before crash: 132
Python Log controller for Nordic as DUT:_  
[Nordic_as_dut_test_log_1b4c56cf5a.zip](https://github.com/user-attachments/files/18346190/Nordic_as_dut_test_log_1b4c56cf5a.zip)

_Git hub hash id: f8d457a446456d5df1e750e728ea48e8fce9d989
Iterations scheduled to run: 500
Iterations executed before crash:  263
Python Log controller for Raspberrypi as DUT:_ 
[rpi_ble-wifi_test_log_f8d457a44.zip](https://github.com/user-attachments/files/18346272/rpi_ble-wifi_test_log_f8d457a44.zip)

_Git hub hash id: f8d457a446456d5df1e750e728ea48e8fce9d989
Iterations scheduled to run: 500
Iterations executed before crash: 25
Python Log controller for Nordic as DUT:_
[Nordic_as_dut_test_log_f8d457a44.zip](https://github.com/user-attachments/files/18346273/Nordic_as_dut_test_log_f8d457a44.zip)

cecille, 2025-01-09:
do you have a backtrace from the core?

Harshith-GRL, 2025-01-10:
> do you have a backtrace from the core?
currently i do not have any backtrace captured.....
would you recommend any specific method or process to achieve this backtrace capture??

bzbarsky-apple, 2025-01-13:
Run your python under a debugger such as gdb or lldb.

Harshith-GRL, 2025-01-15:
I was able to capture the back-trace using the `gdb` with `bt` command when the code crashed 

[trace_back_file.txt](https://github.com/user-attachments/files/18427791/trace_back_file.txt)

Harshith-GRL, 2025-01-15:
I even took generated the core dump file and attaching here

Harshith-GRL, 2025-01-15:
@cecille and @bzbarsky-apple I have captured the core dump using  ` generate-core-file`  Git is not allowing me to upload the file any suggestions on how we could share it to you folks?

Harshith-GRL, 2025-01-16:
I have uploaded  the core dump file here https://graniteriverlabs1-my.sharepoint.com/:f:/g/personal/qanderson_graniteriverlabs_com/ErKYTMry_iZFmhML6I7cVFIBLvCjJwL3aEdiJ2Q7_siM7Q

bzbarsky-apple, 2025-01-17:
Looks like a double-free in `MdnsAvahi`?  @cecille do you know who might know about that code?

andy31415, 2025-01-24:
I have not worked with Avahi - I believe the people that originally implemented the avahi bits are no longer working on the SDK. I am generally familiar with the minMDNS stack (and probably the default maintainer of that) however I do not maintain avahi.

Do we have a printable stacktrace that we could attach to this bug? Might make it easier than download + unzip + gdb + stacktrace

andy31415, 2025-01-24:
I do not actually know how to get the stacktrace without the binary file ... Please upload the stack trace as well as a text file.

Harshith-GRL, 2025-02-20:
apologies for the late response but we are currently not seeing this issue in current execution when we find it we will upload them ASAP

Harshith-GRL, 2025-02-21:
hi @andy31415  we were able to see that error again today, as requested i have taken the stack trace into a text file, and here it is, if you need, i just ran a few commands like `bt full` and `thread apply all bt full` commands when the gdb was enabled, if you need any specific stack trace or logs do provide me with some info on what command u need me to run to extract the data you need

[thread_execution_crash_core_dumped.txt](https://github.com/user-attachments/files/18906310/thread_execution_crash_core_dumped.txt)

andy31415, 2025-02-21:
The stack trace bits seem to be:

```
#8  0x0000fffff1a220d0 in chip::Platform::MemoryFree (p=0xffffe4024fb0) at ../../src/lib/support/CHIPMem-Malloc.cpp:116
        __func__ = "MemoryFree"
#9  0x0000fffff2172558 in chip::Platform::Delete<chip::Dnssd::MdnsAvahi::ResolveContext> (p=0xffffe4024fb0) at ../../src/lib/support/CHIPMem.h:169
#10 0x0000fffff216fc24 in chip::Dnssd::MdnsAvahi::StopResolve (this=Python Exception <class 'RecursionError'>: maximum recursion depth exceeded
0xfffff234a848 <chip::Dnssd::MdnsAvahi::sInstance>, name=0xffffeb40e610 "DEF57AE973AE4C25-00000000BC1E2E02") at ../../src/platform/Linux/DnssdImpl.cpp:848
        it = 0xffffe4024fb0
        truncate_end = 0xffffe4024fb0
#11 0x0000fffff2170bc8 in chip::Dnssd::ChipDnssdResolveNoLongerNeeded (instanceName=0xffffeb40e610 "DEF57AE973AE4C25-00000000BC1E2E02") at ../../src/platform/Linux/DnssdImpl.cpp:1071
#12 0x0000fffff1fdc1f0 in chip::Dnssd::DiscoveryImplPlatform::NodeIdResolutionNoLongerNeeded (this=0xfffff2349a28 <chip::Dnssd::DiscoveryImplPlatform::sManager>, peerId=...) at ../../src/lib/dnssd/Discovery_ImplPlatform.cpp:701
        name = "DEF57AE973AE4C25-00000000BC1E2E02"
#13 0x0000fffff2010304 in chip::AddressResolve::Impl::Resolver::HandleAction (this=0xfffff234a290 <chip::AddressResolve::Resolver::Instance()::gResolver>, current=...) at ../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:355
        action = {mResult = {error = {static kRangeStart = 24, static kRangeLength = 8, static kValueStart = 0, static kValueLength = 24, static kSdkPartStart = 8, static kSdkPartLength = 3, static kSdkCodeStart = 0, static kSdkCodeLength = 8, mError = 50, mFile = 0xfffff222aa40 "src/lib/address_resolve/AddressResolve_DefaultImpl.cpp", mLine = 124}, result = {address = {static kMaxToStringSize = 76, mIPAddress = {static kMaxStringLength = 46, Addr = {50, 19873, 4062358080, 65535}}, mTransportType = 124, mPort = 0, mInterface = {static kMaxIfNameLength = 16, static kPlatformNull = 0, mPlatformInterface = 0}, mRemoteId = 0}, mrpRemoteConfig = {mIdleRetransTimeout = std::chrono::duration = { 0ms }, mActiveRetransTimeout = std::chrono::duration = { 0ms }, mActiveThresholdTime = std::chrono::duration = { 0ms }}, supportsTcpServer = false, supportsTcpClient = false, isICDOperatingAsLIT = false}}, mResultType = chip::AddressResolve::Impl::NodeLookupResult::kLookupError}
        peerId = {mNodeId = 3156094466, mCompressedFabricId = 16065882388800883749}
        listener = 0xffffdca00dc8
#14 0x0000fffff20104c4 in chip::AddressResolve::Impl::Resolver::HandleTimer (this=0xfffff234a290 <chip::AddressResolve::Resolver::Instance()::gResolver>) at ../../src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:384
```

So it seems that a `ChipDnssdResolveNoLongerNeeded` may have tried to double-free something.

andy31415, 2025-02-21:
It looks to me that when we time out a discovery, we get a double-free in python. Unsure why, it would have been great to have info on the stack trace of where the first free happened.

Any chance we can customize the builds? I think if we build the library with asan, we would get both the check and the original location of the free. Otherwise this needs some local reproduction to see.

Harshith-GRL, 2025-02-21:
we can try the customisation build method, kinda new to using 'asan' to build python but will try that, any particular method you would recommend so that we are on the same page for building python using 'asan'

andy31415, 2025-02-21:
I tried to compile locally and it did not work :( asan is undefined. Will need to work a bit to figure out how to set up things. Since this is a crash after another crash, this is not a high priority investigation for me.

LibertasIoT, 2025-04-13:
I encountered the same bug.
We build our own C++ based Matter controller. This is the only unsolved bug so far.
Any progress?

Saravana-kr22, 2025-04-16:
@feasel0 any update on this issue it seems other people in the community is also seeing this issue.

LibertasIoT, 2025-06-04:
It's still a mystery to me. I also spent hours inspecting the code, but found nothing. I don't believe it's caused by thread safety.

I've been running multiple instances since then, and I no longer experience any crashes on both the armv7 and arm64 flavors of Raspberry Pi 2, 4, and 5.

paresy, 2026-02-17:
We are still testing, but for us this patch fixes the issue.

```
diff --git a/src/platform/Linux/DnssdImpl.cpp b/src/platform/Linux/DnssdImpl.cpp
index 802d2fd1b3..be316c8b4e 100644
--- a/src/platform/Linux/DnssdImpl.cpp
+++ b/src/platform/Linux/DnssdImpl.cpp
@@ -839,16 +839,19 @@ void MdnsAvahi::FreeResolveContext(size_t handle)
 
 void MdnsAvahi::StopResolve(const char * name)
 {
-    auto truncate_end = std::remove_if(mAllocatedResolves.begin(), mAllocatedResolves.end(),
-                                       [name](ResolveContext * ctx) { return strcmp(ctx->mName, name) == 0; });
-
-    for (auto it = truncate_end; it != mAllocatedResolves.end(); it++)
-    {
-        (*it)->mCallback((*it)->mContext, nullptr, Span<Inet::IPAddress>(), CHIP_ERROR_CANCELLED);
-        chip::Platform::Delete(*it);
-    }
-
-    mAllocatedResolves.erase(truncate_end, mAllocatedResolves.end());
+    for (auto it = mAllocatedResolves.begin(); it != mAllocatedResolves.end();)
+    {
+        if (strcmp((*it)->mName, name) == 0)
+        {
+            (*it)->mCallback((*it)->mContext, nullptr, Span<Inet::IPAddress>(), CHIP_ERROR_CANCELLED);
+            chip::Platform::Delete(*it);
+            it = mAllocatedResolves.erase(it);
+        }
+        else
+        {
+            ++it;
+        }
+    }
 }
```