project-chip/connectedhomeip #37011: Create an `asserts.py` or similar splitout to share asserts for test cases

andy31415, 2025-01-09:
https://github.com/project-chip/connectedhomeip/pull/36997 updated some asserts to be easier to read, however they are still repeated across different files.

We should enhance existing mobly asserts with some chip asserts for any custom typical operations like `is_valid_....` checks for numbers/strings, validation of MEI values, validation of ranges and so on (e.g. https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/matter_testing_infrastructure/chip/testing/global_attribute_ids.py#L119 has some `is_valid/standard/....` checks but as bools not as assets).

We should build an `asserts` module that builds upon mobly assertions to add matter specific ones somewhere in matter_testing_infrastructure and start using them (we could have something like `matter_asserts.assert_valid_uint32(....)` in the code.

asirko-soft, 2025-01-16:
For this task the approach would be the following:
1) create matter_assers.py that would contain high reuse like "valid string/integer" or "integer in range" and so on
2) is_ checks from global_attribute_ids.py should be wrapped into proper asserts and placed in matter_asserts.py
3) create followup PR to convert existing tests to use the new asserts module
4) update documentation

asirko-soft, 2025-01-16:
First step is ready for review in PR: https://github.com/project-chip/connectedhomeip/pull/37083/