project-chip/connectedhomeip #37041: [BUG] Use-after-free in InteractionModelEngine when accessing Session after WriteHandler::Close()

agatah2333, 2025-01-11:
### Reproduction steps

### Description
Found a use-after-free bug in Matter SDK where `WriteHandler` prematurely releases `ExchangeContext` in its `Close()` method, violating the lifetime rules specified in `ExchangeDelegate` documentation. This causes a crash when `InteractionModelEngine` later attempts to access the freed Session.

### Call Sequence
```cpp
InteractionModelEngine::OnMessageReceived
-> InteractionModelEngine::OnWriteRequest
-> WriteHandler::OnWriteRequest
-> WriteHandler::HandleWriteRequestMessage
-> WriteHandler::Close()  // ExchangeContext is released here
-> Returns to InteractionModelEngine::OnMessageReceived  // Attempts to access Session through ExchangeContext->IsGroupExchangeContext() when Session has already been freed
```

### Root Cause
The issue violates the lifetime rules specified in ExchangeDelegate documentation, which states in [link](https://github.com/project-chip/connectedhomeip/blob/621b9c939c11ba8a4675b329c030960e15b2228d/src/messaging/ExchangeDelegate.h#L36
)
:
On reception of a message on an exchange, the ownership transfers to the OnMessageReceived callee. If you return from OnMessageReceived and no messages were sent on that exchange, the exchange will transfer back to the ExchangeMgr and it will automatically close it.

### Impact

Causes crash due to use-after-free when accessing Session after it's been freed
Potential security implications due to memory corruption

[image]

### Bug prevalence

each time

### GitHub hash of the SDK that was being used

561d23d0db215a99705ff0696e73853c8edf11b2 

### Platform

core

### Platform Version(s)

all versions

bzbarsky-apple, 2025-01-13:
> The issue violates the lifetime rules specified in ExchangeDelegate documentation

Closing an exchange explicitly is not a rules violation.  You _can_ unwind and let auto-close close it, but explicit close is allowed.

But it does look like InteractionModelEngine and WriteHandler might disagree on the API contract between the two of them?

bzbarsky-apple, 2025-01-13:
That said. I am surprised this is use-after-free, not a null-deref.  Could you include the full AddressSanitizer output (ideally as a text attachment), showing where things we deallocated, etc?

bzbarsky-apple, 2025-01-13:
And now I've attempted to reproduce this on SHA 0129a570dcaa0f64b86966004cc8019b932a08b0 and I can't reproduce, using an AddresSanitizer-enabled build that hits the codepath described in the issue.

@agatah2333 I just noticed that the SHA you say you are using (561d23d0db215a99705ff0696e73853c8edf11b2) is quite old (that's the 1.0 release).  Do you see this problem on newer SHAs as well?

agatah2333, 2025-01-13:
> And now I've attempted to reproduce this on SHA [0129a57](https://github.com/project-chip/connectedhomeip/commit/0129a570dcaa0f64b86966004cc8019b932a08b0) and I can't reproduce, using an AddresSanitizer-enabled build that hits the codepath described in the issue.
> 
> @agatah2333 I just noticed that the SHA you say you are using ([561d23d](https://github.com/project-chip/connectedhomeip/commit/561d23d0db215a99705ff0696e73853c8edf11b2)) is quite old (that's the 1.0 release). Do you see this problem on newer SHAs as well?

Thank you for your reply! I discovered this bug in SHA **ba9faf2,** and I forked the project from the v1.4 branch. Apologies for any confusion caused.

I believe this issue violates the lifetime rules specified in the ExchangeDelegate documentation. Specifically, if InteractionModelEngine is the callee of the OnMessageReceived function, it should be responsible for the release. This is outlined in the instructions provided in [this section of the ExchangeDelegate documentation](https://github.com/project-chip/connectedhomeip/blob/621b9c939c11ba8a4675b329c030960e15b2228d/src/messaging/ExchangeDelegate.h#L36). Instead, it seems to rely on WriteHandler::Close() for the release, which I believe is incorrect.

bzbarsky-apple, 2025-01-13:
> Thank you for your reply! I discovered this bug in SHA ba9faf2,

OK.  So presumably I am not in fact exercising the exact codepath you are, since I still can't reproduce, but since your issue does not describe the payload(s) involved and you still haven't provided the cropped-off parts of the AddressSanitizer output, it's really hard to tell what the difference is.

> Specifically, if InteractionModelEngine is the callee of the OnMessageReceived function, it should be responsible for the release. 

Here's the thing: I don't see how ` WriteHandler::Close` is even releasing the exchange.  It calls `Release()` on the _exchange holder_, which is a no-op in terms of exchange ownership unless the exchange is in the "I have sent a message ta the needs a response" or "I will send a message" state.  And I don't see how it would be in one of those states on this codepath, except if we actually sent a response successfully.  But then the IM engine will have `status == Status::Success` so won't touch the exchange.

Some information about which exact codepaths are being followed in your case, or what exact payloads are involved, would be helpful @agatah2333

agatah2333, 2025-01-15:
[crash.txt](https://github.com/user-attachments/files/18430459/crash.txt)

As shown in the attached crash.txt, the `ProcessMessage` interface is indeed a custom testing interface I created to make it easier to test the interaction model message handling flow. The test interface I created (`ProcessMessage`) is designed to directly feed messages into this processing flow, simulating how messages would arrive in a real Matter exchange. In this case, it's specifically sending a Write Request message that triggers the lifetime management issue between WriteHandler and InteractionModelEngine.

Regarding your questions about the exact codepath and payload:

The crash occurs because WriteHandler's Close() triggers a chain of releases that frees the ExchangeContext (and its associated Session) prematurely:

WriteHandler::Close()
-> ExchangeHolder::Release() 
-> ExchangeContext::Abort()
-> ExchangeContext::Release()
-> ExchangeContextDeletor::Release()
-> ExchangeManager::ReleaseContext()
-> HeapObjectPool::ReleaseObject()
-> Platform::Delete()
-> Platform::MemoryFree()

As shown in the AddressSanitizer output, after this chain completes, InteractionModelEngine attempts to access the freed Session.

The crash happens consistently because the WriteHandler's Close() is called within OnWriteRequest before returning control to InteractionModelEngine, which then tries to use the already freed Session.

I understand your point about explicit close being allowed. However, in this case, the issue isn't about whether explicit close is allowed, but rather about the sequencing of operations between WriteHandler and InteractionModelEngine. The WriteHandler is freeing resources that InteractionModelEngine still expects to be valid when processing returns to it.

bzbarsky-apple, 2025-01-15:
>  -> ExchangeContext::Abort()

Why?  What state is the exchange in at this point and why?

Again, what is the actual payload of the WriteRequest message involved?  That would let me stop asking you these questions and just debug with that message myself. 

> The WriteHandler is freeing resources that InteractionModelEngine still expects to be valid when processing returns to it.

That clearly shouldn't happen.  But so far I haven't found how it _can_, so I am not quite sure what the right fix is, so far, because it depends on the circumstances where this could arise.

agatah2333, 2025-01-15:
Here's the exact message payload that triggers the crash (in hex) or you can check in the zip file:
```hex
fb 4a 35 00 00 00 00 fb  ff ff ff 02 00 00 00 00
00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00
00 00 00 00 00 00 00 00  00 00 00 00 00 02 67 2f
69 6d 2f 00 17 24 02 00  24 03 1d 24 04 01 18 18
29 03 11 00 00 00
```

This is a Matter write request message where: fb 4a 35 indicates a write request header

This malformed write request triggers the validation failure in WriteHandler, leading to the premature cleanup sequence described earlier.
The ExchangeContext::Abort() is called as part of the SDK's internal cleanup sequence through ExchangeHolder::Release(), as shown in src/messaging/ExchangeHolder.h.
Would this help with reproducing and debugging the issue?

[crash-86607698799de67c21e6a98067f1e48f898b0f5f.zip](https://github.com/user-attachments/files/18431071/crash-86607698799de67c21e6a98067f1e48f898b0f5f.zip)

bzbarsky-apple, 2025-01-15:
Just to make sure we are talking about the same thing, is that the "Application Payload" after decryption and the message header and protocol header have been stripped off?  So the fact that it's a write request message (06) is not actually in that byte string, right?

I would expect that to start with byte 0x15 for the anonymous TLV structure that represents the payload....  So is this just entirely malformed TLV?  Or are we talking about something else entirely?

(I do see something that looks like potentially some valid TLV at `17 24 02 00  24 03 1d 24 04 01 18` but that's a List, and I would not expect to see any List in a WriteRequest.)

Sorry to keep pressing on this, but I don't want to waste either my time or yours here, so it would really help if we were clear about which payloads are being observed at what level in the stack.

Specifically: is the above the payload that is passed to `InteractionModelEngine::OnMessageReceived` as `aPayload` along with a `PayloadHeader` saying this is a WriteRequest?  Or is this specific dequence of bytes being injected at a different level?

bzbarsky-apple, 2025-01-15:
In particular:

> This is a Matter write request message where: fb 4a 35 indicates a write request header

is not making sense to me so far.  A WriteRequestMessage would be identified by an 06 for the "Protocol Opcode", and the actual payload as I said would start with 0x15....

agatah2333, 2025-01-15:
I print the message when input the crash file:

```cpp
else if (aPayloadHeader.HasMessageType(Protocols::InteractionModel::MsgType::WriteRequest))
{
    uint16_t dataLen = aPayload->DataLength();  
    const uint8_t * data = aPayload->Start();   
    
    ChipLogProgress(InteractionModel, " Payload data:");  
    for (uint16_t i = 0; i < dataLen; ++i)
    {
        ChipLogProgress(InteractionModel, "0x%02X ", data[i]);
    }
    status = OnWriteRequest(apExchangeContext, aPayloadHeader, std::move(aPayload), /* aIsTimedWrite = */ false);
}
```

Then I got (payload length 190):
```
15 29 01 28 00 36 02 15 24 00 4A 37 01 24 01 35 24 02 00 24 03 00 24 04 00 18 35 02 29 01 18 18 
15 24 00 02 37 01 24 01 00 24 02 00 24 03 00 24 04 00 18 35 02 28 01 18 18 
15 24 00 00 37 01 24 01 00 24 02 00 24 03 00 24 04 00 18 35 02 28 01 18 18 
15 24 00 00 37 01 24 01 00 24 02 00 24 03 00 24 04 00 18 35 02 28 01 18 18 
15 24 00 00 37 01 24 01 00 24 02 00 24 03 00 24 04 02 18 35 02 2C 01 04 2F 69 6D 2F 18 18 
15 24 00 00 37 01 24 01 17 24 02 24 24 03 02 24 04 00 18 35 02 29 01 18 18 
15 24 00 01 37 01 24 01 18 24 02 18 24 03 29 24 04 03 18 35 02 18 18 18 24 FF 0B 18
```

bzbarsky-apple, 2025-01-15:
OK, that looks like a TLV payload!  Let me dig into this a bit, thank you!

bzbarsky-apple, 2025-01-16:
@agatah2333 OK, on current tip if I try that payload I do not get any AddressSanitizer errors, and `ExchangeContext::Abort` is _not_ called from `ExchangeHolder::Release`, as expected from code inspection.  This is on SHA 5bd63d5441df6e5edf3792687acdd9bc7a7f507.

Bu looking at your `crash.txt` it looks like this is not in fact being called via the normal message-delivery codepath, but is being called from a fuzzer which is not in-tree, right?  How does this fuzzer set up the exchange?  Because the exchange is expected to keep a reference to itself across its call out to the delegate's `OnMessageReceived`, and this exchange very clearly is not, which is why you are getting a use-after-free....  It also looks like the fuzzer may be putting the exchange in an odd state in terms of its state bits, but it's hard to tell for sure.

Note that the exchange touches its own members after calling `OnMessageReceived`, so if it did not keep a ref to itself it would have use-after-free at _that_ point.  Which is why it holds that reference.

agatah2333, 2025-01-16:
Thanks for the explanation! You're right about the self-reference mechanism. 

I'm directly invoking `OnMessageReceived` because I'm trying to do unit testing to find functional bugs in the message handling logic itself, rather than testing the entire message delivery path.

However, I understand now that bypassing the normal message delivery path means I'm also bypassing the self-reference protection mechanism. Would you suggest a better way to unit test the message handling logic while still maintaining proper object lifetime management?

Current test code:
```cpp
auto exchangeContext = Server::GetInstance().GetExchangeManager().NewContext(
    gMySession,             
    &delegate,             
    /* isInitiator = */ true
);

if (exchangeContext != nullptr) {
    chip::app::InteractionModelEngine::GetInstance()->ProcessMessage(
        exchangeContext, payloadHeader, std::move(payload)
    );
}

bzbarsky-apple, 2025-01-16:
> Current test code:

OK, so you have a few bugs there:

1. isInitiator should be false, if you are _receiving_ the message, no?  That explains why your exchange flags are wrong.
2. The delegate on exchange creation should be gotten via calling OnUnsolicitedMessageReceived on the IM engine (though maybe it is, but that's just not in the paste context?).
3. You should be putting `ExchangeHandle ref(*exchangeContext)` on the stack before the `ProcessMessage` call to replicate the ownership semantics exchange will use on message received.

agatah2333, 2025-01-17:
Thanks! I revised my code by:
1. Setting isInitiator to false since we're simulating message reception
2. Adding ExchangeHandle ref(*exchangeContext) to maintain proper ownership semantics

After running with the payload, so far there is no error generated! Thanks for helping identify the root cause related to exchange state management and lifecycle.

closed 2025-01-17 by agatah2333