project-chip/connectedhomeip #37044: emberAfAttributeChanged callback invoked twice for endpoint 0 's PartsList when endpoint is added

chialunl, 2025-01-11:
When adding an endpoint, emberAfEndpointEnableDisable will be called twice, Once https://github.com/project-chip/connectedhomeip/blob/d5e98293c47c52aa306cf0be766e0b89772db72d/src/app/util/attribute-storage.cpp#L939 and another time https://github.com/project-chip/connectedhomeip/blob/d5e98293c47c52aa306cf0be766e0b89772db72d/src/app/util/attribute-storage.cpp#L950

bzbarsky-apple, 2025-01-13:
Why is this a problem @chialunl?  emberAfAttributeChanged just marks the attribute dirty.  That's an idempotent operation, so calling it multiple times in a row for the same attribute is fine.

bzbarsky-apple, 2025-01-13:
But also, `parentEndpointId` shouldn't ever be 0 in that loop, I would think (unless someone has mis-configured their endpoints: parentEndpointId is for listing the parent endpoint _if it's not the root node endpoint_), so this shouldn't be an issue in practice anyway.