project-chip/connectedhomeip #37072: [Platform] Cannot stop the event loop task correctly

sbh82, 2025-01-15:
### Reproduction steps

For my application I need to stop and restart Matter multiple times, but I can't figure out the right way to do it.
I can't find in the documentation the correct way to make a clean stop of Matter and then a next restart.

Now I'm trying to stop the PlatformManager using:
`chip::DeviceLayer::PlatformMgr().Shutdown();`
and after that call the event loop task is still running. Then if I try to restart it using:
`chip::DeviceLayer::PlatformMgr().InitChipStack();`
I see inside the method it sets a boolean flag to false ( `mShouldRunEventLoop.store(false);`) causing the while inside the event loop task to exit and the ESP32 crashes because there is not a call to `vTaskDelete` at the end of task:
```
template <class ImplClass>
void GenericPlatformManagerImpl_FreeRTOS<ImplClass>::EventLoopTaskMain(void * arg)
{
    ChipLogDetail(DeviceLayer, "CHIP event task running");
    static_cast<GenericPlatformManagerImpl_FreeRTOS<ImplClass> *>(arg)->Impl()->RunEventLoop();
    // TODO: At this point, should we not
    // vTaskDelete(static_cast<GenericPlatformManagerImpl_FreeRTOS<ImplClass> *>(arg)->mEventLoopTask)?
    // Or somehow get our caller to do it once this thread is joined?
}
```
I see that comment and I think the `vTaskDelete` is missing there.

The same problem occurs if I try to simply stop the event loop task using `StopEventLoopTask` method.

The issue #20185 seems to be related to a similar problem.
The last commit of this piece of code was a long time ago (27f9fb49ac45cb3933d96ddf7b25aed6ec9b665d) from @bzbarsky-apple 

### Platform

esp32

### Platform Version(s)

ESP IDF 5.3.1, connectedhomeip commit 43aa98c2

### Type

Manually tested with SDK

bzbarsky-apple, 2025-01-15:
The way this is meant to work in terms of the higher-level API is that the following sequence of calls should be repeatable indefinitely on the PlatformManager:

1. `InitChipStack()`
2. `StartEventLoopTask()` or `RunEventLoop()`
3. `StopEventLoopTask()` (and either wait for it to return or wait for `RunEventLoop()` to return, depending on which option you picked in step 2)
4. `Shutdown()`

If this does not work on some platform, that platform needs fixing.  At some point this did work on esp32, at least the way the tests use it.

For your specific use case, @sbh82 are you using StartEventLoopTask() or RunEventLoop()?  How are you using StopEventLoopTask(), exactly?   What actually causes the crash?  @shubhamdp @dhrishi

sbh82, 2025-01-15:
I tried to stop it using:
```
chip::DeviceLayer::PlatformMgr().StopEventLoopTask();
chip::DeviceLayer::PlatformMgr().Shutdown();
```
but it crashes because in freeRTOS implementation the `vTaskDelete` call is missing as reported previously.
In freeRTOS a task cannot exit without calling `vTaskDelete`

bzbarsky-apple, 2025-01-15:
> I tried to stop it using:

That's after calling `StartEventLoopTask()` or calling `RunEventLoop()`?  It's possible that currently only the `RunEventLoop()` version is exercised in CI...

jadhavrohit924, 2025-01-16:
Hi @sbh82, we are adding Matter shutdown flow for ESP in PR #36972. We will add the required changes to the same PR.

jadhavrohit924, 2025-01-16:
>That's after calling StartEventLoopTask() or calling RunEventLoop()? It's possible that currently only the RunEventLoop() version is exercised in CI...

@bzbarsky-apple ESP uses `StartEventLoopTask()` [see](https://github.com/project-chip/connectedhomeip/blob/master/examples/platform/esp32/common/CHIPDeviceManager.cpp#L57-L87).

sbh82, 2025-01-16:
> > I tried to stop it using:
> 
> That's after calling `StartEventLoopTask()` or calling `RunEventLoop()`? It's possible that currently only the `RunEventLoop()` version is exercised in CI...

To start Matter I simply use the Espressif API `esp_matter::start(...)` that internally calls several other methods and also:
```
PlatformMgr().InitChipStack()
PlatformMgr().StartEventLoopTask()
```

> Hi [@sbh82](https://github.com/sbh82), we are adding Matter shutdown flow for ESP in PR [#36972](https://github.com/project-chip/connectedhomeip/pull/36972). We will add the required changes to the same PR.

@jadhavrohit924 This is a great news! This is exactly what I needed. With these changes will I be able to completely close Matter and free up all memory resources? Can you estimate when this PR will be ready and merged to be released?

jadhavrohit924, 2025-01-17:
@sbh82 We are planning to complete the PR by next week.