project-chip/connectedhomeip #37139: Convert CommandHandlerInterface command listing to provide full data

andy31415, 2025-01-21:
Currently CHI only exposes an "enumerate" command for command IDs (like https://github.com/project-chip/connectedhomeip/blob/master/src/app/CommandHandlerInterface.h#L125)

However this is insufficient as commands have attributes for invoke permissions (several existing ones should be admin) and timed-invoke flags and similar. We need CHI to take full control and return the entire set of "command id + flags + invoke permissions" like defined in https://github.com/project-chip/connectedhomeip/blob/master/src/app/data-model-provider/MetadataTypes.h#L116

This should probably wait for #37127 to be finalized, so that the same structures/builders can be re-used. 

Since this will be an API change, the update should be documented in https://github.com/project-chip/connectedhomeip/blob/master/docs/upgrading.md

khodya, 2025-02-14:
CodegenDataModelProvide.cpp
CommandHandlerInterface
AccepteCommandEntotyFor
there are some clusters that have this implementation already

khodya, 2025-02-14:
lines 387 - 410 should be replaced by 2 lines

ratgr, 2025-03-03:
I will split the problem into 2 pieces, the generation of the Metadata and a proposed better Interface for returning these values,
my reasoning is the following, if we are generating the metadata in its semi-final form, it makes no sense
generating dynamically allocation of it. And the second is how to gnerate the metadata itself

## Metadata Generation

Ideally it replaces current info objects so that we don't duplicate information and 
Possible con, Will need to touch a lot of functions to use this new metadata instead of the old one

``` cpp

// Appending to Zapdata will mean generating this structures dynamically in RAM,
// maybe not allocating but definitely not on Flash
// I believe is better to provide this arrays directly as we can avoid allocation if we do
namespace chip::app::Clusters::NetworkCommissioning{
    namespace Command::Entries {
        constexpr DataModel::AcceptedCommandEntry AcceptedCommandEntries[] = {
            /*ID,       BitFlags<CommandQualityFlags>,     Privilege*/
            { ScanNetworks::Id,             {}, Access::Privilege::kAdminister },   //
            { RemoveNetwork::Id,            {}, Access::Privilege::kAdminister },  //
            { ConnectNetwork::Id,           {}, Access::Privilege::kAdminister }, //
            { ReorderNetwork::Id,           {}, Access::Privilege::kAdminister }, //
            { AddOrUpdateThreadNetwork::Id, {}, Access::Privilege::kAdminister},
            { AddOrUpdateWiFiNetwork::Id,   {}, Access::Privilege::kAdminister},
            { QueryIdentity::Id,            {}, Access::Privilege::kAdminister }
        };

        constexpr DataModel::CommandId GeneratedCommandEntries[] = {
            ScanNetworksResponse::Id,   //
            NetworkConfigResponse::Id,  //
            ConnectNetworkResponse::Id, //
            QueryIdentityResponse::Id   //
        };
    }
    namespace Attribute::Entries{

        using L = AttributeQualityFlags::kListAttribute;
        using F = AttributeQualityFlags::kFabricScoped;
        using S = AttributeQualityFlags::kFabricSensitive;
        using C = AttributeQualityFlags::kChangesOmitted;
        using T = AttributeQualityFlags::kTimed;

        constexpr DataModel::AttributeEntry AttributeEntries[] = {
            /* ID,  BitFlags<AttributeQualityFlags>, RPrivilege, WPrivilege */
            { MaxNetworks::Id,            {F}, {A}, std::nullopt },   
            { Networks::Id,               {L}, {A}, std::nullopt },   
            { ScanMaxTimeSeconds::Id,     {F}, {A}, std::nullopt },   
            { ConnectMaxTimeSeconds::Id,  {F}, {A}, std::nullopt },   
            { InterfaceEnabled::Id,       {},  {V, A}, {V, A}    },   
            { LastNetworkingStatus::Id,   {},  {A}, std::nullopt },   
            { LastNetworkID::Id,          {},  {A}, std::nullopt },   
            { LastConnectErrorValue::Id,  {},  {A}, std::nullopt },   
            { SupportedWiFiBands::Id,     {F}, {V}, std::nullopt },   
            { SupportedThreadFeatures::Id,{F}, {V}, std::nullopt },   
            { ThreadVersion::Id,          {F}, {V}, std::nullopt },   
        };
    }
}

```

Using the info from access.h from all-clusters-app

Actual Solution (ZAP):
    Attributes:  READ (32,32,8): 8  Write (32,32,8): 59 = 4824 bits = 603 bytes // Without Flags!
    Commands:    Privilege (32, 32, 8): 48  = 432 bytes

New Solution (this):
    Attributes:  61 (32, 32, 8, 8) = 610 bytes  With all Flags
    Commands:    48 (32, 32, 8, 8) = 480 bytes 
[... 150 lines cut ...]
    uint64_t buffer; // This might be a ptr if we have more than 127 elements
    
  
};

CHIP_ERROR Instance::EnumerateAcceptedCommands(const ConcreteClusterPath & cluster, FilterSpan<AcceptedCommandEntry>& entries ) {
    // Assuming flags cannot change while running, else remove the const and assign every time

    bool hasThread = mFeatureFlags.Has(Feature::kThreadNetworkInterface);
    bool hasWifi   = mFeatureFlags.Has(Feature::kWiFiNetworkInterface);

    static auto filter = GetEnabledArray(Entries::AcceptedCommandEntries, {
        hasThread || hasWifi, {ScanNetworks::Id, RemoveNetwork::Id, ConnectNetwork::Id, ReorderNetwork::Id}
        hasThread, {}
        hasWifi, {}
    });
    
    entries = {Entries::AcceptedCommandEntries, filter};
    
    return CHIP_NO_ERROR;
}

```

I would prefer a solution like this, it is a little bit more of code but the object makes sure that the lifetime of the filter span is enough and hides the complexity of the bitflags instead of bools, it also never allocates unless more than 127 elements are in the filter

andy31415, 2025-03-03:
@ratgr could you provide the alternative when we do not generate a `includes everything` list? Just the entries?

ratgr, 2025-03-03:
From discussion, I'm going to do the following changes:

- [ ] Implement prototype using a retrocompatible SHIM to compare
     - "No Code Change" SHIM
     -  "CALL ME" SHIM 
- [ ] CodeGenerator for Command::Entry's
- Final Change

andy31415, 2025-03-27:
created #38151 to provide metadata generation that is based on our XMLs (which should be based on spec).
If that is merged, we can either use existing constants or define additional constants (like some "all possible command or attribute entries") to use within the code.

PR #39198 (2025-08-11): CHI updates with Update Shim 
## CHI updates with Update Shim 

Fixes: #37139

## Code TLDR

This code adds 3 big pieces of import to CHIP

### Command Handler Interface Shim

On this change Command Handler Interface is updated to use the new builder based approach and a SHIM is provided to ease into updating

```cpp
template <ClusterId... TClusterIds>
class CommandHandlerInterfaceShim : public CommandHandlerInterface;

```
CommandHandlerInterfaceShim provides an interface compatible with old code, the templated ClusterIds are used for generation of a CompileTimeCalculated Switch statement wit [... cut]
closed 2025-08-11 by mergify[bot]