project-chip/connectedhomeip #37154: [CERT-TEST-FAILURE] [TC-SC-4.3] Test Failure

Ashwinigrl, 2025-01-22:
### Feature Area

Other

### Test Case

TC-SC-4.3

### Reproduction steps

**Command used to launch the DUT :** `./lit-icd-app`
**Command used to run the python script :** `python3 TC_SC_4_3.py --commissioning-method on-network --discriminator 3840 --passcode 20202021 --storage-path admin_storage.json `

While validating the **TC-SC-4.3** python script, Test case is failing at test step 10 `TH performs a DNS-SD browse for _I<hhhh>._sub._matter._tcp.local, where <hhhh> is the 64-bit compressed Fabric identifier, expressed as a fixed-length, sixteen-character hexadecimal string, encoded as ASCII (UTF-8) text using capital letters.` with the below errors.

>  File "/home/ubuntu/connectedhomeip/src/python_testing/TC_SC_4_3.py", line 388, in test_TC_SC_4_3
>     asserts.assert_in(op_sub_type, service_types, f"No PTR record with DNS-SD instance name '{op_sub_type}' wsa found.")
>   File "/home/ubuntu/connectedhomeip/no/lib/python3.12/site-packages/mobly/asserts.py", line 141, in assert_in
>     _call_unittest_assertion(
>   File "/home/ubuntu/connectedhomeip/no/lib/python3.12/site-packages/mobly/asserts.py", line 50, in _call_unittest_assertion
>     raise signals.TestFailure(my_msg, extras=extras)
> mobly.signals.TestFailure: Details='_I77C0200B00B14E37._sub._matter._tcp.local.' not found in ['_matter._tcp.local.', '_matterd._udp.local.'] No PTR record with DNS-SD instance name '_I77C0200B00B14E37._sub._matter._tcp.local.' wsa found., Extras=None
> [MatterTest] 01-22 06:13:51.985 INFO 		***** Test Failure : 
> [MatterTest] 01-22 06:13:51.985 INFO Finished test in 10580893.0ms
> [MatterTest] 01-22 06:13:51.986 ERROR 
> 
> * Test test_TC_SC_4_3 failed for the following reason:
> * Details='_I77C0200B00B14E37._sub._matter._tcp.local.' not found in ['_matter._tcp.local.', '_matterd._udp.local.'] No PTR record with DNS-SD instance name '_I77C0200B00B14E37._sub._matter._tcp.local.' wsa found., Extras=None
> *
> * File "/home/ubuntu/connectedhomeip/src/python_testing/TC_SC_4_3.py", line 388, in test_TC_SC_4_3
> * asserts.assert_in(op_sub_type, service_types, f"No PTR record with DNS-SD instance name '{op_sub_type}' wsa found.")
> *
> * Test step:
> * 10: TH performs a DNS-SD browse for _I<hhhh>._sub._matter._tcp.local, where <hhhh> is the 64-bit compressed Fabric identifier, expressed as a fixed-length, sixteencharacter hexadecimal string, encoded as ASCII (UTF-8) text using capital letters.	Expected outcome: Verify DUT returns a PTR record with DNS-SD instance name set to instance_name
> * Endpoint: None
> [MatterTest] 01-22 06:13:51.986 INFO [Test] test_TC_SC_4_3 FAIL

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

d5e98293c47c52aa306cf0be766e0b89772db72d

### Platform

raspi

### Anything else?

**Sample app used to run the DUT:**
lit-icd-app 

**Test step that require Review:** 
step 10

**Reference Logs:**

[TC-SC-4.3(CNTRL).txt](https://github.com/user-attachments/files/18500932/TC-SC-4.3.CNTRL.txt)
[TC-SC-4.3(DUT).txt](https://github.com/user-attachments/files/18500933/TC-SC-4.3.DUT.txt)

**Python Script References:**
- https://github.com/project-chip/connectedhomeip/blob/master/src/python_testing/TC_SC_4_3.py

cecille, 2025-01-24:
@Ashwinigrl - could you please do a browse on avahi to see if the device is actually advertising that prefix correctly?

Ashwinigrl, 2025-01-31:
@cecille We excuted test step 10 manually using the command  `avahi-browse _IE13B6A3144440C5E._sub._matter._tcp.local.` , (Here <E2CE38EEE4A7A8B0> is a compressed Fabric id.). We got the below response 

> ubuntu@ubuntu:~/connectedhomeip/out/chip-tool$ avahi-browse -r _IE13B6A3144440C5E._sub._matter._tcp
> +   eth0 IPv6 E13B6A3144440C5E-0000000000000001             _matter._tcp         local
> =   eth0 IPv6 E13B6A3144440C5E-0000000000000001             _matter._tcp         local
>    hostname = [E45F010F27D90000.local]
>    address = [fe80::e65f:1ff:fe0f:27d8]
>    port = [5540]
>    txt = ["ICD=0" "T=6" "SAT=5000" "SAI=500" "SII=5500"]