project-chip/connectedhomeip #37171: Subcription failed with chip.exceptions.ChipStackError: IM Error 0x00000580: General error: 0x80 (INVALID_ACTION) .

Saravana-kr22, 2025-01-23:
**Description**
When executing the script TC_Darwin_pair.py as a part of stress testing, in the CASE session when sending wildcard subscription using 2 controller there is a exception being thrown which is  chip.exceptions.ChipStackError: IM Error 0x00000580: General error: 0x80 (INVALID_ACTION) . This script will mimic on a high level how the pairing occurs in the Darwin ecosystem. Here there are two fabrics is created and wildcard subcription is performed using both the controllers. But this INVALID_ACTION error occured only on the 2nd controller. 

GitHub hash of the SDK that were being used are : 36f81cda5e8a1def9d727c9611ea43ba6d997f1a

**Platforms**
I Have performed the execution on Raspberrypi using on-network  provisioning
**Test Bed :**
Controller Platform - python controller running on Raspberrypi 4B -8GB,
RPI as DUT - all-cluster-app running  all-cluster-app running on Raspberrypi 4B -8GB
Provisioning method - On-network

**Test scenario**
Pair/Provision the DUT (All-cluster-app running on raspi) as per drawiin pairing flow using On-network.
Read device information after the device is provisioned
sleep for 2 secs
Unpair the DUT (i.e. removing the fabric)
Factory reset the DUT
repeat step1 - 6 for 500 iterations
Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Darwin_Pair.py

**Execution Details**

In this Test bed for Iteration [ 138, 139, 140, 141]  second controller is failed to perform wildcard subscription 
Iterations scheduled to run: 200
Iterations Failed: 7
Iterations Passed: 193
Execution Logs (logs consists of 1 Pass iteration:142  and 4 Fail Iterations): 

[138_failed.zip](https://github.com/user-attachments/files/18519375/138_failed.zip)
[139_failed.zip](https://github.com/user-attachments/files/18519380/139_failed.zip)
[140_failed.zip](https://github.com/user-attachments/files/18519385/140_failed.zip)
[141_failed.zip](https://github.com/user-attachments/files/18519396/141_failed.zip)
[142_passed.zip](https://github.com/user-attachments/files/18519402/142_passed.zip)

bzbarsky-apple, 2025-01-23:
From [138_failed.zip](https://github.com/user-attachments/files/18519375/138_failed.zip) from the DUT log, here's the relevant bit:

```
[1737585692.750899][29350:29350] CHIP:EM: >>> [E:40690r S:22244 M:49634145] (S) Msg RX from 2:000000000001B66A [99AB] to 0000000012344322 --- Type 0001:03 (IM:SubscribeRequest) (B:51)
...
[1737585718.960936][29350:29350] CHIP:EM: <<< [E:40690r S:22244 M:198522001 (Ack:49634217)] (S) Msg TX from 0000000012344322 to 2:000000000001B66A [99AB] [UDP:[fe80::3c73:93e0:a746:d2c5%wlan0]:43012] --- Type 0001:05 (IM:ReportData) (B:1184)
...
[1737585718.963309][29350:29350] CHIP:EM: >>> [E:40691r S:22244 M:49634218] (S) Msg RX from 2:000000000001B66A [99AB] to 0000000012344322 --- Type 0001:03 (IM:SubscribeRequest) (B:51)
...
[1737585718.964428][29350:29350] CHIP:IM: Deleting previous subscription from NodeId: 000000000001B66A, FabricIndex: 2
...
[1737585720.611132][29350:29350] CHIP:EM: >>> [E:40690r S:22244 M:49634221 (Ack:198522001)] (S) Msg RX from 2:000000000001B66A [99AB] to 0000000012344322 --- Type 0001:01 (IM:StatusResponse) (B:42)
...
[1737585720.611341][29350:29350] CHIP:IM: Msg type 1 not supported
...
[1737585720.613503][29350:29350] CHIP:EM: <<< [E:40690r S:22244 M:198522003 (Ack:49634221)] (S) Msg TX from 0000000012344322 to 2:000000000001B66A [99AB] [UDP:[fe80::3c73:93e0:a746:d2c5%wlan0]:43012] --- Type 0001:01 (IM:StatusResponse) (B:42)
```

So we're in the middle of a priming read for that first SubscribeRequest; the DUT sent a ReportData and are waiting for the StatusResponse.  Then another SubscribeRequest from the same node comes in, with `keepSubscriptions` set to `false`, so that in-progress subscription is torn down.  When the StatusResponse then comes in, the DUT is not expecting it anymore, so it responds INVALID_ACTION.  Then the client tears down _its_ side of the first subscription and reports the error.

But the fundamental problem is the client doing the new subscribe with `keepSubscriptions` set to false before it's done with the previous subscription.

OK, so why is the controller doing that second subscribe?  From the controller side of the log there:

```
2025-01-22 22:40:38,364 - base_tc - INFO - matter_qa_base_test_class.py:_create_iteration_log_file:221 - Started Iteration 138
...
2025-01-22 22:40:41,172 - chip.native.DMG - INFO - __init__.py:_RedirectToPythonLogging:41 - OnResubscribeTimerCallback: ForceCASE = 1
```

Looks to me like there is state left over from previous iterations that then messes with this one.  So basically: the test script is buggy.

Saravana-kr22, 2025-01-28:
@bzbarsky-apple  In this Darwin script we have cleanup function at the end of each iteration, where we Shutdown the subscriptions of the particular iteration.

In the lastest run of the darwin pairing script with the below setup: 
**Platforms**
I Have performed the execution on Raspberrypi as DUT using on-network provisioning
**Test Bed :**
Controller Platform - python controller running on Mac Laptop,
RPI as DUT - all-cluster-app running all-cluster-app running on Raspberrypi 4B -8GB
Provisioning method - On-network

Here i faced the same Invalid action error while the TH2 subscription on iteration [77,79]. Here the 78 iteration is passed but in the controller log i can see the callback of the subcription still this iteration is passed without any issues.

```
2025-01-28 02:28:34,914 - base_tc - INFO - matter_qa_base_test_class.py:_create_iteration_log_file:221 - Started Iteration 78
...
2025-01-28 02:28:36,210 - chip.native.DMG - INFO - __init__.py:_RedirectToPythonLogging:41 - OnResubscribeTimerCallback: ForceCASE = 1
```

So i have a doubt if  the left over subcription of the previous iteration cause this issue why the 78 here with the same is not failed.

 **Execution Logs** : (logs consists of 3 Pass iteration:76,78,80 and 2 Fail Iterations: 77, 79):

[76_passed.zip](https://github.com/user-attachments/files/18573579/76_passed.zip)
[77_failed.zip](https://github.com/user-attachments/files/18573583/77_failed.zip)
[78_passed.zip](https://github.com/user-attachments/files/18573591/78_passed.zip)
[79_failed.zip](https://github.com/user-attachments/files/18573599/79_failed.zip)
[80_passed.zip](https://github.com/user-attachments/files/18573604/80_passed.zip)

**Note:** In this execution `keepSubscriptions` is set to True.

bzbarsky-apple, 2025-01-30:
> where we Shutdown the subscriptions of the particular iteration.

Where is this happening, exactly?  I don't see anything in the script that would do that.  There are sessions being closed, there are DUTs being unpaired, but absolutely nothing that tears down subscriptions that I see.

bzbarsky-apple, 2025-01-30:
@Saravana-kr22 See above.

bzbarsky-apple, 2025-01-30:
Also: does the test script mean to create subscriptions with auto-resubscribe enabled?  If not, maybe it should stop doing that.

bzbarsky-apple, 2025-01-30:
> Note: In this execution keepSubscriptions is set to True.

Not what the DUT claims.  From the [77_failed.zip](https://github.com/user-attachments/files/18573583/77_failed.zip) log:

```
[1738011489.322006][443230:443230] CHIP:DMG: SubscribeRequestMessage =
[1738011489.322061][443230:443230] CHIP:DMG: {
[1738011489.322112][443230:443230] CHIP:DMG: 	KeepSubscriptions = false, 
```

Saravana-kr22, 2025-02-20:
Currently we are  not seeing this issue in current execution after modifing the  KeepSubscriptions = True when we find it we will upload the log  ASAP

Saravana-kr22, 2025-04-30:
This issue is not  observed in recent stress runs after the modification ` KeepSubscriptions = True`. We will proceed to close this ticket for now. So will proceed with closing this issue for now.
If the issue resurfaces in future runs, It can reopen  for further investigation.

closed 2025-04-30 by Saravana-kr22