project-chip/connectedhomeip #37172: Getting UnKnownError as status for Connect Network command.

Saravana-kr22, 2025-01-23:
**Description**
When executing the script TC_Android_pair.py as a part of stress testing, in the Pase session when sending connect network command  we got the respose with status = 12 (UnKnowError) with error value =  -3  as below. This script will mimic on a high level how the pairing occurs in the Android ecosystem.

> 2025-01-17 20:38:04,793 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - InvokeResponseMessage =
2025-01-17 20:38:04,794 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - {
2025-01-17 20:38:04,794 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 	suppressResponse = false, 
2025-01-17 20:38:04,795 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 	InvokeResponseIBs =
2025-01-17 20:38:04,795 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 	[
2025-01-17 20:38:04,796 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 		InvokeResponseIB =
2025-01-17 20:38:04,796 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 		{
2025-01-17 20:38:04,796 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 			CommandDataIB =
2025-01-17 20:38:04,797 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 			{
2025-01-17 20:38:04,797 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 				CommandPathIB =
2025-01-17 20:38:04,797 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 				{
2025-01-17 20:38:04,797 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 					EndpointId = 0x0,
2025-01-17 20:38:04,797 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 					ClusterId = 0x31,
2025-01-17 20:38:04,797 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 					CommandId = 0x7,
2025-01-17 20:38:04,798 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 				},
2025-01-17 20:38:04,798 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 				
2025-01-17 20:38:04,798 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 				CommandFields = 
2025-01-17 20:38:04,798 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 				{
2025-01-17 20:38:04,798 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 					0x0 = 12 (unsigned), 
2025-01-17 20:38:04,798 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 					0x2 = -3 (signed), 
2025-01-17 20:38:04,798 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 				},
2025-01-17 20:38:04,799 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 			},
2025-01-17 20:38:04,799 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 			
2025-01-17 20:38:04,799 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 		},
2025-01-17 20:38:04,799 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 		
2025-01-17 20:38:04,799 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 	],
2025-01-17 20:38:04,799 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 	
2025-01-17 20:38:04,799 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - 	InteractionModelRevision = 11
2025-01-17 20:38:04,799 - chip.native.DMG - DEBUG - __init__.py:_RedirectToPythonLogging:43 - },

GitHub hash of the SDK that were being used are : d96f2b5d7a1e3a88243b02075e6c7ac3a0275938

**Platforms**
I Have performed the execution on Raspberrypi as DUT using ble-wifi  provisioning
**Test Bed :**
Controller Platform - python controller running on Mac Laptop,
Raspberrypi as DUT - all-cluster-app running on Raspberrypi 4B -8GB
Provisioning method - Ble-Wifi

**Test scenario**
Pair/Provision the DUT (All-cluster-app running on raspi) as per drawiin pairing flow using Ble-Wifi.
Read device information after the device is provisioned
sleep for 2 secs
Unpair the DUT (i.e. removing the fabric)
Factory reset the DUT
repeat step1 - 6 for 500 iterations
Script reference: https://github.com/CHIP-Specifications/matter-qa/blob/main/src/matter_qa/scripts/reliability_scripts/TC_Android_Pair.py

**Execution Details**

In this Test bed for Iteration [ 146, 211, 225, 258, 295, 342, 415, 422, 433, 480, 485]  has the  UnKnowError status for the connect network command
Iterations scheduled to run: 500
Iterations Failed: 15
Iterations Passed: 485
Execution Logs (logs consists of 2 Pass iterations  and 11 Fail Iterations): 

[145_passed.zip](https://github.com/user-attachments/files/18520055/145_passed.zip)
[146_Failed.zip](https://github.com/user-attachments/files/18520051/146_Failed.zip)
[147_passed.zip](https://github.com/user-attachments/files/18520059/147_passed.zip)
[211_failed.zip](https://github.com/user-attachments/files/18520050/211_failed.zip)
[225_failed.zip](https://github.com/user-attachments/files/18520049/225_failed.zip)
[258_Failed.zip](https://github.com/user-attachments/files/18520060/258_Failed.zip)
[295_failed.zip](https://github.com/user-attachments/files/18520058/295_failed.zip)
[342_failed.zip](https://github.com/user-attachments/files/18520052/342_failed.zip)
[415_failed.zip](https://github.com/user-attachments/files/18520054/415_failed.zip)
[422_failed.zip](https://github.com/user-attachments/files/18520048/422_failed.zip)
[433_failed.zip](https://github.com/user-attachments/files/18520057/433_failed.zip)
[480_failed.zip](https://github.com/user-attachments/files/18520056/480_failed.zip)
[485_failed.zip](https://github.com/user-attachments/files/18520053/485_failed.zip)

cecille, 2025-04-29:
So the steps here look like the test is doing adn AddOrUpdate (cmd 2), then a ConnectNetwork (cmd 6) and it's connecting to MatterAutomation. 

The connect call prompts the supplicant to do a scan then attempt an association.
In the pass case, this works (145):
```
[1737121611.945399][835475:835476] CHIP:DL: wpa_supplicant:PropertiesChanged:key:State -> 'associating'
[1737121612.043010][835475:835476] CHIP:DL: wpa_supplicant:PropertiesChanged:key:State -> 'completed'
[1737121612.043184][835475:835476] CHIP:DL: wpa_supplicant:PropertiesChanged:key:CurrentBSS -> objectpath '/fi/w1/wpa_supplicant1/Interfaces/0/BSSs/14047'
[1737121612.043264][835475:835476] CHIP:DL: wpa_supplicant:PropertiesChanged:key:CurrentAuthMode -> 'WPA2-PSK+WPA-PSK'
[1737121612.043396][835475:835475] CHIP:DMG: Decreasing reference count for CommandHandlerImpl, remaining 1
```

In the failure case, it doesn't (146):
```
[1737121661.086544][835545:835546] CHIP:DL: wpa_supplicant:PropertiesChanged:key:State -> 'associating'
[1737121661.292380][835545:835546] CHIP:DL: wpa_supplicant:PropertiesChanged:key:AssocStatusCode -> 16
[1737121661.300142][835545:835546] CHIP:DL: wpa_supplicant:PropertiesChanged:key:State -> 'disconnected'
[1737121661.300251][835545:835546] CHIP:ZCL: WiFiDiagnosticsDelegate: OnDisconnectionDetected
```
16 = Authentication rejected due to timeout waiting for next frame in sequence

That's fascinating because the supplicant also reports that the signal for this wifi is -16 (I'm assuming dBm since it's negative) which means these two devices must be living together in a isolation chamber. On mars. Next to a frictionless sphere.

```
[1737121660.880350][835545:835546] CHIP:DL: Network Found: MatterAutomation (0c:0e:76:4e:f3:38) Signal:-16
[1737121660.890268][835545:835546] CHIP:DL: Network Found: GRLPrivate (10:27:f5:37:4e:c6) Signal:-27
[1737121660.899602][835545:835546] CHIP:DL: Network Found: ChipMatter (0c:0e:76:4e:f1:c8) Signal:-34
```

TL;DR - it looks like there's something dodgy with the linux wifi supplicant.

Does this get used in real products.

cjandhyala, 2025-04-30:
@arkq Could you take a look and comment pls ?