project-chip/connectedhomeip #37233: Increase unit test code coverage of controller/ to 80%

HNaftalin, 2025-01-28:


andy31415, 2025-02-05:
Moving this a bit higher in priority - I expect controller to generally be VERY well covered in integration tests, however it also looks like we have several files with absolutely no coverage here where we could trivially add some. `CHIPDeviceController.h/cpp` is likely too complex to cover, however other things seem much simpler and could get coverage easily.

I suggest to try to get some coverage here and re-assess priority after that (we may not get to 80%, however we should be able to get a lot better)

andy31415, 2025-02-07:
link to current: https://matter-build-automation.ue.r.appspot.com/controller/index.html

khodya, 2025-02-07:
Break down:
CHIPDeviceController.cpp - could be tricky to cover
general purpose - check if it is feasible

cecille, 2025-02-07:
Note: we do have some integration tests of the controller and the state machine here: https://github.com/project-chip/connectedhomeip/blob/master/src/controller/python/OpCredsBinding.cpp

We cover some of the window manager behaviour in the certification tests as well.

khodya, 2025-02-13:
Part of the breakdown:
Define (understand) functionalities - what is unit-testable and what is actually unit-tested, what should be.

Elen777300, 2025-08-20:
Working on CommissioningDelegate.h

marybadalyan, 2025-08-26:
Added this [PR](https://github.com/project-chip/connectedhomeip/pull/40693) testing `NOCChainGenerated` with corrupted buffers and empty certificates. These failure paths are **not covered by the existing integration tests** [OpCredsBinding.cpp](https://github.com/project-chip/connectedhomeip/blob/master/src/controller/python/OpCredsBinding.cpp) and increased test coverage by \~0.2%.

**Notes for future:**
Watch out for types like `size_t` , `uint32_t`, `ptrdiff_t`, `intptr_t`, `unsigned long` , etc. when dealing with 32-bit systems that run in the CI.

Elen777300, 2025-09-10:
Addressing review comments on the `CommissioningDelegate` PR:
• Replaced constant→string checks with behavior-focused tests
• Corrected nonce (32B) and signature (64B) lengths; valid Wi-Fi passphrase
• Used realistic PAI/DAC buffers; round-trip assertions only

closed 2025-10-02 by mergify[bot]