project-chip/connectedhomeip #37241: Increase unit test code coverage of lib/dnssd/minimal_mdns/ to 80%

HNaftalin, 2025-01-28:


dmaranges-gd, 2025-05-22:
Currently, the actual coverage of that folder its around the 89%

dmaranges-gd, 2025-05-30:
Respect to that ticket, I checked with the information from Codecov on the PR #38342 and run it locally. All files are covered at least 80%, except the Server.cpp and Parcer.cpp files.

[image]

In general, lines that are not covered are referred to as error returns. Except for three methods on Server.cpp:

[image]

[image]

closed 2025-06-06 by dmaranges-gd