project-chip/connectedhomeip #37257: Ensure no double-work on ::Invoke processing in validating paths

andy31415, 2025-01-28:
See conversation in https://github.com/project-chip/connectedhomeip/pull/37207#discussion_r1931526570_ for reference

In general we currently have IME check that a command path is valid and then calling ::Invoke checks that the command path is valid again. This seems redundant and we can save a bit of resources (both code size and processing time) if we do not get IME to double-check paths and lets the provider handle things.

andy31415, 2025-01-30:
Redundancy move here may not be easy: IME needs attribute flags and this means the path validation is done (we need to traverse the tree anyway and cannot accept commands with unknown metadata).

jennygaz, 2025-09-04:
Currently changed the approach on how invoke is called to ensure it is done only once. However, while doing so, there's a failing test. Right now, I'm having issues compiling and running the tests, so I cannot access the text for the failing test. The failure for compiling, which I'm working on, throws the following:

```
connectedhomeip via 🐍 3.11.3+chromium.29 via pigweed-venv took 4.1s …
➜ ninja -C out/host          
ninja: Entering directory `out/host'
[0/1] Regenerating ninja files
[44/45] ACTION //src/controller/python:chip-repl(//build/toolchain/host:mac_arm64_gcc)
```

jennygaz, 2025-10-06:
Currently working on fixing CI/CD errors, last one was a shadowing variable but fixed it last Friday (03/10/2025) and trying to fix other CI/CD errors. The current approach modifies a bit how the way IME handles status so that it is possible to remember it and CodegenDataModel has new private functions that take advantage of this so that we can avoid checking command path existence or validity twice when calling them.