project-chip/connectedhomeip #37266: [BUG] Fix conformances on example app "lock"

HNaftalin, 2025-01-28:
### Reproduction steps
Need to find and address conformance issues on the example app "lock"
NOTE:  we should remove the door lock and media clusters completely to match the all-clusters

### Bug prevalence
always

### GitHub hash of the SDK that was being used
tot

### Platform
core

### Platform Version(s)
No response

### Anything else?
No response

gd-mauri, 2025-02-06:
This is the PR: https://github.com/project-chip/connectedhomeip/pull/37432

gd-mauri, 2025-02-10:
I see it got fixed in master.

running:

```
scripts/run_in_python_env.sh out/py './scripts/tests/run_python_test.py --app ./out/darwin-arm64-lock-no-ble/chip-lock-app --app-args "--trace-to json:log" --script src/python_testing/TC_DeviceBasicComposition.py --script-args "--qr-code MT:-24J0AFN00KA0648G00"'
```

[image]

closed 2025-02-11 by gd-mauri