project-chip/connectedhomeip #37297: [BUG] Fix conformances on example app "lit-icd-app"

HNaftalin, 2025-01-29:
### Reproduction steps
Need to find and address conformance issues on the example app "lit-icd-app"
NOTE:  we should remove the door lock and media clusters completely to match the all-clusters

### Bug prevalence
always

### GitHub hash of the SDK that was being used
tot

### Platform
core

### Platform Version(s)
No response

### Anything else?
No response

### References:
https://github.com/project-chip/connectedhomeip/issues/29812
https://github.com/project-chip/connectedhomeip/issues/37267

jennygaz, 2025-02-10:
- [ ] Endpoint: 0 - Cluster: 31 (0x1f) AccessControl - Problem: Attribute 0x01 is included, but is disallowed by conformance. Conformance: EXTS, implemented features: spec_location:
- [ ] 
Endpoint: 0 - Cluster: 49 (0x31) NetworkCommissioning - Problem: Attribute 0x02 is included, but is disallowed by conformance. Conformance: WI | TH, implemented features: ET spec_location:
Problem: ProblemSeverity.ERROR
test_name: IDM-10.2
location:
Endpoint: 0,
Cluster: 49 (0x31) NetworkCommissioning
Attribute:3 (0x03)
problem: Attribute 0x03 is included, but is disallowed by conformance. Conformance: WI | TH, implemented features: ET
spec_location:

Problem: ProblemSeverity.ERROR
test_name: IDM-10.3
location:
Endpoint: 0,
Cluster: 40 (0x28) BasicInformation
Attribute:65533 (0xfffd)
problem: Revision found on cluster (3) does not match revision listed in the spec (4)
spec_location:

Problem: ProblemSeverity.ERROR
test_name: IDM-10.3
location:
Endpoint: 0,
Cluster: 48 (0x30) GeneralCommissioning
INFO:chip.native.SC:SecureSession[0x7fbbbc00e570, LSID:14122]: State change 'kActive' --> 'kPendingEviction'
Attribute:65533 (0xfffd)
INFO:chip.native.FP:Forgetting fabric 0x1

Problem: ProblemSeverity.ERROR
INFO:chip.native.TS:Previous Last Known Good Time: 2023-10-14T01:16:48
test_name: IDM-10.3
location:
Endpoint: 0,
INFO:chip.native.TS:Reverted Last Known Good Time to previous value
ERROR:chip.native.CTL:Shutting down the stack...
Cluster: 53 (0x35) ThreadNetworkDiagnostics

Problem: ProblemSeverity.ERROR
test_name: IDM-10.3
location:
Endpoint: 1,
Cluster: 3 (0x03) Identify
Attribute:65533 (0xfffd)
problem: Revision found on cluster (4) does not match revision listed in the spec (5)
spec_location:

Problem: ProblemSeverity.ERROR
test_name: IDM-10.5
location:
DeviceType: 17
ClusterID: 47
problem: Mandatory cluster Power Source for device type Power Source is not present in the server list
spec_location:

Problem: ProblemSeverity.ERROR
test_name: IDM-10.5
location:
Endpoint: 0,
Cluster: 42 (0x2a) OtaSoftwareUpdateRequestor
problem: Extra cluster found on endpoint with device types [Descriptor.Structs.DeviceTypeStruct(deviceType=17, revision=1), Descriptor.Structs.DeviceTypeStruct(deviceType=22, revision=1)]
spec_location:

Problem: ProblemSeverity.ERROR
test_name: IDM-10.6
location:
Endpoint: 0,
Cluster: 29 (0x1d) Descriptor
problem: Expected Device type revision for device type 22 Root Node on endpoint 0 does not match revision on DUT. Expected: 3 DUT: 1
spec_location:

Problem: ProblemSeverity.ERROR
test_name: IDM-10.6
location:
Endpoint: 1,
ERROR:chip.native.-:src/messaging/ExchangeMgr.cpp:185: CHIP Error 0x00000007: No unsolicited message handler at src/protocols/bdx/BdxTransferServer.cpp:55
INFO:chip.native.FP:Shutting down FabricTable
Cluster: 29 (0x1d) Descriptor
INFO:chip.native.TS:Pending Last Known Good Time: 2023-10-14T01:16:48

khodya, 2025-02-12:
PR: https://github.com/project-chip/connectedhomeip/pull/37518

PR #37518 (2025-02-25): Update ZAP file to fix `lit-icd-app` conformance
Updated the ZAP file to fix conformance issues found in the go/matter-sdk-conformance report. Fixes #37297 

- Added `Power Source` cluster with `BAT` basic conformance.
- Bumped versions of several clusters to match the required versions.
- Disabled attributes as specified by conformance.

#### List of found issues

- [x] Endpoint: 0 - Cluster: 31 (0x1f) AccessControl - Problem: Attribute 0x01 is included, but is disallowed by conformance. Conformance: EXTS, implemented features:
- [ ] Endpoint: 0 - Cluster: 49 (0x31) NetworkCommissioning - Pro [... cut]