project-chip/connectedhomeip #37311: [BUG]

agatah2333, 2025-01-29:
### Reproduction steps

# Bug Report: Matter AccessControl Write ACL Command Accepts Invalid Empty Input

## Summary
The all-cluster-app fails to properly validate an empty array parameter `[]` when processing the AccessControl Write ACL command. The application incorrectly accepts this invalid input without any parameter validation. This issue affects all versions of the code.

## Steps to Reproduce
1. Start all-cluster-app
2. Execute the following command:
   ```bash
   ./chip-tool accesscontrol write acl '[]' 0x654324 0
   ```
3. Observe the command execution results

## Current Behavior
The command is accepted and executed by the system without reporting any errors, despite having an invalid empty array parameter.

## Expected Behavior
The system should:
1. Detect that an empty array `[]` is an invalid ACL parameter
2. Reject the command execution
3. Return an appropriate error message indicating the need for valid ACL data

## Impact
- Could lead to unexpected behavior in access control management
- Particularly concerning given the isolation of ACL between controllers

## Additional Notes
- Parameter validation should be implemented at the command parsing level
- This issue affects the basic security functionality of the Matter access control system
- Empty ACL parameters should be explicitly rejected with a meaningful error message

## Differentiation from Related Issues
This issue is distinct from #36535 and #36420. During testing, attempts to use controller A to rewrite the ACL on controller B were unsuccessful, indicating that ACLs between controllers are isolated. This isolation makes proper parameter validation even more critical, as each controller's ACL modifications operate independently.

## Root Cause Analysis
The lack of parameter validation appears to be a separate issue from the controller isolation behavior. While the isolation of ACLs between controllers may be by design, accepting invalid empty ACL parameters is clearly a bug that needs to be addressed independently.

### Bug prevalence

each time

### GitHub hash of the SDK that was being used

57ff3e6

### Platform

other

### Platform Version(s)

all version

bzbarsky-apple, 2025-01-30:
> despite having an invalid empty array parameter.

@agatah2333 What makes you say having an empty list there is invalid, exactly?  Nothing in the spec requires that list to be non-empty, as far as I can tell.

bzbarsky-apple, 2025-01-30:
In particular, please cite the relevant spec text if you decide this issue needs to be reopened.

closed 2025-01-30 by bzbarsky-apple

agatah2333, 2025-02-03:
@bzbarsky-apple I’m noting that setting an empty ACL in the Matter specification can lead to losing all remote access, causing device lock, and potentially needing a factory reset. Is this command intended to lock the device by setting an empty ACL?

bzbarsky-apple, 2025-02-04:
It's not "intended" to do anything, but it's _allowed_ to do that if you really want to.