project-chip/connectedhomeip #37317: [Build] Unable to cross compile the examples for target arm64 linux when running it in native ubuntu OS

aggaddam, 2025-01-30:
### Build issue(s)

I cannot build( Cross-compiling) the matter lights example for target arm64 Linux. Has anyone tried to build the same from a Linux native environment for an arm 64-bit target with Linux? Please share the steps to build cross compiled binary from linux native pc

1. Installed docker in my ubuntu pc
2. Pulled docker image for cross-compiling i.e docker pull ghcr.io/project-chip/chip-build-crosscompile:98
3. docker run --rm -it \
    -v "$(pwd):/workdir" \
    ghcr.io/project-chip/chip-build-crosscompile:98 \
    /bin/bash

`root@111d37527e29:/workdir/connectedhomeip# ./scripts/build/build_examples.py                         --target linux-arm64-chip-cert-clang                         build
2025-01-30 09:09:16 INFO    Building targets: linux-arm64-chip-cert-clang
2025-01-30 09:09:16 INFO    Preparing builder 'linux-arm64-chip-cert-clang'
2025-01-30 09:09:16 INFO    Generating /workdir/connectedhomeip/out/linux-arm64-chip-cert-clang
2025-01-30 09:09:16 INFO    Generating linux-arm64-chip-cert-clang
2025-01-30 09:09:17 INFO    Generating compile_commands took 34ms
2025-01-30 09:09:17 INFO    Done. Made 9202 targets from 414 files in 393ms
2025-01-30 09:09:17 INFO    Command ['bash', '-c', '\nPKG_CONFIG_PATH="/opt/ubuntu-22.04.1-aarch64-sysroot/lib/aarch64-linux-gnu/pkgconfig" \\\n gn gen --check --fail-on-unused-args --export-compile-commands --root=/workdir/connectedhomeip \'--args=is_clang=true target_cpu="arm64" sysroot="/opt/ubuntu-22.04.1-aarch64-sysroot"\' /workdir/connectedhomeip/out/linux-arm64-chip-cert-clang'] completed
2025-01-30 09:09:17 INFO    Building /workdir/connectedhomeip/out/linux-arm64-chip-cert-clang
2025-01-30 09:09:17 INFO    Building linux-arm64-chip-cert-clang
2025-01-30 09:09:17 INFO    ninja: Entering directory `/workdir/connectedhomeip/out/linux-arm64-chip-cert-clang'
2025-01-30 09:09:17 INFO    [1/216] c++ obj/src/ble/libBleLayer.BleUUID.cpp.o
2025-01-30 09:09:17 INFO    FAILED: obj/src/ble/libBleLayer.BleUUID.cpp.o 
2025-01-30 09:09:17 INFO    clang++ -MMD -MF obj/src/ble/libBleLayer.BleUUID.cpp.o.d -Wconversion --target=aarch64-linux-gnu -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wundef -Wimplicit-fallthrough -Wheader-hygiene -Wshorten-64-to-32 -Wformat-type-confusion -Wno-error=shorten-64-to-32 -Wthread-safety -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -fdiagnostics-color --sysroot=/opt/ubuntu-22.04.1-aarch64-sysroot -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS=1 -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Igen/include -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/include -c ../../src/ble/BleUUID.cpp -o obj/src/ble/libBleLayer.BleUUID.cpp.o
2025-01-30 09:09:17 INFO    In file included from ../../src/ble/BleUUID.cpp:20:
2025-01-30 09:09:17 INFO    ../../src/ble/BleUUID.h:25:10: fatal error: 'cstddef' file not found
2025-01-30 09:09:17 INFO       25 | #include <cstddef>
2025-01-30 09:09:17 INFO          |          ^~~~~~~~~
2025-01-30 09:09:17 INFO    1 error generated.
2025-01-30 09:09:17 INFO    [2/216] c++ obj/src/app/util/types.attribute-metadata.cpp.o
2025-01-30 09:09:17 INFO    FAILED: obj/src/app/util/types.attribute-metadata.cpp.o 
2025-01-30 09:09:17 INFO    clang++ -MMD -MF obj/src/app/util/types.attribute-metadata.cpp.o.d --target=aarch64-linux-gnu -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wundef -Wimplicit-fallthrough -Wheader-hygiene -Wshorten-64-to-32 -Wformat-type-confusion -Wno-error=shorten-64-to-32 -Wthread-safety -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -fdiagnostics-color --sysroot=/opt/ubuntu-22.04.1-aarch64-sysroot -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS=1 -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/include -c ../../src/app/util/attribute-metadata.cpp -o obj/src/app/util/types.attribute-metadata.cpp.o
2025-01-30 09:09:17 INFO    In file included from ../../src/app/util/attribute-metadata.cpp:17:
2025-01-30 09:09:17 INFO    In file included from ../../src/app/util/attribute-metadata.h:20:
2025-01-30 09:09:17 INFO    ../../src/app/util/basic-types.h:26:10: fatal error: 'cstdint' file not found
2025-01-30 09:09:17 INFO       26 | #include <cstdint>
2025-01-30 09:09:17 INFO          |          ^~~~~~~~~
2025-01-30 09:09:17 INFO    1 error generated.
2025-01-30 09:09:17 INFO    [3/216] c++ obj/src/ble/libBleLayer.BleError.cpp.o
2025-01-30 09:09:17 INFO    FAILED: obj/src/ble/libBleLayer.BleError.cpp.o 
2025-01-30 09:09:17 INFO    clang++ -MMD -MF obj/src/ble/libBleLayer.BleError.cpp.o.d -Wconversion --target=aarch64-linux-gnu -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wundef -Wimplicit-fallthrough -Wheader-hygiene -Wshorten-64-to-32 -Wformat-type-confusion -Wno-error=shorten-64-to-32 -Wthread-safety -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -fdiagnostics-color --sysroot=/opt/ubuntu-22.04.1-aarch64-sysroot -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS=1 -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Igen/include -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/include -c ../../src/ble/BleError.cpp -o obj/src/ble/libBleLayer.BleError.cpp.o
2025-01-30 09:09:17 INFO    In file included from ../../src/ble/BleError.cpp:25:
2025-01-30 09:09:17 INFO    ../../src/ble/BleError.h:37:10: fatal error: 'cstdint' file not found
2025-01-30 09:09:17 INFO       37 | #include <cstdint>
2025-01-30 09:09:17 INFO          |          ^~~~~~~~~
2025-01-30 09:09:17 INFO    1 error generated.
2025-01-30 09:09:17 INFO    [4/216] c++ obj/src/app/util/types.ember-strings.cpp.o
2025-01-30 09:09:17 INFO    FAILED: obj/src/app/util/types.ember-strings.cpp.o 
2025-01-30 09:09:17 INFO    clang++ -MMD -MF obj/src/app/util/types.ember-strings.cpp.o.d --target=aarch64-linux-gnu -O0 -g2 -fno-common -ffunction-sections -fdata-sections -fno-exceptions -fno-unwind-tables -fno-asynchronous-unwind-tables -fPIC -Wall -Werror -Wextra -Wshadow -Wunreachable-code -Wvla -Wformat -Wformat-nonliteral -Wformat-security -Wundef -Wimplicit-fallthrough -Wheader-hygiene -Wshorten-64-to-32 -Wformat-type-confusion -Wno-error=shorten-64-to-32 -Wthread-safety -Wno-deprecated-declarations -Wno-missing-field-initializers -Wno-unknown-warning-option -Wno-unused-parameter -fdiagnostics-color --sysroot=/opt/ubuntu-22.04.1-aarch64-sysroot -fno-strict-aliasing -fmacro-prefix-map=../../= -std=gnu++17 -fno-rtti -Wnon-virtual-dtor -D_LIBCPP_ENABLE_THREAD_SAFETY_ANNOTATIONS=1 -DCHIP_HAVE_CONFIG_H=1 -I../../src/include -I../../src -Igen/include -I../../zzz_generated/app-common -I../../config/standalone -I../../third_party/nlassert/repo/include -I../../third_party/nlio/repo/include -I../../third_party/nlfaultinjection/include -c ../../src/app/util/ember-strings.cpp -o obj/src/app/util/types.ember-strings.cpp.o
2025-01-30 09:09:17 INFO    In file included from ../../src/app/util/ember-strings.cpp:17:
2025-01-30 09:09:17 INFO    ../../src/app/util/ember-strings.h:19:10: fatal error: 'cstddef' file not found
2025-01-30 09:09:17 INFO       19 | #include <cstddef>
2025-01-30 09:09:17 INFO          |          ^~~~~~~~~
2025-01-30 09:09:17 INFO    1 error generated.
[... cut ...]
  File "/workdir/connectedhomeip/scripts/build/builders/builder.py", line 112, in _Execute
    self._runner.Run(cmdarray, title=title)
  File "/workdir/connectedhomeip/scripts/build/runner/shell.py", line 73, in Run
    raise Exception('Command %r failed: %d' % (cmd, code))
Exception: Command ['bash', '-c', '\nPKG_CONFIG_PATH="/opt/ubuntu-22.04.1-aarch64-sysroot/lib/aarch64-linux-gnu/pkgconfig" \\\n ninja -C /workdir/connectedhomeip/out/linux-arm64-chip-cert-clang src/tools/chip-cert'] failed: 1
root@111d37527e29:/workdir/connectedhomeip# 

`

mRoudierLg, 2025-01-30:
There may be several options:

- Either you are missing the "Ubuntu-22.04.1-aarch64-sysroot" folder which you need for the sysroot argument
- You can try running this command: `./scripts/run_in_build_env.sh "./scripts/build/build_examples.py --target linux-arm64-chip-cert-clang build"`
- Maybe but not sure, certain libs would be missing

If ever, you can refer to how the CI does it: [.yaml Linux CI](https://github.com/project-chip/connectedhomeip/blob/master/.github/workflows/examples-linux-arm.yaml)

aggaddam, 2025-01-30:
@mRoudierLg  I can build the examples with the docker image for target Arm 64-bit Linux. Now, the issue is I am unable to execute the example application on arm target hardware - getting an error like missing a few object files. Any thoughts on it?

`echo "Install Dependencies..."
sudo apt-get install -y git gcc g++ pkg-config libssl-dev libdbus-1-dev \
    libglib2.0-dev libavahi-client-dev ninja-build python3-venv python3-dev \
    python3-pip unzip libgirepository1.0-dev libcairo2-dev libreadline-dev \
    default-jre
sudo apt-get install -y libsdl2-dev    

echo "Clone repository..."
git clone --depth=1 --branch v1.4.0.0 https://github.com/project-chip/connectedhomeip.git
cd connectedhomeip
./scripts/checkout_submodules.py --shallow --platform linux

echo "Pull Docker Image"
docker pull ghcr.io/project-chip/chip-build-crosscompile:98
         
echo "Run Docker Image"
docker run --rm -it \
    -v "$(pwd):/var/connectedhomeip" \
    ghcr.io/project-chip/chip-build-crosscompile:98 \
    /bin/bash
    
echo "Build"
cd /var/connectedhomeip

git config --global --add safe.directory /var/connectedhomeip

source ./scripts/bootstrap.sh
source ./scripts/activate.sh

./scripts/run_in_build_env.sh \
 "./scripts/build/build_examples.py \
    --target linux-arm64-chip-tool-clang \
    build"`

andy31415, 2025-02-04:
https://github.com/project-chip/connectedhomeip/actions/workflows/examples-linux-arm.yaml does a cross compile test (runs with ubunt 24.04 sysroot described in https://github.com/project-chip/connectedhomeip/blob/master/integrations/docker/images/stage-1/chip-build-crosscompile/README.md

You require a similar O/S and the same libraries installed. In particullar current https://github.com/project-chip/connectedhomeip/blob/master/integrations/docker/images/stage-1/chip-build-crosscompile/start-sysroot-vm.sh#L112 seems to do:

```yaml
packages:
  - g++
  - gcc
  - libavahi-client-dev
  - libcairo2-dev
  - libdbus-1-dev
  - libgirepository1.0-dev
  - libglib2.0-dev
  - libpcsclite-dev
  - libreadline-dev
  - libsdl2-dev
  - libssl-dev
  - ninja-build
  - pkg-config
  - unzip
```

Your error seems to indicate you do not have libssl.