project-chip/connectedhomeip #37342: Implement spec requirement of "ACL check before existence check"

andy31415, 2025-01-31:
This request is to implement https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/9024

This should be done after #37322 merges and we should fix the TODO in `WriteHandler::CheckWriteAllowed`.

The issue is that we the existing ordering allows someone with insufficient rights on a cluster still determine what attributes it supports by seeing a difference between "access denied" and "unsupported attribute".

#35336 tried to address it a long time ago, however it was failing CI and code was restructured significantly since then. We can pick up some bits/ideas from there ... like we need to fix BOTH attributes and commands. However overall I expect we need very different/new code (we do not need to be inside ember/ember-compatibility anymore).

khodya, 2025-02-07:
1) Get ACL
2) throw error if it is not found 
3) only then read attribute

Alami-Amine, 2025-06-04:
@bzbarsky-apple do you mind elaborating on what exactly needs to be fixed in spec?

closed 2025-07-10 by mergify[bot]