project-chip/connectedhomeip #37355: [Documentation] Which XML Definition for Door Lock Cluster is Used in Upstream Code?

pimpalemahesh, 2025-02-03:
### Documentation issues

I've noticed two different XML definitions for the clusters in the codebase:
[HandWritten](https://github.com/project-chip/connectedhomeip/blob/master/src/app/zap-templates/zcl/data-model/chip/door-lock-cluster.xml) and [Spec Generated](https://github.com/project-chip/connectedhomeip/blob/master/data_model/1.4/clusters/DoorLock.xml)

I have few questions:

1. Which XML file is currently being used by the ZAP tool and code generation?
2. What is the purpose of maintaining two separate definitions?
3. Is there a plan to consolidate these definitions or maintain them separately?

### Anything else?

I found that some commands are missing in the handwritten XML file, and the emberAfCommandCallbacks in src/app/clusters/server.cpp files are only defined for commands present in the handwritten version. Additionally, there are commands present in the spec-generated XML that do not have corresponding callbacks defined in the server.cpp files

andy31415, 2025-02-03:
ZAP uses https://github.com/project-chip/connectedhomeip/tree/master/src/app/zap-templates/zcl/data-model/chip

The other ones are used by validation/testing scripts as they were always script-generated from the spec (so more correct). We are also transitioning to encourage/enforce generation for the ZAP XMLs via https://github.com/project-chip/alchemy, however that is a recent development.

andy31415, 2025-02-04:
https://github.com/project-chip/connectedhomeip/blob/master/src/app/common/templates/config-data.yaml defines lists of items that are using interface-only clusters (specifically CommandHandlerInterfaceOnlyClusters) and those will not have the ember callbacks used as they assume a CommandHandlerInterface would handle them.

closed 2025-02-04 by andy31415

pimpalemahesh, 2025-02-05:
@andy31415 Thanks for the reply!