project-chip/connectedhomeip #37359: [BUG] Need update BLE related part for bluez 5.78

TE-N-ElvenWang, 2025-02-03:
### Reproduction steps

On Linux environment, the bluez 5.78 updated the api which the indication won't feed back via old APIs.
An issue for bluez upgrading are in: https://github.com/bluez/bluez/issues/1081 

This will cause that the end device will always waiting for GATT ready for send signal and then timeout.

Tested on latest master tree still have such issue.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

master

### Platform

IMX8, core

MatjazH-sue, 2025-06-26:
You can quickly fix this without adverse problems like this:
`--- a/src/platform/Linux/bluez/BluezEndpoint.cpp
+++ b/src/platform/Linux/bluez/BluezEndpoint.cpp
@@ -189,11 +189,11 @@ gboolean BluezEndpoint::BluezCharacteristicAcquireNotify(BluezGattCharacteristic
     }
 
     conn->SetupNotifyHandler(fds[0], isAdditionalAdvertising);
-    bluez_gatt_characteristic1_set_notify_acquired(aChar, TRUE);
+    //bluez_gatt_characteristic1_set_notify_acquired(aChar, TRUE);
     conn->SetNotifyAcquired(true);
 
     GUnixFDList * fdList = g_unix_fd_list_new_from_array(&fds[1], 1);
-    bluez_gatt_characteristic1_complete_acquire_notify(aChar, aInvocation, fdList, g_variant_new_handle(0), conn->GetMTU());
+    //bluez_gatt_characteristic1_complete_acquire_notify(aChar, aInvocation, fdList, g_variant_new_handle(0), conn->GetMTU());
     g_object_unref(fdList);
 
     BLEManagerImpl::HandleTXCharCCCDWrite(conn);
`

**My best understanding of what's going wrong:**
CHIP's implementation of the BlueZ GATT APIs is incorrect and sets up two separate channels for receiving BLE Indication Confirmations; the Confirm method is a "D-BUS channel" and the AcquireNotify method establishes a "UNIX socket" channel.

And while the D-BUS handler is implemented on the CHIP side, the socket handler is left as a non-functional stub for some reason (`NotifyHandlerCallback` in `BluezConnection.cpp`).

BlueZ API documentation states that only one channel may be used for a given BLE connection, and that if AcquireNotify is used, Indication Confirmations will _only_ be passed through the socket. However, in practice this was not the case on BlueZ as there was a bug sending confirmations on both channels regardless, so the CHIP implementation worked only by accident. This bug was fixed by BlueZ 5.78 (specifically, here:  and now the indications only come through the UNIX socket and are not handled by CHIP. This breaks Matter Bluetooth Transport Protocol (and thus network setup).

The fix above works because as long as long as BlueZ hasn't been given a socket to send notifications over, it will continue to use the Confirm method (as it has until 5.78). However, a proper fix should probably either abandon using AcquireNotify in favor of StartNotify (I think this is preferable as Matter BTP causes very little traffic) or implement the socket callback handler properly.

Eldho1416, 2025-07-01:
Hi @MatjazH-sue ,

Thanks for the patch, it works. But this be a critical bug in future since people will start using latest bluez.

How do we raise this issue directly to the CHIP moderators?