project-chip/connectedhomeip #37391: Ensure all zapxml clusters are generated by alchemy. Record errors for the ones that have differences and resolve.

andy31415, 2025-02-04:
https://github.com/project-chip/connectedhomeip/tree/master/src/app/zap-templates/zcl/data-model/chip contains our zapxml data model.

Historically we have found that manual editing based on spec easily contains errors and we are encouraging people to use Alchemy - https://github.com/project-chip/alchemy - to generate them.

Only a subset is currently generated like that (grep `XML generated by Alchemy; DO NOT EDIT.`):

```yaml
- src/app/zap-templates/zcl/data-model/chip/camera-av-settings-user-level-management-cluster.xml
- src/app/zap-templates/zcl/data-model/chip/camera-av-stream-management-cluster.xml
- src/app/zap-templates/zcl/data-model/chip/chime-cluster.xml
- src/app/zap-templates/zcl/data-model/chip/device-energy-management-cluster.xml
- src/app/zap-templates/zcl/data-model/chip/device-energy-management-mode-cluster.xml
- src/app/zap-templates/zcl/data-model/chip/energy-evse-cluster.xml
- src/app/zap-templates/zcl/data-model/chip/energy-evse-mode-cluster.xml
- src/app/zap-templates/zcl/data-model/chip/push-av-stream-transport-cluster.xml
- src/app/zap-templates/zcl/data-model/chip/tls-certificate-management-cluster.xml
- src/app/zap-templates/zcl/data-model/chip/water-heater-management-cluster.xml
- src/app/zap-templates/zcl/data-model/chip/water-heater-mode-cluster.xml
- src/app/zap-templates/zcl/data-model/chip/webrtc-provider-cluster.xml
- src/app/zap-templates/zcl/data-model/chip/webrtc-requestor-cluster.xml
- src/app/zap-templates/zcl/data-model/chip/zone-management-cluster.xml
```

All others should be converted to use alchemy. In case alchemy shows differences, we should determine how to resolve as they represent mismatch between specification and SDK (or alchemy bugs are also possible, as well as unparsable SDK differences).

We need:
   - convert every cluster
   - when conversion does not work, create an issue to resolve (and we will follow up on it)

tersal, 2025-02-19:
Used Alchemy to generate Thermostat and AirPurifier, there are some error not specifically related to said modules but looking to confirm if the output for a "non-error reported" cluster is OK. Also, going through the currently reported errors to assess the severity of them.

tersal, 2025-02-24:
Generated XML and ZAP files for AirQuality cluster without apparent issues and now in review.

However there were issues with the next two clusters:

- Thermostat, generated normally with Alchemy but there are errors when using zap_regen to build afterwards. Seem to be SQL related errors.
- ColorControl, generated correctly with Alchemy and zap_regen. However some example apps are failing during compilation most likely because a type change (from uint_8 to a specific ColorControl defined enum).

tersal, 2025-05-14:
There is a rather comprehensive list of discrepancies in the Media clusters described in [here](https://github.com/project-chip/connectedhomeip/issues/32669).

tersal, 2025-08-26:
OnOff cluster in review.

ApplicationLauncher and ApplicationBasic migration is in review with requested changes to the spec of ApplicationBasic. 

There are some changes to consider but working through them with help from the media team.

tersal, 2025-10-02:
Created [PR in spec repo](https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/12267) for ICDManagement, Administrator Commissioning, TimeSync and SmokeCOAlarm so errata can be used to solve discrepancies. Still need approvals or merge so it can be used.

khodya, 2026-11-03:
Missing clusters: 
- Channel 
- ModeBase