project-chip/connectedhomeip #37393: Run `include-what-you-use` as part of CI, enforce it

andy31415, 2025-02-04:
We have frequent bad includes logic, which often can hide layering issues and sometimes makes the includes overly complicated (include much more than what we need).

#32046 has an example of a manual IWYU run, however it also describes that manual fixes were needed.

We should establish a process of running IWYU and see if we can enforce it in CI (even for a subset of files that we keep increasing) the same like we enfordce other linters like clang-tidy, manual linters and zap regen.

I expect we want:
- [ ] ensure iwyu_tool is available in CI/documented how to run
- [ ] update configurations to not require manual fixes
- [ ] make "fix my includes" easy to run manually (likely on linux and darwin, some script in scripts/tools or similar would be useful)
- [ ] make the check part of CI

soares-sergio, 2025-02-04:
Maybe we could run this on every PR but only on the changed files in the respective PR.

andy31415, 2025-02-06:
This likely also needs some investigation in what "include what you use" tools are available/possible. I was told it is quite a complicated problem in general, we want something reasonably maintained and working.