project-chip/connectedhomeip #37422: [CERT-TEST-FAILURE] [TC-DD-1.3] Python script should provide more descriptive steps for ATL representatives

Rajashreekalmane, 2025-02-06:
### Feature Area

Other

### Test Case

TC-DD-1.3

### Reproduction steps

**Issue Description**
The Python script for TC-DD-1.3 lacks clear instructions for ATL representatives on how to execute the test. The current steps are too generic and require more details about expected behavior, verification criteria, and necessary inputs.

- Observe that some test steps do not clearly specify what actions the ATL representative needs to take.
- Some steps assume knowledge about NFC readers and payload verification, which should be explicitly detailed.

**Expected Behavior**
The script should provide clear, step-by-step guidance on what the ATL rep should do.
Each verification step should include expected values or responses for better understanding.
Additional comments or logging should be added to guide test execution.

**Python script reference :** https://github.com/project-chip/matter-test-scripts/blob/main/onboarding_payload_test_suite/tcdd13/tcdd13.py

### Bug prevalence

whevenr I do

### GitHub hash of the SDK that was being used

dfghbjk

### Platform

raspi

doublemis1, 2025-02-06:
Talking about the issue F2F.
The issue is not functional but the description transparency.
The issue should not be a cert blocker but as an enhancement.
Based on F2F discussion, such test case description enhancement should not be a part of Matter 1.4.1 (since it does not fix any functional test step).
The issue should be assigned to the Matter 1.5.0

cjandhyala, 2025-02-06:
@raul-marquez-csa could you take a look at this ?

raul-marquez-csa, 2025-07-29:
@Rajashreekalmane will provide suggestions/examples where the steps can be made more descriptive for ATL representatives. Will also identify and reference the unclear steps specifically to improve clarity.

Rajashreekalmane, 2025-07-29:
Hi @raul-marquez-csa, here are a few suggestions to make the script more descriptive and helpful for ATL representatives:

- Improve step labels in create_test_steps() to clearly state what each step is verifying.
- Add informative log messages in the execute() method to guide ATL reps through the process (e.g., "Scan NFC tag now", "Verifying payload prefix...").
- Include expected value context in logs (e.g., "Expecting a 12-bit discriminator", "Prefix should start with 'MT:'").
- Comment key parts of the code where values are parsed or validated, to help ATL reps understand the logic and data flow.

Let me know if you'd like me to elaborate on any of these.

raul-marquez-csa, 2025-07-31:
PR
[TC-DD-1.3 script clarification update #671](https://github.com/project-chip/matter-test-scripts/pull/671)

closed 2025-10-08 by raul-marquez-csa