project-chip/connectedhomeip #37426: [Build] Ninja error while setting up CHIP Tool

KoertSchoen, 2025-02-06:
### Build issue(s)

1. I started setting up the CHIP Tool on Linux (Ubuntu 22.04.2 LTS), following the official documentation: https://project-chip.github.io/connectedhomeip-doc/guides/BUILDING.html
2. Environment is setup and passes all checks (command: source scripts/activate.sh)
3. The following command "gn gen out/host" also gives no errors.
4. But then running "ninja -C out/host check" gives the error:
ninja: Entering directory `out/host'
ninja: error: '../../third_party/jsoncpp/repo/src/lib_json/json_reader.cpp', needed by 'obj/third_party/jsoncpp/repo/src/lib_json/jsoncpp.json_reader.cpp.o', missing and no known rule to make it
5. I've looked online to find a solution to solve this error, but I haven't gotten closer to a solution, does anyone know how to solve this?
Additional information: I forked the connectedhomeip repository and ran the commands in the upper directory, so in ~/connectedhomeip.
Thanks in advance!

bzbarsky-apple, 2025-02-07:
@KoertSchoen Just to check: you have the jsoncpp submodule initialized?  Does the file `third_party/jsoncpp/repo/src/lib_json/json_reader.cpp` exist?  If not, why not?

KoertSchoen, 2025-02-10:
@bzbarsky-apple No it was not initialized, I assumed this was automatically done when running the activate script. Do you know if there is a command to initialize all the submodules at once? Thanks for the quick reply.

closed 2025-02-10 by KoertSchoen