project-chip/connectedhomeip #37497: [BUG] v1.4.1 Not able to run Python  test using TH

rquidute, 2025-02-10:
### Reproduction steps

We built a `chip-cert-bin` image using TOT at the time of `v1.4-branch` (SHA `37a2d68a92f9d379ea46a236a9653cde05eeeaa3`). 

When we try to run any Python Test we are getting this error:

> INFO       | 2025-02-10 18:05:15.824977 | Sending command to SDK container: python3 /root/python_testing/scripts/sdk/test_harness_client.py commission --trace-to json:log --commissioning-method on-network --discriminator 3840 --passcode 20202021
> PYTHON_TEST | 2025-02-10 18:05:16.948861 | Traceback (most recent call last):
> PYTHON_TEST | 2025-02-10 18:05:16.951518 |   File "/root/python_testing/scripts/sdk/test_harness_client.py", line 25, in <module>
> PYTHON_TEST | 2025-02-10 18:05:16.956753 | from matter_testing_support import (
> PYTHON_TEST | 2025-02-10 18:05:16.959531 |   File "/root/python_testing/scripts/sdk/matter_testing_support.py", line 47, in <module>
> PYTHON_TEST | 2025-02-10 18:05:16.964367 | from chip import ChipDeviceCtrl  # Needed before chip.FabricAdmin
> PYTHON_TEST | 2025-02-10 18:05:16.967166 |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> PYTHON_TEST | 2025-02-10 18:05:16.969789 |   File "/usr/local/lib/python3.12/dist-packages/chip/ChipDeviceCtrl.py", line 48, in <module>
> PYTHON_TEST | 2025-02-10 18:05:16.975739 | from . import FabricAdmin
> PYTHON_TEST | 2025-02-10 18:05:16.978357 |   File "/usr/local/lib/python3.12/dist-packages/chip/FabricAdmin.py", line 24, in <module>
> PYTHON_TEST | 2025-02-10 18:05:16.984311 | from chip import CertificateAuthority, ChipDeviceCtrl
> PYTHON_TEST | 2025-02-10 18:05:16.987055 |   File "/usr/local/lib/python3.12/dist-packages/chip/CertificateAuthority.py", line 27, in <module>
> PYTHON_TEST | 2025-02-10 18:05:16.992624 | from chip import ChipStack, FabricAdmin
> PYTHON_TEST | 2025-02-10 18:05:16.995501 |   File "/usr/local/lib/python3.12/dist-packages/chip/ChipStack.py", line 39, in <module>
> PYTHON_TEST | 2025-02-10 18:05:17.000485 | from .clusters import Attribute as ClusterAttribute
> PYTHON_TEST | 2025-02-10 18:05:17.003209 |   File "/usr/local/lib/python3.12/dist-packages/chip/clusters/__init__.py", line 18, in <module>
> PYTHON_TEST | 2025-02-10 18:05:17.009069 | from . import Attribute, CHIPClusters, Command, Objects  # noqa: F401
> PYTHON_TEST | 2025-02-10 18:05:17.014440 | ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
> PYTHON_TEST | 2025-02-10 18:05:17.016647 |   File "/usr/local/lib/python3.12/dist-packages/chip/clusters/Objects.py", line 6913
> PYTHON_TEST | 2025-02-10 18:05:17.018669 |     <<<<<<< HEAD
> PYTHON_TEST | 2025-02-10 18:05:17.020905 |     ^^
> PYTHON_TEST | 2025-02-10 18:05:17.025678 | SyntaxError: invalid syntax

TH execution log: 

[\[SDK\]TH_Python_tests_Error_in_tot_v1_4_branch.log](https://github.com/user-attachments/files/18740830/SDK.TH_Python_tests_Error_in_tot_v1_4_branch.log)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

37a2d68a92f9d379ea46a236a9653cde05eeeaa3

### Platform

raspi

rquidute, 2025-02-10:
Hi @cecille are you aware of this issue?

raju-apple, 2025-02-10:
@rquidute can you also post the SHA on v1.4-branch where the TH worked fine ?

rquidute, 2025-02-10:
@raju-apple We built an image using this SHA `ebf0be48787a05e85fdc904aca9dd361fdf8f81a` and it was fine.

cecille, 2025-02-10:
Pretty sure that was a bad merge on that last cherry pick. I will put up a fix with just that change.

cecille, 2025-02-10:
https://github.com/project-chip/connectedhomeip/pull/37500