project-chip/connectedhomeip #37537: Remove temporary aliases from matter_testing.py

asirko-soft, 2025-02-12:
Remove temporary function aliases in matter_testing.py

Currently matter_testing.py contains temporary aliases for functions from the matchers, timeoperations, and conversions modules. These aliases should be removed after transition period and all code should be updated to use the module-qualified names directly.

For example:
- Replace `type_matches` with `matchers.is_type`
- Replace `id_str` with `conversions.format_decimal_and_hex`
- Replace `cluster_id_str` with `conversions.cluster_id_str`

Steps:
1. Search for all usages of these aliased functions
2. Replace with module-qualified versions
3. Remove the aliases from matter_testing.py
4. Update any documentation referring to these functions

abalestra22, 2025-07-01:
PR: https://github.com/project-chip/connectedhomeip/pull/39796

abalestra22, 2025-07-14:
PR: https://github.com/project-chip/connectedhomeip/pull/40004

abalestra22, 2025-07-15:
Created a new PR with a clean tree history: https://github.com/project-chip/connectedhomeip/pull/40031
Had some issues with some submodules and .gitmodules

abalestra22, 2025-08-25:
Working on fixing conflicts in first part PR, also will remove timeoperations module changes (based on PR comments)

abalestra22, 2025-08-26:
Fixed conflicts, waiting for reviews so we can merge first PR and continue with the rest of small PRs.

abalestra22, 2025-09-04:
I followed the approach of creating small PRs for the refactor. Right now, I’m waiting for reviews on the first PR before merging it. The remaining PRs will each address one alias — decorators, runner, and conversions — so in theory, there should be three more in total. After merging the first PR, this work will be on hold for now since I’m focusing on the SU tickets.

abalestra22, 2025-09-10:
CI/CD jobs are now failing and I'm researching if those issues are related to my changes.

closed 2025-09-16 by mergify[bot]